intel/ipu6-drivers #254: Dragonfly G4 laptop webcam with Ubuntu 24.04 not working/recognized

dillrich, 2024-07-15:


andyyeh75, 2024-07-29:
I am from Intel camera driver team. Kindly be informed that the integration and validation are required for power on a IPU Webcam. We didn't receive PC vendor's request to enable the camera on this device yet. Suggest to get contact with HP customer service. Hope to hear good news from HP.

goinglinux, 2024-12-10:
Affects me, too.

tejasraman, 2025-02-09:
I would love to see support for the G4 webcam in Linux - now that there's sound support for it the webcam is the last real issue with the G4 on Linux

yrenisio, 2025-02-09:
@tejasraman, could you point to the sound fix please ? Sorry for the out of subject question.

tejasraman, 2025-02-09:
> @tejasraman, could you point to the sound fix please ? Sorry for the out of subject question. 
> 
> 
> 

no problem! Here it is: https://github.com/nicholascw/aur-snd-hda-scodec-cs35l41-sandwalker

dillrich, 2025-02-09:
What do I do to install the fix?

Rich

tejasraman, 2025-02-09:
> What do I do to install the fix?
> 
> Rich
> 
> 
> > @tejasraman <https://github.com/tejasraman>, could you point to the sound
> > fix please ? Sorry for the out of subject question.
> >
> > no problem! Here it is:
> > https://github.com/nicholascw/aur-snd-hda-scodec-cs35l41-sandwalker

You have to compile the kernel module

DavidDohmen, 2025-04-22:
Very disappointing that even in such an expensive business laptop and at these times, vendors like HP still do not even try to make the hardware run under linux. This will probably be my last HP device. Has anybody reached out to their support as suggested by andyyeh75? Couldn't find a fitting post on the HP support forums. Just someone complaining about the broken sound.

tejasraman, 2025-04-22:
> Very disappointing that even in such an expensive business laptop and at these times, vendors like HP still do not even try to make the hardware run under linux. This will probably be my last HP [...]

I'm considering reaching out to HP's EliteBook support team and asking for an escalation to someone who can make that request. I would love to have Linux support on this laptop but I don't think it'll happen until HP gets on board. 

They have made Linux patches upon request before (e.g a BIOS patch for the dragonfly G3 sound was made after a support request) so maybe they'll do it again...

dillrich, 2025-04-22:
You should reach out!  The laptop is awesome, just doesn't work with the integrated camera :(

DavidDohmen, 2025-04-22:
Let's all do it. Hearing it several times is probably moving the needle more than if just one of us is doing it.

DavidDohmen, 2025-05-15:
Hi @andyyeh75 , has anybody from HP reached out yet? It's shameful, there still is no proper linux support for this device.

tejasraman, 2025-05-15:
> Let's all do it. Hearing it several times is probably moving the needle more than if just one of us is doing it. I can only call them (no way to email them) but will do that anyway, tomorrow.
> 
> EDIT:
> I posted in their support forum. Maybe like and coment for more visibility https://h30434.www3.hp.com/t5/Notebook-Video-Display-and-Touch/HP-Dragonfly-G4-no-linux-driver-for-camera/m-p/9376224#M284032
> 
> EDIT: 
> Called the service today and let them know I'd like to have linux drivers for its webcam. The gentlemen promised to forward it to the respective people. Let's keep it coming!

@DavidDohmen that is amazing! I'll do the same when I have time (final exams here have been CRAZY lol)

andyyeh75, 2025-05-17:
> Hi [@andyyeh75](https://github.com/andyyeh75) , has anybody from HP reached out yet? It's shameful, there still is no proper linux support for this device.

Because of duty shift, Kindly contact with @sujimmy for following up your technical support request. Thanks.

DavidDohmen, 2025-05-23:
Thanks for the info!
@sujimmy welcome to this topic and thanks for taking over!
As a direct hp and indirect intel customer, I'd like to understand how this works.
Why are you depending on hp requesting the drivers for this webcam? Can you not just commit the drivers to the linux kernel?

My post on the hp support forum had several +1s but so far we're being ignored there. Via phone support I also raised the problem and they said they'd "forward" it but I don't know, if anybody has reached out to you. Can you give us some more insights from your side on that?

DavidDohmen, 2025-07-03:
Congrats for getting it to work.
Just a quick question from someone who has never before patched a Linux kernel.
Will this survive kernel upgrades and OS updates in general or will it have to be recompiled and applied everytime there is a new kernel version? Is there a difference between minor and major kernel updates?

nicholascw, 2025-08-03:
Just adding a data point here, I tried to use the DKMS method on 6.15 without a success, but it seems the camera working out of the box with 6.16 mainline kernel on Arch Linux at this point.

nicholascw, 2025-08-05:
> That's fantastic! Is the sound also working by now, or is still necessary to patch this part here: https://github.com/nicholascw/aur-snd-hda-scodec-cs35l41-sandwalker

It's pretty off-topic, but yes, we still need to patch it as I did not follow through the LKML to get it merged.

tejasraman, 2025-08-05:
> > That's fantastic! Is the sound also working by now, or is still necessary to patch this part here: https://github.com/nicholascw/aur-snd-hda-scodec-cs35l41-sandwalker
> 
> It's pretty off-topic, but yes, we still need to patch it as I did not follow through the LKML to get it merged.

tysm for the quick response @nicholascw - hope to see it get merged soon!

DavidDohmen, 2025-09-04:
> I am on latest kernel (6.16.4) but the camera doesn't seem to be working.

Which OS are you running? Maybe it's some Arch linux specific additions that not all distros have?

niyoko, 2025-09-06:
> > I am on latest kernel (6.16.4) but the camera doesn't seem to be working.
> 
> Which OS are you running? Maybe it's some Arch linux specific additions that not all distros have?

It's base archlinux, not manjaro or something.

kolAflash, 2026-01-18:
Dragonfly G4  
[Fedora-43 live system _(Linux-6.17, 2025-10-25)_](https://dl.fedoraproject.org/pub/fedora/linux/releases/43/Workstation/x86_64/iso/Fedora-Workstation-Live-43-1.6.x86_64.iso): webcam works, speakers work sometimes _(couldn't yet figure out why)_  
[Ubuntu-25.10 live system _(Linux-6.17, image 2025-10-07)_](https://releases.ubuntu.com/25.10/ubuntu-25.10-desktop-amd64.iso): audio works, but webcam does NOT  
Couldn't yet figure out how to get both working in one distro.