intel/ipu6-drivers #258: Doesn't work on ThinkPad X1 Carbon Gen 10 with Ubuntu 24.04

80kk, 2024-07-23:
I have Lenovo ThinkPad X1 Carbon Gen 10 with Ubuntu 24.04LTS and 6.8.0-38-generic stock kernel installed. Using apt I've installed `intel-usbio-dkms` and then all four repositories as mentioned on https://github.com/intel/ipu6-drivers?tab=readme-ov-file For ipu6-drivers I decided to use DKMS method, it built and loaded without any issues:
```
# dkms status
ipu6-drivers/0.0.0, 6.8.0-38-generic, x86_64: installed
usbio-drivers/0~git202312141918.78ffb706-0ubuntu2, 6.8.0-38-generic, x86_64: installed
```
I've also tried Intel Integrated Image Processing Unit 6 driver from Additional drivers but it didn't make any difference except ipu6-drivers module didn't build because it was already provided.
```
# cat /sys/class/video4linux/*/name
Intel IPU6 CSI-2 0
Intel IPU6 CSI2 BE SOC 2
Intel IPU6 CSI2 BE SOC 3
Intel IPU6 CSI2 BE SOC 4
Intel IPU6 CSI2 BE SOC 5
Intel IPU6 CSI2 BE SOC 6
Intel IPU6 CSI2 BE SOC 7
ov2740 13-0036
Intel IPU6 CSI-2 1
Intel IPU6 CSI-2 2
Intel IPU6 CSI-2 3
Intel IPU6 CSI-2 4
Intel IPU6 CSI-2 5
Intel IPU6 CSI-2 6
Intel IPU6 CSI-2 7
Intel IPU6 CSI2 BE SOC 0
Intel IPU6 CSI2 BE SOC 1
Intel IPU6 BE SOC capture 0
Intel IPU6 BE SOC capture 1
Intel IPU6 BE SOC capture 2
Intel IPU6 BE SOC capture 3
Intel IPU6 BE SOC capture 4
Intel IPU6 BE SOC capture 5
Intel IPU6 BE SOC capture 6
Intel IPU6 BE SOC capture 7
```
```
# lspci 
00:00.0 Host bridge: Intel Corporation Alder Lake-U15 Host and DRAM Controller (rev 04)
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-UP3 GT2 [Iris Xe Graphics] (rev 0c)
00:04.0 Signal processing controller: Intel Corporation Alder Lake Innovation Platform Framework Processor Participant (rev 04)
00:05.0 Multimedia controller: Intel Corporation Alder Lake Imaging Signal Processor (rev 04)
00:06.0 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x4 Controller #0 (rev 04)
00:07.0 PCI bridge: Intel Corporation Alder Lake-P Thunderbolt 4 PCI Express Root Port #0 (rev 04)
00:07.2 PCI bridge: Intel Corporation Alder Lake-P Thunderbolt 4 PCI Express Root Port #2 (rev 04)
00:08.0 System peripheral: Intel Corporation 12th Gen Core Processor Gaussian & Neural Accelerator (rev 04)
00:0a.0 Signal processing controller: Intel Corporation Platform Monitoring Technology (rev 01)
00:0d.0 USB controller: Intel Corporation Alder Lake-P Thunderbolt 4 USB Controller (rev 04)
00:0d.2 USB controller: Intel Corporation Alder Lake-P Thunderbolt 4 NHI #0 (rev 04)
00:0d.3 USB controller: Intel Corporation Alder Lake-P Thunderbolt 4 NHI #1 (rev 04)
00:12.0 Serial controller: Intel Corporation Alder Lake-P Integrated Sensor Hub (rev 01)
00:14.0 USB controller: Intel Corporation Alder Lake PCH USB 3.2 xHCI Host Controller (rev 01)
00:14.2 RAM memory: Intel Corporation Alder Lake PCH Shared SRAM (rev 01)
00:14.3 Network controller: Intel Corporation Alder Lake-P PCH CNVi WiFi (rev 01)
00:15.0 Serial bus controller: Intel Corporation Alder Lake PCH Serial IO I2C Controller #0 (rev 01)
00:16.0 Communication controller: Intel Corporation Alder Lake PCH HECI Controller (rev 01)
00:19.0 Serial bus controller: Intel Corporation Alder Lake-P Serial IO I2C Controller #0 (rev 01)
00:1c.0 PCI bridge: Intel Corporation Device 51b8 (rev 01)
00:1f.0 ISA bridge: Intel Corporation Alder Lake PCH eSPI Controller (rev 01)
00:1f.3 Audio device: Intel Corporation Alder Lake PCH-P High Definition Audio Controller (rev 01)
00:1f.4 SMBus: Intel Corporation Alder Lake PCH-P SMBus Host Controller (rev 01)
00:1f.5 Serial bus controller: Intel Corporation Alder Lake-P PCH SPI Controller (rev 01)
04:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller PM9A1/PM9A3/980PRO
08:00.0 Wireless controller [0d40]: MEDIATEK Corp. Device 4d75 (rev 01)
```
```
# lsusb 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 06cb:00fc Synaptics, Inc. 
Bus 003 Device 003: ID 8086:0b63 Intel Corp. USB Bridge
Bus 003 Device 004: ID 8087:0033 Intel Corp. AX211 Bluetooth
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

What else I can do to narrow down the issue?

alesrebec, 2024-09-30:
Same situation.
Lenovo X1 Carbon G10 with MIPI camera
6.8.0-45-generic
Upgraded today to Ubuntu 24.04.1 LTS
 
# dkms status
```
ipu6-drivers/0.0.0: added
ipu6-drivers/0~git202406240945.aecec2aa-0ubuntu2~24.04.1, 6.8.0-45-generic, x86_64: installed
v4l2loopback/0.12.7, 6.8.0-45-generic, x86_64: installed (WARNING! Diff between built and installed module!)
```

Running "sudo gst-launch-1.0 icamerasrc ! autovideosink"
I get:
`WARNING: erroneous pipeline: no element "icamerasrc"
`

alesrebec, 2024-09-30:
I just found this command:
`sudo -E gst-launch-1.0 icamerasrc buffer-count=7 device-name=ov2740-uf ! video/x-raw,format=NV12,width=1280,height=720 ! v4l2sink device=${DEVICE}`

If I run it my camera status light turns on. 
I then visited this web cam test page https://webcamtests.com/check and the camera is working!

80kk, 2024-10-01:
> Was just able to get it working on PopOs under kernel `6.9.3-76060903-generic`. Also installed through dkms. Do you get any output when running this?
> 
> ```shell
> sudo gst-launch-1.0 icamerasrc ! autovideosink
> ```
> 
> Will hopefully make a more detailed tutorial on the specific setup i got it working under.

I am getting:
```
# gst-launch-1.0 icamerasrc ! autovideosink
WARNING: erroneous pipeline: no element "icamerasrc"
# dkms status
ipu6-drivers/0.0.0, 6.8.0-41-generic, x86_64: installed
ipu6-drivers/0.0.0, 6.8.0-45-generic, x86_64: installed
usbio-drivers/0~git202312141918.78ffb706-0ubuntu2.1, 6.8.0-41-generic, x86_64: installed
usbio-drivers/0~git202312141918.78ffb706-0ubuntu2.1, 6.8.0-45-generic, x86_64: installed
```
I am guessing gstreamer can#t see this plugin but I can't find how to fix it.

80kk, 2024-10-01:
> I just found this command: `sudo -E gst-launch-1.0 icamerasrc buffer-count=7 device-name=ov2740-uf ! video/x-raw,format=NV12,width=1280,height=720 ! v4l2sink device=${DEVICE}`
>
> If I run it my camera [...]

```
WARNING: erroneous pipeline: no element "icamerasrc"
```
no luck here.

80kk, 2024-10-02:
> @80kk I also get this error if I run it again now but it works if it's placed in a script like this:
> 
> ```
> #!/bin/bash
> 
> DEVICE=/dev/video0
> 
> # Set the path to the GStreamer plugins
> export GST_PLUGIN_PATH=/usr/lib/gstreamer-1.0:/usr/local/lib/gstreamer-1.0
> 
> # Run camera command
> gst-launch-1.0 icamerasrc buffer-count=7 device-name=ov2740-uf ! video/x-raw,format=NV12,width=1280,height=720 ! v4l2sink device=${DEVICE}
> ```
> 
> Name it "run_camera.sh", make it executable:
> 
> `chmod +x run_camera.sh`
> 
> and then run it as sudo:
> 
> `sudo ./run_camera.sh `

I don't have `v4l2sink` module but I can't see it is required? However setting a path gives me:
```
# ls -la /usr/lib/gstreamer-1.0/
total 1468
drwxr-xr-x   2 root root    4096 Oct  1 22:10 .
drwxr-xr-x 116 root root   12288 Oct  1 22:10 ..
-rwxr-xr-x   1 root root    1157 Oct  1 22:10 libgsticamerasrc.la
-rwxr-xr-x   1 root root 1478968 Oct  1 22:10 libgsticamerasrc.so

# export GST_PLUGIN_PATH=/usr/lib/gstreamer-1.0
# gst-launch-1.0 icamerasrc ! autovideosink
[10-02 16:40:14.453] CamHAL[ERR] load_camera_hal_library, failed to open PCI device. error: (null)
[10-02 16:40:14.453] CamHAL[ERR] get_number_of_cameras, function call is nullptr
[10-02 16:40:14.453] CamHAL[ERR] get_number_of_cameras, function call is nullptr

(gst-plugin-scanner:11228): GLib-GObject-CRITICAL **: 16:40:14.453: g_param_spec_enum: assertion 'g_enum_get_value (enum_class, default_value) != NULL' failed

(gst-plugin-scanner:11228): GLib-GObject-CRITICAL **: 16:40:14.453: validate_pspec_to_install: assertion 'G_IS_PARAM_SPEC (pspec)' failed

(gst-plugin-scanner:11228): GLib-GObject-CRITICAL **: 16:40:14.453: g_param_spec_ref_sink: assertion 'G_IS_PARAM_SPEC (pspec)' failed

(gst-plugin-scanner:11228): GLib-GObject-CRITICAL **: 16:40:14.454: g_param_spec_unref: assertion 'G_IS_PARAM_SPEC (pspec)' failed
[10-02 16:40:14.472] CamHAL[ERR] load_camera_hal_library, failed to open PCI device. error: (null)
[10-02 16:40:14.472] CamHAL[ERR] get_number_of_cameras, function call is nullptr
[10-02 16:40:14.472] CamHAL[ERR] get_number_of_cameras, function call is nullptr

(gst-launch-1.0:11227): GLib-GObject-CRITICAL **: 16:40:14.472: g_param_spec_enum: assertion 'g_enum_get_value (enum_class, default_value) != NULL' failed

(gst-launch-1.0:11227): GLib-GObject-CRITICAL **: 16:40:14.472: validate_pspec_to_install: assertion 'G_IS_PARAM_SPEC (pspec)' failed

(gst-launch-1.0:11227): GLib-GObject-CRITICAL **: 16:40:14.472: g_param_spec_ref_sink: assertion 'G_IS_PARAM_SPEC (pspec)' failed

(gst-launch-1.0:11227): GLib-GObject-CRITICAL **: 16:40:14.472: g_param_spec_unref: assertion 'G_IS_PARAM_SPEC (pspec)' failed
[10-02 16:40:14.472] CamHAL[ERR] get_number_of_cameras, function call is nullptr
WARNING: erroneous pipeline: could not link camerasrc0 to autovideosink0
```

alesrebec, 2024-10-03:
For running that command it's required to have v4l2sink. It takes the video data and outputs it to a device compatible with Video4Linux (V4L2)
Try installing it and maybe adjusting the command to eventually make it work. (maybe different device-name, video format, etc..)