intel/ipu6-drivers #262: Impossible to build on 6.10.3-arch1-2

amvidalrc, 2024-08-11:
Impossible to build on latest Arch stable Kernel 6.10.3-arch1-2 on a Samsung Galaxy Book 4.

Tested:

Aur Package
Manually

Always same error:

`[andromeda@Uranium ipu6-drivers]$  make -j`nproc`
make -C /lib/modules/6.10.3-arch1-2/build M=/home/andromeda/Desarrollo/ipu6-drivers modules
make[1]: Entering directory '/usr/lib/modules/6.10.3-arch1-2/build'
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/i2c/hm11b1.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/i2c/ov01a1s.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/i2c/ov01a10.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/i2c/ov02c10.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/i2c/ov02e10.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/i2c/ov05c10.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/i2c/ov2740.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/i2c/hm2170.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/i2c/hm2172.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/i2c/hi556.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-bus.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-dma.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-mmu.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-buttress.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-trace.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-cpd.o
  CC [M]  /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/ipu6.o
/home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu.c: In function 'request_cpd_fw':
/home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu.c:432:29: error: implicit declaration of function 'vmalloc'; did you mean 'kvmalloc'? [-Wimplicit-function-declaration]
  432 |                 tmp->data = vmalloc(fw->size);
      |                             ^~~~~~~
      |                             kvmalloc
/home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu.c:432:27: error: assignment to 'const u8 *' {aka 'const unsigned char *'} from 'int' makes pointer from integer without a cast [-Wint-conversion]
  432 |                 tmp->data = vmalloc(fw->size);
      |                           ^
make[5]: *** [scripts/Makefile.build:244: /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu.o] Error 1
make[5]: *** Waiting for unfinished jobs....
/home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-mmu.c: In function 'ipu_mmu_alloc':
/home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-mmu.c:564:28: error: implicit declaration of function 'vzalloc'; did you mean 'kzalloc'? [-Wimplicit-function-declaration]
  564 |         mmu_info->l2_pts = vzalloc(ISP_L2PT_PTES * sizeof(*mmu_info->l2_pts));
      |                            ^~~~~~~
      |                            kzalloc
/home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-mmu.c:564:26: error: assignment to 'u32 **' {aka 'unsigned int **'} from 'int' makes pointer from integer without a cast [-Wint-conversion]
  564 |         mmu_info->l2_pts = vzalloc(ISP_L2PT_PTES * sizeof(*mmu_info->l2_pts));
      |                          ^
/home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-mmu.c:583:9: error: implicit declaration of function 'vfree'; did you mean 'kvfree'? [-Wimplicit-function-declaration]
  583 |         vfree(mmu_info->l2_pts);
      |         ^~~~~
      |         kvfree
make[5]: *** [scripts/Makefile.build:244: /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6/../ipu-mmu.o] Error 1
make[4]: *** [scripts/Makefile.build:485: /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel/ipu6] Error 2
make[3]: *** [scripts/Makefile.build:485: /home/andromeda/Desarrollo/ipu6-drivers/drivers/media/pci/intel] Error 2
make[2]: *** [/usr/lib/modules/6.10.3-arch1-2/build/Makefile:1934: /home/andromeda/Desarrollo/ipu6-drivers] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.10.3-arch1-2/build'
make: *** [Makefile:121: all] Error 2`

SkeLLLa, 2024-08-12:
Same on fedora.
Slightly better formatted log attached

[0.0-13.20240226gitfb4c170-for-6.10.3-200.fc40.x86_64.failed.log](https://github.com/user-attachments/files/16580055/0.0-13.20240226gitfb4c170-for-6.10.3-200.fc40.x86_64.failed.log)

amvidalrc, 2024-08-12:
I solved the library problems and solved the pfn library deletion:

on ipu-psys.c

`ret = pin_user_pages(addr, 1, FOLL_WRITE | FOLL_LONGTERM, &page);
if (ret < 0)
    goto error_up_read;

pfn = page_to_pfn(page);  // Usa 'pfn' ya declarado anteriormente
pages[nr] = pfn_to_page(pfn);

unpin_user_pages(&page, 1);`

And i compiled it succesfull and the modules are active, but the camera is not working yet on cheese or gnome camera.