intel/ipu6-drivers #263: Dell Latitude 9330 2-in-1 webcam not recognised with kernel 6.10.5-arch1-1

Xwang1976, 2024-08-18:
The webcam is not recognised on my Dell Latitude 9330 2-in-1 laptop.

These are the details of my pc configuration:

Operating System: Arch Linux 
KDE Plasma Version: 6.1.4
KDE Frameworks Version: 6.5.0
Qt Version: 6.7.2
Kernel Version: 6.10.5-arch1-1 (64-bit)
Graphics Platform: Wayland
Processors: 12 × 12th Gen Intel® Core™ i5-1240U
Memory: 15.3 GiB of RAM
Graphics Processor: Mesa Intel® Graphics
Manufacturer: Dell Inc.
Product Name: Latitude 9330

These are the outputs of some command I tried to try to understand where the issue is from (without success):
sudo dmesg | grep ipu6
```
[    3.544961] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    3.545361] intel-ipu6 0000:00:05.0: IPU6 in non-secure mode touch 0x0 mask 0xff
[    3.552847] intel-ipu6 0000:00:05.0: FW version: 20230925
[    3.558108] intel-ipu6 0000:00:05.0: IPU6-v3[465d] hardware version 5
```
sudo LIBCAMERA_LOG_LEVELS=0 cam --list
```
[0:03:52.542661954] [3163] DEBUG IPAModule ipa_module.cpp:334 ipa_ipu3.so: IPA module /usr/lib/libcamera/ipa_ipu3.so is signed
[0:03:52.542694759] [3163] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/libcamera/ipa_ipu3.so'
[0:03:52.542711514] [3163] DEBUG IPAModule ipa_module.cpp:334 ipa_vimc.so: IPA module /usr/lib/libcamera/ipa_vimc.so is signed
[0:03:52.542718489] [3163] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/libcamera/ipa_vimc.so'
[0:03:52.542730024] [3163] ERROR IPAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
[0:03:52.542732644] [3163] ERROR IPAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info
[0:03:52.542745989] [3163]  INFO Camera camera_manager.cpp:313 libcamera v0.3.1
[0:03:52.542971665] [3164] DEBUG Camera camera_manager.cpp:69 Starting camera manager
[0:03:52.549879824] [3164] DEBUG DeviceEnumerator device_enumerator.cpp:230 New media device "intel-ipu6" created from /dev/media0
[0:03:52.551146448] [3164] DEBUG DeviceEnumerator device_enumerator.cpp:258 Added device /dev/media0: intel-ipu6
[0:03:52.551306888] [3164] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'vimc'
[0:03:52.551319683] [3164] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'uvcvideo'
[0:03:52.551322816] [3164] DEBUG Camera camera_manager.cpp:134 Found registered pipeline handler 'ipu3'
Available cameras:
```
pacman -Ql linux-firmware | grep ipu6
```
linux-firmware /usr/lib/firmware/intel/ipu/ipu6_fw.bin.zst
linux-firmware /usr/lib/firmware/intel/ipu/ipu6ep_fw.bin.zst
linux-firmware /usr/lib/firmware/intel/ipu/ipu6epadln_fw.bin.zst
linux-firmware /usr/lib/firmware/intel/ipu/ipu6epmtl_fw.bin.zst
linux-firmware /usr/lib/firmware/intel/ipu/ipu6se_fw.bin.zst
```
v4l2-ctl --all
```
Driver Info:
        Driver name      : isys
        Card type        : ipu6
        Bus info         : 
        Driver version   : 6.10.5
        Capabilities     : 0xa4a00001
                Video Capture
                Metadata Capture
                I/O MC
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x24a00001
                Video Capture
                Metadata Capture
                I/O MC
                Streaming
                Extended Pix Format
Priority: 2
Video input : 0 (Intel IPU6 ISYS Capture 0: ok)
Format Video Capture:
        Width/Height      : 1920/1080
        Pixel Format      : 'BG12' (12-bit Bayer BGBG/GRGR)
        Field             : None
        Bytes per Line    : 3840
        Size Image        : 4151040
        Colorspace        : Raw
        Transfer Function : Default (maps to None)
        YCbCr/HSV Encoding: Default (maps to ITU-R 601)
        Quantization      : Default (maps to Full Range)
        Flags             : 
Format Metadata Capture:
        Sample Format   : 'MET8' (8-bit Generic Metadata)
        Buffer Size     : 9600
```
sudo cam -l
```
[0:07:22.249654633] [6769] ERROR IPAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
[0:07:22.249665907] [6769] ERROR IPAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info
[0:07:22.249678756] [6769]  INFO Camera camera_manager.cpp:313 libcamera v0.3.1
Available cameras: 
```

Cyborgscode, 2024-09-13:
Device: TerraPad 1262v2
OS: Fedora 40

Mor eor less the same issues here, a different cam resolution, but the same result: it creates a /dev/media0 device, but no cams are detected.

Cyborgscode, 2024-09-13:
[   42.743875] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.HEC.TSR1], AE_NOT_FOUND (20240322/psargs-330)
[   42.743895] ACPI Error: Aborting method \_SB.PC00.LPCB.H_EC.SEN1._TMP due to previous error (AE_NOT_FOUND) (20240322/psparse-529)
[   52.474606] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.HEC.TSR1], AE_NOT_FOUND (20240322/psargs-330)
[   52.474622] ACPI Error: Aborting method \_SB.PC00.LPCB.H_EC.SEN1._TMP due to previous error (AE_NOT_FOUND) (20240322/psparse-529)
[   67.318562] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.HEC.TSR1], AE_NOT_FOUND (20240322/psargs-330)
[   67.318589] ACPI Error: Aborting method \_SB.PC00.LPCB.H_EC.SEN1._TMP due to previous error (AE_NOT_FOUND) (20240322/psparse-529)
[   89.333934] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.HEC.TSR1], AE_NOT_FOUND (20240322/psargs-330)
[   89.333958] ACPI Error: Aborting method \_SB.PC00.LPCB.H_EC.SEN1._TMP due to previous error (AE_NOT_FOUND) (20240322/psparse-529)
[  124.148854] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.HEC.TSR1], AE_NOT_FOUND (20240322/psargs-330)
[  124.148872] ACPI Error: Aborting method \_SB.PC00.LPCB.H_EC.SEN1._TMP due to previous error (AE_NOT_FOUND) (20240322/psparse-529)
[  173.300552] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.HEC.TSR1], AE_NOT_FOUND (20240322/psargs-330)
[  173.300585] ACPI Error: Aborting method \_SB.PC00.LPCB.H_EC.SEN1._TMP due to previous error (AE_NOT_FOUND) (20240322/psparse-529)
[  178.937214] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[  178.937581] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x0 mask 0x0
[  178.938141] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[  178.938668] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[  178.939140] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[  178.939660] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[  178.940134] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[  178.940654] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[  178.941136] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[  178.941659] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[  178.942131] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[  178.942653] intel-ipu6 0000:00:05.0: Unexpected CSR 0x0
[  178.986108] intel-ipu6 0000:00:05.0: FW version: 20230925
[  178.987274] intel-ipu6 0000:00:05.0: Found supported sensor INT3474:01
[  178.987763] intel-ipu6 0000:00:05.0: Found supported sensor INT3474:02
[  178.987973] intel-ipu6 0000:00:05.0: Connected 2 cameras
[  178.989561] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[  178.999153] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[  179.068307] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[  179.068357] intel-ipu6 0000:00:05.0: IPU6-v3[465d] hardware version 5
[  179.128451] ov2740 i2c-INT3474:02: error -EREMOTEIO: failed to find sensor
[  179.129799] ov2740 i2c-INT3474:02: probe with driver ov2740 failed with error -121
[  179.194393] ov2740 i2c-INT3474:01: error -EREMOTEIO: failed to find sensor
[  179.196197] ov2740 i2c-INT3474:01: probe with driver ov2740 failed with error -121
[  247.025895] ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PC00.LPCB.HEC.TSR1], AE_NOT_FOUND (20240322/psargs-330)
[  247.025929] ACPI Error: Aborting method \_SB.PC00.LPCB.H_EC.SEN1._TMP due to previous error (AE_NOT_FOUND) (20240322/psparse-529)

Cyborgscode, 2024-09-13:
it also creates 32 video devices:

crw-rw----+ 1 root video 81,  0 14. Sep 00:30 /dev/video0
crw-rw----+ 1 root video 81,  1 14. Sep 00:30 /dev/video1
crw-rw----+ 1 root video 81, 10 14. Sep 00:30 /dev/video10
crw-rw----+ 1 root video 81, 11 14. Sep 00:30 /dev/video11
crw-rw----+ 1 root video 81, 12 14. Sep 00:30 /dev/video12
crw-rw----+ 1 root video 81, 13 14. Sep 00:30 /dev/video13
crw-rw----+ 1 root video 81, 14 14. Sep 00:30 /dev/video14
crw-rw----+ 1 root video 81, 15 14. Sep 00:30 /dev/video15
crw-rw----+ 1 root video 81, 16 14. Sep 00:30 /dev/video16
crw-rw----+ 1 root video 81, 17 14. Sep 00:30 /dev/video17
crw-rw----+ 1 root video 81, 18 14. Sep 00:30 /dev/video18
crw-rw----+ 1 root video 81, 19 14. Sep 00:30 /dev/video19
crw-rw----+ 1 root video 81,  2 14. Sep 00:30 /dev/video2
crw-rw----+ 1 root video 81, 20 14. Sep 00:30 /dev/video20
crw-rw----+ 1 root video 81, 21 14. Sep 00:30 /dev/video21
crw-rw----+ 1 root video 81, 22 14. Sep 00:30 /dev/video22
crw-rw----+ 1 root video 81, 23 14. Sep 00:30 /dev/video23
crw-rw----+ 1 root video 81, 24 14. Sep 00:30 /dev/video24
crw-rw----+ 1 root video 81, 25 14. Sep 00:30 /dev/video25
crw-rw----+ 1 root video 81, 26 14. Sep 00:30 /dev/video26
crw-rw----+ 1 root video 81, 27 14. Sep 00:30 /dev/video27
crw-rw----+ 1 root video 81, 28 14. Sep 00:30 /dev/video28
crw-rw----+ 1 root video 81, 29 14. Sep 00:30 /dev/video29
crw-rw----+ 1 root video 81,  3 14. Sep 00:30 /dev/video3
crw-rw----+ 1 root video 81, 30 14. Sep 00:30 /dev/video30
crw-rw----+ 1 root video 81, 31 14. Sep 00:30 /dev/video31
crw-rw----+ 1 root video 81,  4 14. Sep 00:30 /dev/video4
crw-rw----+ 1 root video 81,  5 14. Sep 00:30 /dev/video5
crw-rw----+ 1 root video 81,  6 14. Sep 00:30 /dev/video6
crw-rw----+ 1 root video 81,  7 14. Sep 00:30 /dev/video7
crw-rw----+ 1 root video 81,  8 14. Sep 00:30 /dev/video8
crw-rw----+ 1 root video 81,  9 14. Sep 00:30 /dev/video9