intel/ipu6-drivers #265: Dell Precision 5490 camera not working on Arch 6.10.9

Gerico0889, 2024-09-11:
The webcam is being recognized correctly but it does not work. It seems that the drivers are correctly installed. The webcam cannot be found with `lspci` or `lsusb`. The weird thing is that there seem to be a lot of `/dev/video*/ folders and none of this output seem to work. I'll leave some info down here but happy to provide more information if needed:

PC Specs:
```
❯ fastfetch --logo none
    コンピューター
┌──────────────────────────────────────────┐
   OS : EndeavourOS x86_64
   Kernel : Linux 6.10.9-arch1-2
   Packages : 1270 (pacman)
   Display : 2560x1600 @ 60 Hz (as 1535x959) in 14″ [Built-in]
   WM : Hyprland (Wayland)
   Terminal : kitty 0.36.2
└──────────────────────────────────────────┘
┌──────────────────────────────────────────┐
   CPU : Intel(R) Core(TM) Ultra 7 165H
   GPU : RTX 2000 Ada Generation Laptop GPU
   GPU : Intel Arc Graphics
   GPU Driver : nvidia (proprietary) 560.35.03
   GPU Driver : i915
  ﬙ Memory : 3.18 GiB / 30.80 GiB (10%)
  󱦟 OS Age  : 1 days
  󱫐 Uptime  : 16 mins
└──────────────────────────────────────────┘
```
```
❯ lspci
0000:00:00.0 Host bridge: Intel Corporation Device 7d01 (rev 04)
0000:00:01.0 PCI bridge: Intel Corporation Device 7ecc (rev 10)
0000:00:02.0 VGA compatible controller: Intel Corporation Meteor Lake-P [Intel Arc Graphics] (rev 08)
0000:00:04.0 Signal processing controller: Intel Corporation Meteor Lake-P Dynamic Tuning Technology (rev 04)
0000:00:05.0 Multimedia controller: Intel Corporation Meteor Lake IPU (rev 04)
0000:00:06.0 PCI bridge: Intel Corporation Device 7e4d (rev 20)
0000:00:07.0 PCI bridge: Intel Corporation Meteor Lake-P Thunderbolt 4 PCI Express Root Port #0 (rev 10)
0000:00:07.1 PCI bridge: Intel Corporation Meteor Lake-P Thunderbolt 4 PCI Express Root Port #1 (rev 10)
0000:00:07.2 PCI bridge: Intel Corporation Meteor Lake-P Thunderbolt 4 PCI Express Root Port #2 (rev 10)
0000:00:07.3 PCI bridge: Intel Corporation Meteor Lake-P Thunderbolt 4 PCI Express Root Port #3 (rev 10)
0000:00:08.0 System peripheral: Intel Corporation Meteor Lake-P Gaussian & Neural-Network Accelerator (rev 20)
0000:00:0a.0 Signal processing controller: Intel Corporation Meteor Lake-P Platform Monitoring Technology (rev 01)
0000:00:0b.0 Processing accelerators: Intel Corporation Meteor Lake NPU (rev 04)
0000:00:0d.0 USB controller: Intel Corporation Meteor Lake-P Thunderbolt 4 USB Controller (rev 10)
0000:00:0d.2 USB controller: Intel Corporation Meteor Lake-P Thunderbolt 4 NHI #0 (rev 10)
0000:00:0d.3 USB controller: Intel Corporation Meteor Lake-P Thunderbolt 4 NHI #1 (rev 10)
0000:00:0e.0 RAID bus controller: Intel Corporation Volume Management Device NVMe RAID Controller Intel Corporation
0000:00:12.0 Serial controller: Intel Corporation Meteor Lake-P Integrated Sensor Hub (rev 20)
0000:00:14.0 USB controller: Intel Corporation Meteor Lake-P USB 3.2 Gen 2x1 xHCI Host Controller (rev 20)
0000:00:14.2 RAM memory: Intel Corporation Device 7e7f (rev 20)
0000:00:15.0 Serial bus controller: Intel Corporation Meteor Lake-P Serial IO I2C Controller #0 (rev 20)
0000:00:15.3 Serial bus controller: Intel Corporation Meteor Lake-P Serial IO I2C Controller #3 (rev 20)
0000:00:16.0 Communication controller: Intel Corporation Meteor Lake-P CSME HECI #1 (rev 20)
0000:00:1c.0 PCI bridge: Intel Corporation Device 7e3f (rev 20)
0000:00:1f.0 ISA bridge: Intel Corporation Device 7e02 (rev 20)
0000:00:1f.3 Multimedia audio controller: Intel Corporation Meteor Lake-P HD Audio Controller (rev 20)
0000:00:1f.4 SMBus: Intel Corporation Meteor Lake-P SMBus Controller (rev 20)
0000:00:1f.5 Serial bus controller: Intel Corporation Meteor Lake-P SPI Controller (rev 20)
0000:01:00.0 3D controller: NVIDIA Corporation AD107GLM [RTX 2000 Ada Generation Laptop GPU] (rev a1)
0000:02:00.0 Network controller: Intel Corporation Wi-Fi 7(802.11be) AX1775*/AX1790*/BE20*/BE401/BE1750* 2x2 (rev 1a)
0000:ab:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5260 PCI Express Card Reader (rev 01)
10000:e0:06.0 System peripheral: Intel Corporation RST VMD Managed Controller
10000:e0:06.1 PCI bridge: Intel Corporation Device 7eca (rev 10)
10000:e1:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller S4LV008[Pascal]
```
```
❯ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 8086:0b63 Intel Corp. USB Bridge
Bus 003 Device 003: ID 27c6:634c Shenzhen Goodix Technology Co.,Ltd. Goodix Fingerprint USB Device
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
[... 125 lines cut ...]
	/dev/video27
	/dev/video28
	/dev/video29
	/dev/video30
	/dev/video31
	/dev/video32
	/dev/video33
	/dev/video34
	/dev/video35
	/dev/video36
	/dev/video37
	/dev/video38
	/dev/video39
	/dev/video40
	/dev/video41
	/dev/video42
	/dev/video43
	/dev/video44
	/dev/video45
	/dev/video46
	/dev/video47

ipu6 (PCI:0000:00:05.0):
	/dev/media0
```

iiztp, 2024-09-18:
Same problem here, on the same laptop though I have Debian Unstable isntead of Endeavour

Gerico0889, 2024-09-19:
Just to keep the Issue up to date. I also updated to 6.10.10-arch1-2 but still have the exact same problem.

iiztp, 2024-09-19:
> Just to keep the Issue up to date. I also updated to 6.10.10-arch1-2 but still have the exact same problem.

I thought you were also on 6.10.10, I must have misread, I am on same Kernel.
FYI, I tried recompiling kernel of 6.11 because I saw that it might work from this [link](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1074441)
But can't seem to succeed to build it, it says "missing IPU6 drivers" or something like that...

Cyborgscode, 2024-09-22:
BTW:  the real issue is here: the i2c system can't talk to the chip.

https://github.com/intel/ipu6-drivers/issues/252

Gerico0889, 2024-09-22:
Mmm I'm not sure if that is the same problem. As you can see from my dmesg output I do not get that `error -EREMOTEIO: failed to find sensor` error

fmonteghetti, 2024-10-01:
On both 6.10 and 6.11,, with the dell precision 5490, I have this error:
```
vsc-tp spi-INTC10D0:00: probe with driver vsc-tp failed with error -22
```
`vsc` means (Intel) Vision Sensing Controller, so this is likely what prevents the camera from being accessed. I am on the latest firmware 1.6.0.

alexhulbert, 2024-10-27:
I can confirm that the mei_vsc_hw module seems to be the root of the problem for me as well. I have a Dell XPS 9640 and I get the same exact error (`spi-INTC10D0:00` for me too). Theoretically, on the latest kernel we shouldn't even need these out-of-tree drivers. Fedora 41 users are reporting that similar meteor lake devices work out of the box one the spi device is setup and visible from `/sys/bus/i2c/devices/`

@fmonteghetti Did you make any progress with the issue?

fmonteghetti, 2024-10-27:
> @fmonteghetti Did you make any progress with the issue?

Not really. I am waiting for the release of Fedora 41 in november to get an additional data point. (Fedora 41 plans to ship with support for ipu6 camera through an open source stack https://fedoraproject.org/wiki/Changes/IPU6_Camera_support)

randoogle, 2024-12-05:
> > @fmonteghetti Did you make any progress with the issue?
> 
> Not really. I am waiting for the release of Fedora 41 in november to get an additional data point. (Fedora 41 plans to ship with support for ipu6 camera through an open source stack https://fedoraproject.org/wiki/Changes/IPU6_Camera_support)
> 
> Edit: the Dell Precision 5490 webcam works out of the box with Fedora 41. The kernel log contains 'Found supported sensor OVTI01A0:00', which I could not get on arch linux 6.11 kernel.

Odd, I have a new Precision 5490 with a fresh install of Fedora 41, and the webcam isn't working. 
I do see this in my dmesg logs:

> [   24.660997] pci 0000:00:05.0: Found supported sensor OVTI01A0:00
[   24.661063] pci 0000:00:05.0: Connected 1 cameras

But that doesn't seem to translate into a working webcam. Maybe it's something to do with me running Silverblue? I doubt it, but I guess it's possible.

fmonteghetti, 2024-12-05:
> Odd, I have a new Precision 5490 with a fresh install of Fedora 41, and the webcam isn't working. I do see this in my dmesg logs:
>
>
> But that doesn't seem to translate into a working webcam. Maybe [...]

I tested on the regular Fedora 41 Workstation iso. I had the same kernel messages as you: I tested the camera in cheese (aka gnome camera) only.