intel/ipu6-drivers #272: Add support for GC2607 sensor

mark-herbert42, 2024-10-05:
Unfortuantely could not find the suitable driver for it.

The path on laptop :  pci0000:00/0000:00:19.1/i2c_designware.5/i2c-5/i2c-GCTI2607:00
ACPI device GCTI2607:00

Windows ini file - saying it is Intel made driver
[gc2607.txt](https://github.com/user-attachments/files/17266804/gc2607.txt)

bit, 2025-03-23:
Looks like GC2607 is used in the HUAWEI MateBook X Pro 2024

mark-herbert42, 2025-03-24:
Exactly - that's where it is used. Unfortunately it seems to be the only PC product where that sensor is used, the rest is in mobiles. But the camera driver for Huawei/Windows is provided by Intel.

yuyalun-allen, 2025-07-20:
Is there any progress for supporting this sensor in the driver?