intel/ipu6-drivers #282: Lenovo Thinkbook X AI 2024 with IPU6 get kernel panic

peigongdsd, 2024-10-14:
uname -a output:
```
Linux ThinkBookX 6.11.3 #1-NixOS SMP PREEMPT_DYNAMIC Thu Oct 10 10:04:18 UTC 2024 x86_64 GNU/Linux
```
CPU: Core Ultra 9 185H
OS: Nixos 24.11(unstable) 

Kernel panic happenes several times with all of them seems related to ipu6 kernel module. Call stack looks like
```
<4>[    8.888110] Call Trace:
<4>[    8.888115]  <IRQ>
<4>[    8.888119]  ? __die+0x23/0x80
<4>[    8.888138]  ? page_fault_oops+0x173/0x5b0
<4>[    8.888149]  ? search_module_extables+0x19/0x70
<4>[    8.888157]  ? search_bpf_extables+0x5f/0x90
<4>[    8.888167]  ? exc_page_fault+0x155/0x160
<4>[    8.888178]  ? asm_exc_page_fault+0x26/0x30
<4>[    8.888187]  ? complete+0x43/0x80
<4>[    8.888201]  ipu6_buttress_isr+0x186/0x3f0 [intel_ipu6]
<4>[    8.888221]  __handle_irq_event_percpu+0x4a/0x1b0
<4>[    8.888231]  handle_irq_event+0x3e/0x90
<4>[    8.888238]  handle_fasteoi_irq+0x91/0x260
<4>[    8.888246]  __common_interrupt+0x3e/0xc0
<4>[    8.888254]  common_interrupt+0x82/0xa0
<4>[    8.888263]  </IRQ>
<4>[    8.888266]  <TASK>
```

Here are more crashdumps from pstore

[1.txt](https://github.com/user-attachments/files/17367268/1.txt)
[2.txt](https://github.com/user-attachments/files/17367269/2.txt)
[3.txt](https://github.com/user-attachments/files/17367270/3.txt)
[4.txt](https://github.com/user-attachments/files/17367271/4.txt)
[5.txt](https://github.com/user-attachments/files/17367272/5.txt)
[6.txt](https://github.com/user-attachments/files/17367273/6.txt)
[8.txt](https://github.com/user-attachments/files/17367274/8.txt)
[9.txt](https://github.com/user-attachments/files/17367275/9.txt)

peigongdsd, 2024-10-14:
BTW this crash happenes quite randomly, even when the ipu6 device or camera is not accessed. Seems only related to the kernel module.

peigongdsd, 2024-10-15:
Found bug report https://github.com/linux-surface/linux-surface/issues/1516 which might be related.

sgruszka, 2024-11-04:
Hi, you can try this patch:
https://lore.kernel.org/linux-media/20241031102321.409454-1-stanislaw.gruszka@linux.intel.com/

peigongdsd, 2024-11-05:
> Hi, you can try this patch: https://lore.kernel.org/linux-media/20241031102321.409454-1-stanislaw.gruszka@linux.intel.com/

Doesn't seem to crash anymore in 10 hours. I'll keep testing but seems like this is the fix. Thanks a lot!

sgruszka, 2024-11-05:
Thanks for testing. BTW how did you configure pstore? It's supported out of the box on NixOS ?

peigongdsd, 2024-11-05:
I think this is finally fixed. And for the pstroe thing, yes. I believe that this function is provided by systemd-pstore, that is, systemd will automantically transfer your crash dump in UEFI pstore into /var/lib/systemd/pstore. I havn't enabled anything explicitly, maybe nixos set some default options.

closed 2024-11-05 by peigongdsd