intel/ipu6-drivers #288: Dell XPS 9320 camera not found

IvanVojtko, 2024-10-28:
Hello. I have Dell XPS 9320 and my camera stopped working (not sure when). It used to work at least few months ago, but now nothing works
```
System Information
        Manufacturer: Dell Inc.
        Product Name: XPS 9320
        Version: Not Specified
        Serial Number: 113B2T3
        UUID: 4c4c4544-0031-3310-8042-b1c04f325433
        Wake-up Type: Power Switch
        SKU Number: 0AF3
        Family: XPS
```
```
cam -l
[0:20:12.063201636] [3990] ERROR IPAModule ipa_module.cpp:171 Symbol ipaModuleInfo not found
[0:20:12.063223978] [3990] ERROR IPAModule ipa_module.cpp:291 v4l2-compat.so: IPA module has no valid info
[0:20:12.063236186] [3990]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2
Available cameras:
```
```
v4l2-ctl --all
Cannot open device /dev/video0, exiting.
```
```
sudo dmesg | grep "ipu6"                                             
[    3.810676] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    3.810849] intel-ipu6 0000:00:05.0: IPU6 in non-secure mode touch 0x0 mask 0xff
[    3.814705] intel-ipu6 0000:00:05.0: FW version: 20230925
[    3.816781] intel-ipu6 0000:00:05.0: IPU6 in non-secure mode touch 0x0 mask 0xff
[    3.819303] intel-ipu6 0000:00:05.0: FW version: 20230925
[    3.822411] intel-ipu6 0000:00:05.0: IPU6 in non-secure mode touch 0x0 mask 0xff
[    3.826056] intel-ipu6 0000:00:05.0: FW version: 20230925
```
```
ls /lib/firmware/intel/ipu/
ipu6epadln_fw.bin.zst  ipu6ep_fw.bin.zst  ipu6epmtl_fw.bin.zst  ipu6_fw.bin.zst  ipu6se_fw.bin.zst  irci_irci_ecr-master_20161208_0213_20170112_1500.bin.zst  shisp_2400b0_v21.bin.zst  shisp_2401a0_v21.bin.zst
```
If I list i2c devices, only i2c-OVTI01AB:00 is showing, which is presumably IR camera.
```
ls /sys/bus/i2c/devices  
i2c-0  i2c-1  i2c-10  i2c-11  i2c-12  i2c-13  i2c-14  i2c-15  i2c-16  i2c-17  i2c-18  i2c-2  i2c-3  i2c-4  i2c-5  i2c-6  i2c-7  i2c-8  i2c-9  i2c-ELAN900C:00  i2c-OVTI01AB:00  i2c-VEN_04F3:00
```
If I check /sys/devices, I can see also acpi:OVTI01A0:, which should be regular camera.
```
find /sys/devices -name modalias | xargs grep -h acpi | grep -Ev 'PNP|LNX' | sort -u
acpi:ACPI0003:
acpi:ACPI0007:
acpi:ACPI000C:
acpi:ACPI000E:
acpi:INT3472:
acpi:INTC1041:
acpi:INTC1046:
acpi:INTC1055:
acpi:INTC1070:
acpi:INTC1094:
acpi:INTC1095:
acpi:INTC1096:
acpi:INTC1097:
acpi:INTC1098:
acpi:OVTI01A0:
acpi:OVTI01AB:
```
Also this line is in dmesg.
```
[    5.029553] vsc-tp spi-INTC1094:00: probe with driver vsc-tp failed with error -22
```
I'm running 6.11 linux kernel and latest BIOS. Any idea what to check, what can be the issue?

catwell, 2024-10-29:
I have the same issue on a XPS 3915. Only VEN_0488 and OVTI01AB appear in `/sys/bus/i2c/devices`, not OVTI01A0. When I load intel_ipu6 I only have those lines in dmesg:

```
[  886.207888] intel-ipu6 0000:00:05.0: IPU6 in non-secure mode touch 0x0 mask 0xff
[  886.209989] intel-ipu6 0000:00:05.0: FW version: 20230925
```

This error also happens one during boot:

```
[37.360686] vsc-tp spi-INTC1094:00: probe with driver vsc-tp failed with error -22
```

androw, 2024-11-01:
Same problem here with XPS 9315

jat255, 2024-11-01:
I seem to have the same camera IDs with a Dell Precision 5490 on 6.11.5

androw, 2024-11-03:
> @catwell @androw Give [this](https://github.com/intel/ipu6-drivers/issues/228#issuecomment-2453317943) a try, just ensure you have the linux-headers-xxx package for your kernel version already installed.

Header are installed on my side and I'm running Arch with kernel:
`Linux 6.11.6-zen1-1.1-zen #1 ZEN SMP PREEMPT_DYNAMIC Sat, 02 Nov 2024 20:51:49 +0000 x86_64 GNU/Linux`

odontomachus, 2024-11-18:
Same problem with Dell precision 5680, can't get the camera to work.

sysmso, 2024-11-18:
Same problem with Dell precision 5480...

androw, 2024-11-19:
I tried to contact Kernel developer here: https://lore.kernel.org/linux-media/DM8PR11MB565358EF5A7F96CD1A364C5F99202@DM8PR11MB5653.namprd11.prod.outlook.com/T/#t

They recommend to contact vendor (Dell). No luck to get to the good support level. Atm, they recommend installing the provided Ubuntu 20.04

Seems to also be linked to #291

twouters, 2024-11-19:
> I tried to contact Kernel developer here: https://lore.kernel.org/linux-media/DM8PR11MB565358EF5A7F96CD1A364C5F99202@DM8PR11MB5653.namprd11.prod.outlook.com/T/#t
>
> They recommend to contact vendor [...]

I don't know why they are blaming this on the device being in non-secure mode. It works in non-secure mode as I reported in #291 even though it is very unreliable (I've only seen the camera work on two out of twenty reboots - with the upstream kernel drivers, not the driver from this repository.)

androw, 2024-11-19:
I do not really know what this secure or not mode imply.
Maybe you can try to add that point on the kernel mailing list?
I tried the patch provided in #291 but it didn't work on my side.

twouters, 2024-11-19:
> I do not really know what this secure or not mode imply. Maybe you can try to add that point on the kernel mailing list? I tried the patch provided in #291 but it didn't work on my side.

Yeah, I was messing around with some patches and when the camera worked I thought I had found a fix but I believe it was a one-off success.

markedmunds, 2024-11-20:
The same problem applies to XPS 16 (9640) with kernel 6.11.9 and sensor OVTI02C1. Hans De Geode from RedHat has also been working on this or a similar issue: https://hansdegoede.dreamwidth.org/28841.html