intel/ipu6-drivers #291: Deferred probe pending: intel-ipu6: IPU6 bridge init failed

bnicolae, 2024-11-04:
The IPU6 kernel driver does not properly initialize the sensor. I am running the following recent kernel on a Dell XPS 14 9440:
6.11.6-arch1-1 #1 SMP PREEMPT_DYNAMIC

```
[    5.073429] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    5.073895] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.087818] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.093275] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.099067] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.106581] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.109721] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.112224] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.114954] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.118321] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.121034] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.130421] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.133523] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.141270] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.144274] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.150025] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.152758] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.166290] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.169562] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.172759] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.175878] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.223814] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.226571] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.250669] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.253306] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.254506] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.257363] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.262313] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.265965] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.271877] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.275943] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.325626] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.330049] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.330508] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.334360] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.334666] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.337339] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.342627] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.345604] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.345916] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.348643] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.359526] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.362934] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.364088] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.367566] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.375836] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.379069] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.384590] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.388002] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.389829] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.393326] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.394925] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.397991] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.398760] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.402016] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.426353] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.429870] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.430845] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.436087] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.437074] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.443561] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.488426] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.491322] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.507740] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[    5.514108] intel-ipu6 0000:00:05.0: FW version: 20230925
[    5.562299] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[... 48 lines cut ...]
[   10.806938] intel-ipu6 0000:00:05.0: FW version: 20230925
[   11.050902] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[   11.064712] intel-ipu6 0000:00:05.0: FW version: 20230925
[   11.123009] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[   11.140002] intel-ipu6 0000:00:05.0: FW version: 20230925
[   11.183208] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[   11.186472] intel-ipu6 0000:00:05.0: FW version: 20230925
[   11.186952] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[   11.190272] intel-ipu6 0000:00:05.0: FW version: 20230925
[   11.360072] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[   11.374693] intel-ipu6 0000:00:05.0: FW version: 20230925
[   11.378212] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[   11.381087] intel-ipu6 0000:00:05.0: FW version: 20230925
[   11.381559] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[   11.384373] intel-ipu6 0000:00:05.0: FW version: 20230925
[   11.557586] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[   11.571662] intel-ipu6 0000:00:05.0: FW version: 20230925
[   16.260498] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[   16.275988] intel-ipu6 0000:00:05.0: FW version: 20230925
[   16.276793] pci 0000:00:05.0: deferred probe pending: intel-ipu6: IPU6 bridge init failed
[  187.128318] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[  187.143001] intel-ipu6 0000:00:05.0: FW version: 20230925
[  187.146021] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[  187.149356] intel-ipu6 0000:00:05.0: FW version: 20230925
```

sendevent, 2024-11-06:
The same thing is on the Dell XPS 16 9640 (OV02C10) running `6.12.0-rc2-2-MANJARO`

jsenecal, 2024-11-10:
Same thing on a Dell Latitude 5430 with 6.11.6-arch1-1 (on Archlinux as well)

Camera chipset is seen as OVTI01AB, if that helps.

filiurskyi, 2024-11-15:
Dell XPS 13 9340, can confirm similar issue on arch. Kernel `6.11.7-arch1-1`

twouters, 2024-11-18:
> After patching the Arch kernel (6.11.7-arch1-1) with the following patch, i'm getting new results on XPS 9315: https://lore.kernel.org/all/20241031102321.409454-1-stanislaw.gruszka@linux.intel.com/
> 
> ```
> Nov 18 10:29:33 kernel: intel-ipu6 0000:00:05.0: FW version: 20230925
> Nov 18 10:29:33 kernel: intel-ipu6 0000:00:05.0: IPU6 in non-secure mode touch 0x0 mask 0xff
> Nov 18 10:29:33 kernel: intel-ipu6 0000:00:05.0: FW version: 20230925
> Nov 18 10:29:33 kernel: intel-ipu6 0000:00:05.0: IPU6 in non-secure mode touch 0x0 mask 0xff
> Nov 18 10:29:33 kernel: intel-ipu6 0000:00:05.0: FW version: 20230925
> Nov 18 10:29:33 kernel: intel-ipu6 0000:00:05.0: Found supported sensor OVTI01A0:00
> Nov 18 10:29:33 kernel: intel-ipu6 0000:00:05.0: Connected 1 cameras
> Nov 18 10:29:33 kernel: intel-ipu6 0000:00:05.0: IPU6-v3[465d] hardware version 5
> ```
> 
> @jwrdegoede fyi
> 
> (Edit: I'm even able to get an image with qcam, but probably need to do something about `/dev/udmabuf` permissions)

I tried to reproduce by rebooting and rebuilding the kernel with only mentioned patch (I did have a few others applied, but none of them resulted in the camera becoming available) but each boot since earlier today resulted in an undetected camera, even the kernel I booted from where the camera did work with earlier no longer detects the camera.

So I don't know what caused it to succeed this one time :disappointed:

twouters, 2024-11-18:
One difference I noticed from the boot where the camera worked, but it doesn't tell me much

```
Nov 18 10:29:22 kernel: vsc-tp spi-INTC1094:00: Probing device: IRQ=183
```

vs those when it doesn't work

```
Nov 18 15:55:20 kernel: vsc-tp spi-INTC1094:00: Probing device: IRQ=-517
```

```
diff --git a/drivers/misc/mei/vsc-tp.c b/drivers/misc/mei/vsc-tp.c
index 1618cca9a731..fa1d846d27fe 100644
--- a/drivers/misc/mei/vsc-tp.c
+++ b/drivers/misc/mei/vsc-tp.c
@@ -484,6 +484,7 @@ static int vsc_tp_probe(struct spi_device *spi)
                .id = PLATFORM_DEVID_NONE,
        };
        struct device *dev = &spi->dev;
+       dev_info(dev, "Probing device: IRQ=%d\n", spi->irq);
        struct platform_device *pdev;
        struct acpi_device *adev;
        int ret;
```

sgruszka, 2024-11-21:
> One difference I noticed from the boot where the camera worked, but it doesn't tell me much
> Nov 18 15:55:20 kernel: vsc-tp spi-INTC1094:00: Probing device: IRQ=-517
> ```
> 
> ```
> diff --git a/drivers/misc/mei/vsc-tp.c b/drivers/misc/mei/vsc-tp.c
> index 1618cca9a731..fa1d846d27fe 100644
> --- a/drivers/misc/mei/vsc-tp.c
> +++ b/drivers/misc/mei/vsc-tp.c
> @@ -484,6 +484,7 @@ static int vsc_tp_probe(struct spi_device *spi)
>                 .id = PLATFORM_DEVID_NONE,
>         };
>         struct device *dev = &spi->dev;
> +       dev_info(dev, "Probing device: IRQ=%d\n", spi->irq);

Hi , for this probing issue I attached proposed fix 
https://bugzilla.redhat.com/show_bug.cgi?id=2324683#c11

bnicolae, 2024-12-09:
Seems like I am hitting on a related issue, also mentioned here: https://bugzilla.redhat.com/show_bug.cgi?id=2324683#c13

```
[    7.483604] intel_vsc intel_vsc: hw_reset failed ret = -110
[    8.390076] intel_vsc intel_vsc: hw_reset failed ret = -110
[    9.300415] intel_vsc intel_vsc: hw_reset failed ret = -110
[    9.300701] intel_vsc intel_vsc: reset: reached maximal consecutive resets: disabling the device
[    9.301068] intel_vsc intel_vsc: reset failed ret = -19
[    9.301357] intel_vsc intel_vsc: link layer initialization failed.
[    9.301653] intel_vsc intel_vsc: error -ENODEV: init hw failed
```

Is there any news about this one?

cpence, 2025-02-03:
I think I have a similar issue on Arch 6.12.10-arch1-1, though I don't even get the "bridge init failed" message. Just the loop of:

```
[ 1097.799421] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x0 mask 0x0
[ 1097.822634] intel-ipu6 0000:00:05.0: FW version: 20230925
[ 1097.844334] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x0 mask 0x0
[ 1097.849527] intel-ipu6 0000:00:05.0: FW version: 20230925
```

ad infinitum (122 times since I booted up 21 minutes ago).

bnicolae, 2025-02-03:
The issue is still present in 6.13.1-arch1-1

sgruszka, 2025-02-18:
Hi, 
 Hans posted fix for the "intel_vsc intel_vsc: hw_reset failed ret = -110" issue. 
https://lore.kernel.org/lkml/20250214212425.84021-1-hdegoede@redhat.com/T/#r75250e63c3065475e4c83a55bf3d325a662369b2

bnicolae, 2025-03-04:
I can confirm the fix for "intel_vsc intel_vsc: hw_reset failed ret = -110" works on 6.13.5. Now my camera is attached correctly:
```
[   16.728659] intel-ipu6 0000:00:05.0: Connected 1 cameras
[   16.729842] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[   16.766321] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[   16.816461] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[   16.816478] intel-ipu6 0000:00:05.0: IPU6-v4[7d19] hardware version 6
```

However, the issue itself ("IPU6 bridge init failed" and a long list of retries) still remains and probably should be fixed. Therefore, I will keep this issue open for now.

closed 2025-05-29 by bnicolae