intel/ipu6-drivers #294: Latitude 7320 detachable on Ubuntu 24.10

thekswenson, 2024-11-10:
Hello folks.  Is there any reason why these drivers should not work for the Dell Latitude 7320 detachable?

If it is not hopeless, I would like to post my issue here. Using the PPA I have a camera that seems to be recognized by the driver, but the video using, e.g. `ffplay /dev/video0`, is only black.
```
=> sudo lshw -C multimedia
  *-multimedia:0            
       description: Multimedia controller
       product: Intel Corporation
       vendor: Intel Corporation
       physical id: 5
       bus info: pci@0000:00:05.0
       version: 01
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi pm bus_master cap_list
       configuration: driver=intel-ipu6 latency=0
       resources: iomemory:600-5ff irq:169 memory:6052000000-6052ffffff
```

thekswenson, 2024-11-18:
Even a message such as, "this will take a significant amount of work on the driver" would be helpful.  Basically, if it probably won't take a huge effort, I'm willing to dig into this a bit on my own.

Tiger862000, 2024-11-23:
Hi @thekswenson, I'm in the same Situation as you. With Ubuntu it is not only the camera that's not working, also the Pen is not working properly. For you?

I'm in contact with developer from Fedora who implemented the IPU6 drivers and modules for several platforms. Theoretically it should just work, sadly it's not. https://fedoraproject.org/wiki/Changes/IPU6_Camera_support
I'll keep you posted on the Fedora support.

Have you checked the Ubuntu image for the non-detachable aka Laptop? Because that one is certified by Ubuntu to work.
I'll also test that and report back.

Regards

thekswenson, 2024-11-26:
Hi @Tiger862000 , it's nice to hear from you.

I bought my machine in 2021, so it's possible that we have slightly different hardware. My pen works well, although something _did_ change with the buttons when I upgraded to Ubuntu 24.04. There is an [Ubuntu Bug](https://bugs.launchpad.net/ubuntu/+source/gnome-control-center/+bug/2083102), that links to an upstream bug related to configuring the pen in gnome-control-center.

In short, I don't know why your pen isn't working, and I'm no expert.

In 2021, I [submitted information](https://github.com/linuxwacom/wacom-hid-descriptors/issues/174#issue-1035004205) so that they could add the HID descriptor for my pen.  [It looks like](https://github.com/linuxwacom/wacom-hid-descriptors/issues/366) something may have changed for the newer models. But if you read the [libwacom overview](https://github.com/linuxwacom/libwacom?tab=readme-ov-file#overview), you'll see that the HID descriptor and [tablet definition](https://github.com/linuxwacom/libwacom/issues/432) may not have anything to do with the pen actually working, as it looks like [libinput](https://wayland.freedesktop.org/libinput/doc/latest/tablet-support.html) may be doing the actual work.

When you say "Ubuntu image for the non-detachable", I actually can't find this on the internet?  Do you have this?
Any help you could offer would be greatly appreciated!

danielw01, 2025-02-07:
adding this to libwacom.stylus got pen working for me on fedora 41 

[0x211]
# Dell     ; VID_NONE     | 0x0000 | BAT_SWAP
Name=Dell 7320 Pen
Group=isdv4-aes
Buttons=2
EraserType=Button
Axes=Pressure
Type=Mobile