intel/ipu6-drivers #298: Dell XPS 13 9340 HID webcam sensor is missing

filiurskyi, 2024-11-15:
Missing mapping for sensor HIMX1092  for ( `root/drivers/media/pci/intel/ipu-bridge.c`):

```
ls /sys/bus/i2c/devices
i2c-0  i2c-1  i2c-10  i2c-11  i2c-12  i2c-13  i2c-14  i2c-15  i2c-16  i2c-17  i2c-18  i2c-2  i2c-3  i2c-4  i2c-5  i2c-6  i2c-7  i2c-8  i2c-9  i2c-HIMX1092:00  i2c-VEN_04F3:00
```

filiurskyi, 2024-11-15:
Hi and thank you for a fast reply!
dmesg:
```
[    0.000000] Linux version 6.11.8-arch1-2 (linux@archlinux) (gcc (GCC) 14.2.1 20240910, GNU ld (GNU Binutils) 2.43.0) #1 SMP PREEMPT_DYNAMIC Fri, 15 Nov 2024 15:35:07 +0000
[    0.000000] Command line: initrd=\initramfs-linux.img cryptdevice=PARTUUID=57ad1b44-adf0-4e8d-8491-7cc4f5c71c80:root root=/dev/mapper/root zswap.enabled=0 rw rootfstype=ext4
[    0.000000] x86/split lock detection: #AC: crashing the kernel on kernel split_locks and warning on user-space split_locks
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000004d514fff] usable
[    0.000000] BIOS-e820: [mem 0x000000004d515000-0x0000000051510fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000051511000-0x0000000051d71fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000051d72000-0x0000000051ffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000051fff000-0x0000000051ffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000052000000-0x00000000687fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c0000000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000f9ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000088fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] APIC: Static calls initialized
[    0.000000] efi: EFI v2.7 by Dell
[    0.000000] efi: ACPI=0x51ffe000 ACPI 2.0=0x51ffe014 SMBIOS=0x4dda5000 TPMFinalLog=0x51c78000 ESRT=0x4dd35718 MEMATTR=0x46709018 RNG=0x51f36f18 INITRD=0x466cc218 TPMEventLog=0x51f20018 
[    0.000000] random: crng init done
[    0.000000] efi: Remove mem96: MMIO range=[0xc0000000-0xcfffffff] (256MB) from e820 map
[    0.000000] e820: remove [mem 0xc0000000-0xcfffffff] reserved
[    0.000000] efi: Remove mem97: MMIO range=[0xf8000000-0xf9ffffff] (32MB) from e820 map
[    0.000000] e820: remove [mem 0xf8000000-0xf9ffffff] reserved
[    0.000000] efi: Remove mem99: MMIO range=[0xff000000-0xffffffff] (16MB) from e820 map
[    0.000000] e820: remove [mem 0xff000000-0xffffffff] reserved
[    0.000000] SMBIOS 3.6 present.
[    0.000000] DMI: Dell Inc. XPS 13 9340/0CH5X9, BIOS 1.6.0 05/09/2024
[    0.000000] DMI: Memory slots populated: 8/8
[    0.000000] tsc: Detected 3000.000 MHz processor
[    0.000000] tsc: Detected 2995.200 MHz TSC
[    0.000008] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000010] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000014] last_pfn = 0x890000 max_arch_pfn = 0x400000000
[    0.000018] MTRR map: 8 entries (3 fixed + 5 variable; max 23), built from 10 variable MTRRs
[    0.000020] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000464] x2apic: enabled by BIOS, switching to x2apic ops
[    0.000466] last_pfn = 0x52000 max_arch_pfn = 0x400000000
[    0.005098] esrt: Reserving ESRT space from 0x000000004dd35718 to 0x000000004dd357a0.
[    0.005107] Using GB pages for direct mapping
[    0.005301] Secure boot disabled
[    0.005301] RAMDISK: [mem 0x3e42b000-0x3f920fff]
[    0.005388] ACPI: Early table checksum verification disabled
[    0.005391] ACPI: RSDP 0x0000000051FFE014 000024 (v02 DELL  )
[    0.005394] ACPI: XSDT 0x0000000051F29228 00016C (v01 DELL   Dell Inc 00000002      01000013)
[    0.005399] ACPI: FACP 0x0000000051FDD000 000114 (v06 DELL   Dell Inc 00000002      01000013)
[    0.005402] ACPI: DSDT 0x0000000051F50000 0891D3 (v02 DELL   Dell Inc 00000002      01000013)
[    0.005404] ACPI: FACS 0x0000000051CA2000 000040
[    0.005406] ACPI: SSDT 0x0000000051FFC000 00038C (v02 PmaxDv Pmax_Dev 00000001 INTL 20210930)
[    0.005408] ACPI: SSDT 0x0000000051FFB000 000689 (v02 PmRef  Cpu0Ist  00003000 INTL 20210930)
[    0.005410] ACPI: SSDT 0x0000000051FFA000 0005E6 (v02 PmRef  Cpu0Hwp  00003000 INTL 20210930)
[    0.005411] ACPI: SSDT 0x0000000051FF9000 0001AB (v02 PmRef  Cpu0Psd  00003000 INTL 20210930)
[    0.005413] ACPI: SSDT 0x0000000051FF8000 000540 (v02 PmRef  Cpu0Cst  00003001 INTL 20210930)
[    0.005415] ACPI: SSDT 0x0000000051FF6000 001BAF (v02 PmRef  ApIst    00003000 INTL 20210930)
[    0.005417] ACPI: SSDT 0x0000000051FF4000 001620 (v02 PmRef  ApHwp    00003000 INTL 20210930)
[    0.005418] ACPI: SSDT 0x0000000051FF2000 001349 (v02 PmRef  ApPsd    00003000 INTL 20210930)
[    0.005420] ACPI: SSDT 0x0000000051FF1000 000FBB (v02 PmRef  ApCst    00003000 INTL 20210930)
[    0.005422] ACPI: SSDT 0x0000000051FED000 003BC8 (v02 CpuRef CpuSsdt  00003000 INTL 20210930)
[    0.005424] ACPI: DTPR 0x0000000051FEC000 000098 (v01                 00000000      00000000)
[    0.005426] ACPI: SSDT 0x0000000051FEA000 00128C (v02 INTEL  PDatTabl 00001000 INTL 20210930)
[    0.005427] ACPI: SSDT 0x0000000051FE9000 000A54 (v02 INTEL  IgfxSsdt 00003000 INTL 20210930)
[    0.005429] ACPI: SSDT 0x0000000051FDF000 009A89 (v02 INTEL  TcssSsdt 00001000 INTL 20210930)
[    0.005431] ACPI: SSDT 0x0000000051FDE000 0000F9 (v02 INTEL  IpuSsdt  00001000 INTL 20210930)
[    0.005433] ACPI: HPET 0x0000000051FDC000 000038 (v01 DELL   Dell Inc 00000002      01000013)
[... cut ...]
cec                    94208  3 drm_display_helper,xe,i915
i8042                  57344  1 dell_laptop
wmi                    36864  8 dell_wmi_sysman,video,dell_wmi_ddv,dell_wmi,wmi_bmof,dell_smm_hwmon,dell_smbios,dell_wmi_descriptor
serio                  28672  6 serio_raw,atkbd,psmouse,i8042
```

in the meantime i found somewhere that HIMX1092 is related to infrared sensors, not camera itself so i event cant figure out (according to https://hansdegoede.dreamwidth.org/28841.html) what my camera model is...

filiurskyi, 2024-11-15:
Hi and thank you for a fast reply!

In the meantime i figured out that HIMX1092 is related to infrared sensors, not camera itself

lsmod:
```
Module                  Size  Used by
udp_diag               12288  0
tcp_diag               12288  0
inet_diag              28672  2 tcp_diag,udp_diag
ccm                    20480  9
snd_seq_dummy          12288  0
snd_hrtimer            12288  1
rfcomm                102400  4
snd_seq               131072  7 snd_seq_dummy
snd_seq_device         16384  1 snd_seq
snd_soc_sof_sdw       114688  4
snd_soc_intel_hda_dsp_common    16384  1 snd_soc_sof_sdw
dell_pc                12288  0
snd_sof_probes         28672  0
snd_ctl_led            24576  0
platform_profile       12288  1 dell_pc
cmac                   12288  2
algif_hash             12288  1
algif_skcipher         12288  1
af_alg                 32768  6 algif_hash,algif_skcipher
dell_laptop            36864  0
bnep                   36864  2
snd_soc_rt715_sdca     49152  1
snd_soc_rt1318_sdw     28672  2
regmap_sdw_mbq         12288  1 snd_soc_rt715_sdca
regmap_sdw             12288  2 snd_soc_rt715_sdca,snd_soc_rt1318_sdw
snd_hda_codec_hdmi     98304  1
snd_soc_dmic           12288  0
joydev                 28672  0
snd_sof_pci_intel_mtl    40960  0
snd_sof_intel_hda_generic    40960  1 snd_sof_pci_intel_mtl
soundwire_intel        81920  4 snd_sof_intel_hda_generic
soundwire_cadence      45056  1 soundwire_intel
snd_sof_intel_hda_common   200704  2 snd_sof_pci_intel_mtl,snd_sof_intel_hda_generic
snd_sof_intel_hda_mlink    36864  3 soundwire_intel,snd_sof_intel_hda_common,snd_sof_intel_hda_generic
snd_sof_intel_hda      20480  2 snd_sof_intel_hda_common,snd_sof_intel_hda_generic
snd_sof_pci            24576  2 snd_sof_pci_intel_mtl,snd_sof_intel_hda_generic
snd_sof_xtensa_dsp     16384  1 snd_sof_intel_hda_generic
snd_sof               462848  6 snd_sof_pci_intel_mtl,snd_sof_pci,snd_sof_intel_hda_common,snd_sof_intel_hda_generic,snd_sof_probes,snd_sof_intel_hda
snd_sof_utils          16384  1 snd_sof
snd_soc_hdac_hda       28672  1 snd_sof_intel_hda_common
hid_sensor_custom_intel_hinge    12288  0
snd_hda_ext_core       36864  4 snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_sof_intel_hda_mlink,snd_sof_intel_hda
snd_soc_acpi_intel_match   110592  2 snd_sof_pci_intel_mtl,snd_sof_intel_hda_generic
soundwire_generic_allocation    12288  1 soundwire_intel
snd_soc_acpi           16384  2 snd_soc_acpi_intel_match,snd_sof_intel_hda_generic
soundwire_bus         147456  8 snd_soc_sof_sdw,snd_soc_rt715_sdca,regmap_sdw,regmap_sdw_mbq,soundwire_intel,soundwire_generic_allocation,snd_soc_rt1318_sdw,soundwire_cadence
hid_sensor_als         16384  0
hid_sensor_trigger     20480  4 hid_sensor_custom_intel_hinge,hid_sensor_als
vfat                   24576  1
intel_uncore_frequency    12288  0
snd_soc_core          454656  9 snd_soc_sof_sdw,snd_soc_rt715_sdca,soundwire_intel,snd_sof,snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_soc_rt1318_sdw,snd_sof_probes,snd_soc_dmic
industrialio_triggered_buffer    12288  1 hid_sensor_trigger
intel_uncore_frequency_common    16384  1 intel_uncore_frequency
fat                   110592  1 vfat
snd_compress           28672  2 snd_soc_core,snd_sof_probes
kfifo_buf              12288  1 industrialio_triggered_buffer
iwlmvm                749568  0
ac97_bus               12288  1 snd_soc_core
hid_sensor_iio_common    20480  3 hid_sensor_trigger,hid_sensor_custom_intel_hinge,hid_sensor_als
snd_pcm_dmaengine      16384  1 snd_soc_core
x86_pkg_temp_thermal    16384  0
industrialio          143360  5 industrialio_triggered_buffer,hid_sensor_trigger,hid_sensor_custom_intel_hinge,kfifo_buf,hid_sensor_als
hid_sensor_custom      32768  0
[... 1342 lines cut ...]
[ 6862.796183] pci 0000:00:05.0: Setting to D3hot
[ 6862.807128] pci 0000:00:0b.0: Setting to D3hot
[ 6862.819393] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/mtl_guc_70.bin version 70.29.2
[ 6862.827350] i8042 kbd 00:03: activated
[ 6862.827363] i8042 aux 00:04: activated
[ 6862.829453] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[ 6862.829458] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[ 6862.829678] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[ 6862.830184] i915 0000:00:02.0: [drm] GT1: GuC firmware i915/mtl_guc_70.bin version 70.29.2
[ 6862.830186] i915 0000:00:02.0: [drm] GT1: HuC firmware i915/mtl_huc_gsc.bin version 8.5.4
[ 6862.833047] nvme nvme0: D3 entry latency set to 10 seconds
[ 6862.840329] nvme nvme0: 22/0/0 default/read/poll queues
[ 6862.843502] i915 0000:00:02.0: [drm] GT1: GUC: submission enabled
[ 6862.843506] i915 0000:00:02.0: [drm] GT1: GUC: SLPC enabled
[ 6862.843596] i915 0000:00:02.0: [drm] GT1: GUC: RC enabled
[ 6863.165386] OOM killer enabled.
[ 6863.165389] Restarting tasks ... done.
[ 6863.166574] random: crng reseeded on system resumption
[ 6863.175225] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[ 6863.180299] intel-ipu6 0000:00:05.0: FW version: 20230925
[ 6863.181720] mei_gsc_proxy 0000:00:16.0-0f73db04-97ab-4125-b893-e904ad0d5464: bound 0000:00:02.0 (ops i915_gsc_proxy_component_ops [i915])
[ 6863.182047] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[ 6863.186047] intel-ipu6 0000:00:05.0: FW version: 20230925
[ 6863.215528] PM: suspend exit
```

sgruszka, 2024-11-18:
Yes, this require more work. But it this laptop there is ljca usb expander that power up camera sensor, so we do not need new usbio driver and possibly this can be fixed quicker.