intel/ipu6-drivers #299: Latest patches failing on X1 Carbon Gen 11 using latest Ubuntu 24.04 oem kernel package

The9Cat, 2024-11-19:
## Summary

Following the discussion in https://github.com/intel/ipu6-drivers/issues/187#issuecomment-2484864556, this documents my attempt to build and use the patched kernel.  This is at the request of @sgruszka.

### Steps to build the kernel
1. Begin with a standard Ubuntu 24.04 build environment
2. Add "deb-src" to the types line in the `/etc/apt/sources.list.d/ubuntu.sources` file.  Execute `sudo apt update`.
3. Install source and build packages as follows: 
``
sudo apt update
sudo apt build-dep -y linux linux-image-unsigned-6.11.0-1007-oem
sudo apt install -y fakeroot llvm libncurses-dev dwarves
``
I'm using an unsigned schema because I can't build the signed Ubuntu package, obviously...
4. Get the source package: `apt source linux-image-unsigned-6.11.0-1007-oem`
5. `cd` to the kernel source directory and apply the following 4 patches using `patch -p1 < file`.  
[patch1.txt](https://github.com/user-attachments/files/17821262/patch1.txt)
[patch2.txt](https://github.com/user-attachments/files/17821263/patch2.txt)
[patch3.txt](https://github.com/user-attachments/files/17821264/patch3.txt)
[patch4.txt](https://github.com/user-attachments/files/17821265/patch4.txt)
6. Use the debian scripts to prepare the source:
``
chmod a+x debian/rules
chmod a+x debian/scripts/*
chmod a+x debian/scripts/misc/*
fakeroot debian/rules clean
``
7. Edited the source name in the `debian.oem/changelog` file to build `*.deb` packages with a distinguishing version number.  I arbitrarily picked `1000` for this.
8. Build the kernel with the debian scripts.  This takes a while, of course:
``
fakeroot debian/rules clean
fakeroot debian/rules binary
``
9. This gives a bunch of `*.deb` files in the `..` directory.  I installed these with the command: `sudo dpkg -i linux-oem-6.11-headers-6.11.0-1007_6.11.0-1007.1000_all.deb linux-headers-6.11.0-1007-oem_6.11.0-1007.1000_amd64.deb linux-modules-6.11.0-1007-oem_6.11.0-1007.1000_amd64.deb linux-modules-ipu6-6.11.0-1007-oem_6.11.0-1007.1000_amd64.deb linux-image-unsigned-6.11.0-1007-oem_6.11.0-1007.1000_amd64.deb`
10. Finally: reboot
11. I need to turn off secure boot in order to launch the unsigned kernel

### Notes/problems/issue with this build
- If I use the Ubuntu noble-proposed kernel packages for `6.11.0-1007-oem`.  The kernel modules fail to correctly configure the ov2740 and v4l2_loopback some large fraction of the time, perhaps 50-60% of the time.  This can often be recovered by a few `modprobe -r ov2740` + `modprobe ov2740` followed by a `sudo systemctl restart v4l2-relayd.service` to restart the gstreamer.
- While the camera can be used in this state, it is not 100% reliable.  It often requires restarts and resets by removing and inserting modules.
- When using the newly compiled kernel with the patches, the `video4linux` virtual devices seem not to be configured and therefore the gstreamer process fails to start from the service file.
- As mentioned in #187, this may be a consequence of using an unsigned kernel package with signed firmware.  I'm not sure if this is the root cause of this failure or not.

sgruszka, 2024-11-20:
Thanks, I will go through those steps.

sgruszka, 2024-11-24:
Hi @The9Cat . I built kernel/modules packages and installed. 
After reboot  some V4L devices are created, but not sure if configured properly.
```
stasiu@yoga:~$ ls /sys/class/video4linux/
v4l-subdev0  v4l-subdev10  v4l-subdev12  v4l-subdev14  v4l-subdev16  v4l-subdev3  v4l-subdev5  v4l-subdev7  v4l-subdev9  video1  video3  video5  video7
v4l-subdev1  v4l-subdev11  v4l-subdev13  v4l-subdev15  v4l-subdev2   v4l-subdev4  v4l-subdev6  v4l-subdev8  video0       video2  video4  video6
stasiu@yoga:~$ sudo cat /sys/class/video4linux/*/name
Intel IPU6 CSI-2 0
Intel IPU6 CSI2 BE SOC 2
Intel IPU6 CSI2 BE SOC 3
Intel IPU6 CSI2 BE SOC 4
Intel IPU6 CSI2 BE SOC 5
Intel IPU6 CSI2 BE SOC 6
Intel IPU6 CSI2 BE SOC 7
ov2740 18-0036
Intel IPU6 CSI-2 1
Intel IPU6 CSI-2 2
Intel IPU6 CSI-2 3
Intel IPU6 CSI-2 4
Intel IPU6 CSI-2 5
Intel IPU6 CSI-2 6
Intel IPU6 CSI-2 7
Intel IPU6 CSI2 BE SOC 0
Intel IPU6 CSI2 BE SOC 1
Intel IPU6 BE SOC capture 0
Intel IPU6 BE SOC capture 1
Intel IPU6 BE SOC capture 2
Intel IPU6 BE SOC capture 3
Intel IPU6 BE SOC capture 4
Intel IPU6 BE SOC capture 5
Intel IPU6 BE SOC capture 6
Intel IPU6 BE SOC capture 7
```
How can this be tested ? How to install rest of the stack on ubuntu ?

sgruszka, 2024-11-24:
One thing is strange, linux-modules-ipu6-6.11.0-1008-oem package seems to use isys module from intel/ipu6-drivers. So patching in kernel ipu6 isys should have no impact.

The9Cat, 2024-11-24:
Interesting.  You definitely got further than I did!  found *no* v4l devices.   Did you need to add something not described in my list of steps?  Otherwise, I can't understand why it would work for you and not me...something different in the build environment, I guess, but what?

To test, I would do the following:
```
sudo add-apt-repository ppa:oem-solutions-group/intel-ipu6
sudo apt update
sudo apt install gstreamer1.0-icamera libcamhal0 v4l2-relayd v4l-utils
```
That should give you everything you need to try capturing the camera stream with `vlc` or using the [gUM test page](https://mozilla.github.io/webrtc-landing/gum_test.html)

The9Cat, 2024-11-25:
And, I must have misunderstood the situation with isys.  I had assumed that the linux-modules packages was using the kernel drivers, but truth is that I didn't checked...

sgruszka, 2024-11-25:
I had to skip step 7:
> Edited the source name in the debian.oem/changelog file to build *.deb packages with a distinguishing version number. I arbitrarily picked 1000 for this.
Because I got some build error when changed version number in the change log file. Apparently did not make this correctly, but ..

I think it's the issue could be that: with the changed kernel version some external modules are not loaded i.e. v4l2loopback .
You can try to build without version modification and assume that the -1008 is patched and rollback to -1007 or any other older version to use unpatched kernel.

sgruszka, 2024-11-25:
> sudo add-apt-repository ppa:oem-solutions-group/intel-ipu6
> sudo apt update
> sudo apt install gstreamer1.0-icamera libcamhal0 v4l2-relayd v4l-utils
> ```
>
> That should give you everything you need to try [...]

This overwrite custom build oem-1008 installed kernel and I got ov2740 sensor error again when reboot second time.
I checked there was no /sys/bus/usb/devices/3-8\:1.0/ljca_version file which should be present on patched ljca driver.  
When realized that, I just reinstall kernel image and modules:
 dpkg -i linux-image-unsigned-6.11.0-1008-oem_6.11.0-1008.8_amd64.deb linux-modules-6.11.0-1008-oem_6.11.0-1008.8_amd64.deb
 And that make camera works on gUM page. Checked 3 reboots.

sgruszka, 2024-11-25:
One more thing. During the process of above testing, something changed permission or owners of  /dev/video* devices like this:
crw-rw---- 1 root video 81,  0 Nov 25 15:45 /dev/video0
So I could not use the devices with my previous setup. I had to added myself to video group to get the acces again.

The9Cat, 2024-11-25:
Curious.  In Step 7, I changed the version number not the name in the `changelog`.  E.g. the line:
```
linux-oem-6.11 (6.11.0-1007.9) noble; urgency=medium
```
to
```
linux-oem-6.11 (6.11.0-1007.1000) noble; urgency=medium
```
so I don't think that should confuse the dependencies but your thought in https://github.com/intel/ipu6-drivers/issues/299#issuecomment-2497209254 is worth checking.   My understanding is that the debian scripts use that for tagging the modules with the version name.  The kernel name should remain unchanged (the stuff before the final '.' in the changelog entry).   I could try reverting to the the original `changelog` and try again?  Is that the only change in the steps that you recommend?

My build was very clearly a *fail* with respect the the camera.  I was a bit surprised and confused.  Perhaps I didn't investigate hard enough.

The permissions of the `/dev/video*` is the Ubuntu policy.  I agree: you need to your username in the `video` group to get this to work in general.

sgruszka, 2024-11-25:
I think you have right: changing to  6.11.0-1007.1000 should not change the kernel version , which should stay 6.11.0-1007-oem.

Anything obvious from logs what when wrong ?

I recommend install ppe packages and then just replace with custom kernel and modules , like described in https://github.com/intel/ipu6-drivers/issues/299#issuecomment-2497303597

The9Cat, 2024-11-26:
Do you mean behavior of the camera, or byte difference?  The kernel modules do get compiled with the patches, I believe.    The 1007-oem packages in the Ubuntu repo *do* work, albeit with occasional need for module reloads and gstreamer time outs.  The patched ones did not work.  There was nothing obvious in the system log.  The `intel-ipu6` seems to configure itself, finds the FW, sensor, and _connects_ the camera.  But there are *no* isys or psys messages.

The9Cat, 2024-11-26:
## Success

Looks like a reliable ipu6 webcam so far.

## The details
I followed the same steps as in https://github.com/intel/ipu6-drivers/issues/299#issue-2673546368 with two changes:
1. I used `linux-image-unsigned-6.11.0-1008-oem` instead of `linux-image-unsigned-6.11.0-1007-oem`
2. I did not update the version number in the `changelog`, following @sgruszka 

After building and installing the newly minted packages, the ipu6 works as expected!

## Tests
1. I booted 5 times in a row with the working ipu6 camera up and running
2. Checked using the gUM test page, Zoom, and vlc.  I started and stopped each app multiple times, changed camera settings, changed background in Zoom, etc.  This would usually be enough to require a v4l2 reset.  But could not get it to fail!

## Comments
- Still not sure why my first set of packages did not work.  The only difference was renaming the version number, following the standard Debian/Ubuntu method.  I'm not sure it's worth rebuilding just to change the version number, but I am curious.
- I did not have to worry about `/dev/video*` device permissions since my user is already in the video group
- I did not have to worry about the overwrite described in https://github.com/intel/ipu6-drivers/issues/299#issuecomment-2497303597 given that those packages were installed before the kernel upgrade

I'd like to thank @sgruszka for going through my Ubuntu _recipe_ and making suggestions!

The9Cat, 2024-11-26:
Hopefully, these patches will get into mainline soon.  Meanwhile, I'll be checking ipu6 performance.

The9Cat, 2024-11-27:
Also, a note to any Ubuntu user: I recommend putting these packages on hold using `sudo apt-mark hold <packages>` after installation to prevent `apt upgrade` from updating them.

trygvis, 2024-11-27:
Another success story on Debian/sid here. With the patches included in this issue, I got a working video stream from a kernel built on Debian with these commands:

    apt source -t sid linux
    cd linux-6.11.10
    time debian/bin/test-patches ../299-patches/patch*

Where `../299-patches/patch*` are the patches from https://github.com/intel/ipu6-drivers/issues/299#issue-2673546368.

Installation is: `sudo dpkg -i ../*6.11.10*.deb`.

I'm not using v4l2-relayd, and GNOME Snapshot and the GUM pages are working.

The9Cat, 2024-11-27:
@trygvis This is very interesting.  I didn't realize that there was another way here...if you are not using the /dev/video* API along with `v4l2-relayd`, what service is providing the video stream?

The9Cat, 2024-12-02:
@sgruszka Saw your patches in 6.13 (RC).  Have requested that the Ubuntu oem kernel maintainer pull in those patches.  We'll see...

rmasad, 2024-12-02:
I tried compiling the kernel as shown here and it still doesn't work. I got the following error:
```
sudo journalctl -p err -b
dic 02 15:42:06 x1 systemd-modules-load[428]: Failed to find module 'v4l2loopback'
...
```

I tried installing v4l2loopback-dkms and it stopped showing that error but it still doesn't work. I already deleted it.

The9Cat, 2024-12-02:
@rmasad Did you install the modules from Step 9 above?  Specifically,  the `linux-modules-6.11.0-1008-oem` module that you compiled (where 1008 might be another version depending on which source you grabbed) supplies the `v4l2loopback` driver.

Can you remind me which hardware you have?  Is it a Lenovo?  These patches seem to have fixed the issues for me on my Lenovo X1C11 although there may be other issues on other hardware.

Finally, I'm really sorry you are having trouble with this!  I share your frustration and have been there, too.   If you have suggestions on what I can do to improve the instructions above, LMK.

sgruszka, 2024-12-03:
> @sgruszka Saw your patches in 6.13 (RC). Have requested that the Ubuntu oem kernel maintainer pull in those patches. We'll see...

The patches should be  added next week to -stable linux releases as well i.e. 6.12.2, 6.11.11 ... , usually distribution update to latest stable minor release.

rmasad, 2024-12-05:
I recompiled with the instructions (but with linux-image-unsigned-6.11.0-1008-oem and without changing the version number, step 7) and I no longer have the error with v4l2loopback but the camera still doesn't work (it doesn't even turn on the light).

I have a Lenovo X1 Gen12

The9Cat, 2024-12-05:
@rmasad This is very strange.  I've been finding this to be very robust since the patches.  Were there any relevant error messages in `syslog`?  What are the results of:

1. `sudo dmesg |& grep ipu6`
2. `v4l2-ctl --list-devices`
3. Maybe look at `media-ctl -p` as well

rmasad, 2024-12-05:
The camera has not worked for me at all (even though I rebooted several times). Here are the results of the commands:

```
sudo dmesg |& grep ipu6
[    4.054467] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    4.054655] intel-ipu6 0000:00:05.0: Device 0x7d19 (rev: 0x4)
[    4.054679] intel-ipu6 0000:00:05.0: physical base address 0x4059000000
[    4.054680] intel-ipu6 0000:00:05.0: mapped as: 0x00000000d28b4772
[    4.054795] intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.054799] intel-ipu6 0000:00:05.0: IPU secure touch = 0x80000000
[    4.054801] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    4.067760] intel-ipu6 0000:00:05.0: IPC reset done
[    4.073936] intel-ipu6 0000:00:05.0: FW version: 20230925
[    4.075926] intel-ipu6 0000:00:05.0: Found supported sensor OVTI08F4:00
[    4.075977] intel-ipu6 0000:00:05.0: Connected 1 cameras
[    4.080590] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[    4.117751] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[    4.165815] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[    4.165854] intel-ipu6 0000:00:05.0: IPU6-v4 driver version 1.0
[    4.466236] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[    4.466403] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
```

```
v4l2-ctl --list-devices
ipu6 (PCI:pci:pci0000:00):
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/video6
	/dev/video7
	/dev/video8

ipu6 (pci:pci0000:00):
	/dev/media0

Intel MIPI Camera (platform:v4l2loopback-000):
	/dev/video0
```

```
media-ctl -p
Media controller API version 6.11.0

Media device information
------------------------
driver          intel-ipu6-isys
model           ipu6
serial          
bus info        pci:pci0000:00
hw revision     0x0
driver version  6.11.0

Device topology
- entity 1: Intel IPU6 CSI-2 0 (2 pads, 8 links, 0 routes)
            type V4L2 subdev subtype Unknown flags 0
            device node name /dev/v4l-subdev0
	pad0: Sink
		[stream:0 fmt:Y8_1X8/4096x3072 field:none]
	pad1: Source
		[stream:0 fmt:Y8_1X8/4096x3072 field:none]
		-> "Intel IPU6 CSI2 BE SOC 0":0 []
		-> "Intel IPU6 CSI2 BE SOC 1":0 []
		-> "Intel IPU6 CSI2 BE SOC 2":0 []
		-> "Intel IPU6 CSI2 BE SOC 3":0 []
		-> "Intel IPU6 CSI2 BE SOC 4":0 []
		-> "Intel IPU6 CSI2 BE SOC 5":0 []
		-> "Intel IPU6 CSI2 BE SOC 6":0 []
[... 306 lines cut ...]

```
sudo systemctl status v4l2-relayd.service

● v4l2-relayd.service - v4l2-relay daemon service
     Loaded: loaded (/usr/lib/systemd/system/v4l2-relayd.service; enabled; preset: enabled)
     Active: active (running) since Thu 2024-12-05 17:44:20 -03; 8min ago
   Main PID: 1589 (v4l2-relayd)
      Tasks: 5 (limit: 76201)
     Memory: 408.7M (peak: 409.5M)
        CPU: 1.990s
     CGroup: /system.slice/v4l2-relayd.service
             └─1589 /usr/bin/v4l2-relayd -i "icamerasrc buffer-count=7" -o "appsrc name=appsrc caps=video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! videoconvert ! v4l2s>

dic 05 17:50:14 x1 sh[1589]: [12-05 17:50:14.841] CamHAL[ERR] invalid media format, default value used.
dic 05 17:50:14 x1 sh[1589]: [12-05 17:50:14.841] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
dic 05 17:50:14 x1 sh[1589]: [12-05 17:50:14.841] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
dic 05 17:50:14 x1 sh[1589]: [12-05 17:50:14.841] CamHAL[ERR] invalid media format, default value used.
dic 05 17:50:14 x1 sh[1589]: [12-05 17:50:14.841] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
dic 05 17:50:14 x1 sh[1589]: [12-05 17:50:14.841] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
dic 05 17:50:14 x1 sh[1589]: [12-05 17:50:14.841] CamHAL[ERR] invalid media format, default value used.
dic 05 17:50:14 x1 sh[1589]: [12-05 17:50:14.841] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
dic 05 17:50:14 x1 sh[1589]: [12-05 17:50:14.841] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
dic 05 17:50:14 x1 sh[1589]: [12-05 17:50:14.842] CamHAL[ERR] invalid media format, default value used.
```

The9Cat, 2024-12-05:
Hm.   To me, it looks like the sensor isn't detected.   I don't see the `isys bind` message.   I recall from previous posts that Device 0x7d19 should be handled by the `ov02c10` module.  Can you check if that's loaded?    E.g. what is `lsmod | grep ov`.  If it is loaded, maybe try cycling it with a `modprobe -r` and a `modprobe`?

Honestly, I'm not 100% sure what's up with this.

sgruszka, 2024-12-06:
@rmasad please provide full output of dmesg command as txt file. What is your laptop/system ?

rmasad, 2024-12-06:
ov02c10 is not loaded by default. Doing sudo modprobe ov02c10 (and removing it with sudo modprobe -r ov02c10 and loading it again a few times) does not work.

Attached the log of sudo dmesg: [dmesg.log](https://github.com/user-attachments/files/18041946/dmesg.log)

sgruszka, 2024-12-06:
```
[   17.617886] int3472-discrete INT3472:0b: cannot find GPIO chip INTC1007:00, deferring
[   17.619049] int3472-discrete INT3472:0c: cannot find GPIO chip INTC1007:00, deferring
[   17.619261] platform INT3472:0b: deferred probe pending: int3472-discrete: Failed to get GPIO
[   17.619265] platform INT3472:0c: deferred probe pending: int3472-discrete: Failed to get GPIO
```
Seems USB GPIO expander is missing . It's either ljca or usbio . Ljca modules should be already installed,  usbio is provided external linux-modules-usbio*.deb package.

rmasad, 2024-12-06:
I finally got it working. I had intel-usbio-dkms installed (I deleted it) and installed the module that you get by compiling the kernel with these patches.

I hope these patches will be released soon.

Thanks @The9Cat and @crisidev

The9Cat, 2024-12-07:
@sgruszka Thanks!  Yes, that makes sense.