intel/ipu6-drivers #303: Linux kernel 6.12.1 (regression): VIDEO_INTEL_IPU6 depends on ARCH_HAS_DMA_OPS

nagmat84, 2024-11-27:
I reported [issue 219534 in the Linux kernel bug tracker](https://bugzilla.kernel.org/show_bug.cgi?id=219534) and was told to find the actual maintainer. (I hope here is the right place.)

Since kernel 6.12.1 the configuration option `VIDEO_INTEL_IPU6` depends on `ARCH_HAS_DMA_OPS`. (It did not in kernel 6.11.*.) The option `ARCH_HAS_DMA_OPS` is only implicitly activated by enabling GART_IOMMU or XEN. IMHO, this appears like a bug (at least it is odd.) For a kernel which is tailored to only support Intel CPUs, `GART_IOMMU` is not available either, because it only is meaningful for AMD CPUs. (You need to set `CPU_SUP_AMD` for `GART_IOMMU`). I also do not intend to run the kernel as a XEN guest and do not need XEN guest support in the kernel.

Expected behavior: `VIDEO_INTEL_IPU6` should be selectable without `CPU_SUP_AMD` or `XEN`.

sgruszka, 2024-11-28:
Looks like issue introduced by:

commit de6c85bf918ea52d5c680f0d130b37ee2ff152d6
Author: Christoph Hellwig <hch@lst.de>
Date:   Wed Aug 28 09:02:47 2024 +0300

    dma-mapping: clearly mark DMA ops as an architecture feature
    
and later (for 6.13) fixed by 3 patches - the last one changed the dependecy:
 
 commit c8e9120c2065868d97e9e94bceee777e5db08c3e
Author: Bingbu Cao <bingbu.cao@intel.com>
Date:   Wed Oct 16 15:53:03 2024 +0800

    media: ipu6: remove architecture DMA ops dependency in Kconfig

Not really sure what to do for 6.12 ...

sgruszka, 2024-11-29:
Perhaps the best would be just backport ipu6 dma related changes to 6.12 , but not sure. 

@nagmat84 please write email to Christoph,  Bingbu, Sakari Ailus and linux-media@vger.kernel.org regarding issue.

bingbucao, 2024-12-02:
> Looks like issue introduced by:
> 
> commit de6c85bf918ea52d5c680f0d130b37ee2ff152d6 Author: Christoph Hellwig [hch@lst.de](mailto:hch@lst.de) Date: Wed Aug 28 09:02:47 2024 +0300
> 
> ```
> dma-mapping: clearly mark DMA ops as an architecture feature
> ```
> 
> and later (for 6.13) fixed by 3 patches - the last one changed the dependecy:
> 
> commit c8e9120c2065868d97e9e94bceee777e5db08c3e Author: Bingbu Cao [bingbu.cao@intel.com](mailto:bingbu.cao@intel.com) Date: Wed Oct 16 15:53:03 2024 +0800
> 
> ```
> media: ipu6: remove architecture DMA ops dependency in Kconfig
> ```
> 
> Not really sure what to do for 6.12 ...

Yes, if you want to use IPU6 on 6.12.
1. Backport patches from 6.13 to 6.12
2. Add some hack in Makefile to make IPU6 not depends on ARCH_HAS_DMA_OPS

nagmat84, 2024-12-02:
> Yes, if you want to use IPU6 on 6.12.
> 
>     1. Backport patches from 6.13 to 6.12

Anything I can do? I know how to clone the repo, pick the relevant commits and back them (locally). But what needs to be done to get them into the mainline tree for the next 6.12.z patch release? I have never done kernel programming myself.

sgruszka, 2024-12-16:
This is fixed in 6.12.5