intel/ipu6-drivers #306: intel-ipu6: CSE authenticate_run failed + error -EIO: FW authentication failed

nagmat84, 2024-11-30:
I get the following error on kernel 6.12.1 with a custom kernel config
```
intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
Loading firmware: intel/ipu/ipu6epmtl_fw.bin
intel-ipu6 0000:00:05.0: FW version: 20230925
intel-ipu6 0000:00:05.0: Found supported sensor OVTI08F4:00
intel-ipu6 0000:00:05.0: Connected 1 cameras
intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
intel-ipu6 0000:00:05.0: expected resp: 0x2, IPC response: 0xd20 
intel-ipu6 0000:00:05.0: CSE authenticate_run failed
intel-ipu6 0000:00:05.0: error -EIO: FW authentication failed
intel-ipu6 0000:00:05.0: probe with driver intel-ipu6 failed with error -5
```

Expected behaviour: AUTHENTICATE_RUN should succeed.

A necessary kernel option might be missing in the custom kernel configuration, because FW loading works with the distribution-provided kernel. However, if this was indeed the reason, that would indicate a issue with Kconfig for the IPU6 driver as Kconfig should ensure that the kernel includes all necessary options.

Pastebins:

 - [kernel-6.12.1-gentoo-dist, working](https://pastebin.com/q1Tnzw69)
 - [kernel-6.12.1-custom, CSE authenticate_run fails](https://pastebin.com/rmKdRR6q)

manwegit, 2024-12-03:
Do you have SecureBoot enabled?
I had to add intel keys: https://dgpu-docs.intel.com/driver/configuring-secure-boot.html

nagmat84, 2024-12-03:
Yes, secure boot is enabled. I replaced the platform key (PK) with my own key, re-signed the Microsoft KEKs 2011 and 2023, Canonical KEK with my own PK and imported the DB keys from Microsoft (as I need dual boot with Windows) and Canonical as well as my own DB key to sign my custom Linux kernel.

Where do I get the Intel keys?

nagmat84, 2024-12-03:
I added the Intel keys to the UEFI firmware. It didn't help (as expected).

For my custom-built kernel, the CSE still fails to authenticate the firmware. It works with a self-compiled kernel based on the distribution-provided kernel configuration.

bingbucao, 2024-12-04:
Which UEFI firmware version are you using? It looks like the CSE firmware does not match the signed IPU firmware.

nagmat84, 2024-12-04:
- UEFI: LENOVO 21KCCTO1WW/21KCCTO1WW, BIOS N3YET71W (1.36 ) 09/04/2024
 - IPU6 FW: 20230925

But please keep in mind that FW loading and authentication works with a distribution kernel, but not with the self-configured kernel *on the same machine using the same firmware.* So it shouldn't be a hardware or firmware problem.

bingbucao, 2024-12-04:
> * UEFI: LENOVO 21KCCTO1WW/21KCCTO1WW, BIOS N3YET71W (1.36 ) 09/04/2024
> * IPU6 FW: 20230925
>
> But please keep in mind that FW loading and authentication works with a distribution kernel, but not with [...]

Are you sure using same IPU firmware binary with the distribution kernel? I see that the distribution kernel is using a built-in firmware binary.

```
CONFIG_EXTRA_FIRMWARE="intel/ibt-0180-0041.ddc intel/ibt-0180-0041.sfi i915/mtl_guc_70.bin i915/mtl_huc_gsc.bin i915/mtl_gsc_1.bin i915/mtl_dmc.bin intel/ipu/ipu6epadln_fw.bin intel/ipu/ipu6epmtl_fw.bin intel/ipu/ipu6ep_fw.bin intel/vpu/vpu_37xx_v0.0.bin intel/sof-ipc4/mtl/sof-mtl.ri intel-ucode/06-aa-01 intel-ucode/06-aa-02 intel-ucode/06-aa-04 iwlwifi-ma-b0-gf-a0-89.ucode iwlwifi-ma-b0-gf-a0.pnvm regulatory.db regulatory.db.p7s rtl_nic/rtl8153b-2.fw intel/sof-ace-tplg/sof-hda-generic-2ch.tplg"
CONFIG_EXTRA_FIRMWARE_DIR="/lib/firmware"
```

nagmat84, 2024-12-04:
> Are you sure using same IPU firmware binary with the distribution kernel? I see that the distribution kernel is using a built-in firmware binary.

Yes, I am. Once again: *Both* kernels are self-compiled. The kernels are built on the target machine (and both kernels include the same, identical firmware file from `/lib/firmware`. The only difference between the kernels is that one kernel is based on the distribution-provided configuration, while the other kernel uses a customized, stripped-down configuration to save compile time.

Quite obviously, I disabled too much, i.e. left out a necessary kernel option or module.

nagmat84, 2024-12-09:
I have experimented a bit and at least was able to achieve some different behavior. (But not good though.)

## Summary (TLDR)

There seems at least to be three issues here, but I do not know how they are related or if they are independent:

1. **Issue 1:** Successful CSE authentication for IPU6 requires at least `INTEL_MEI_GSC` to be enabled and maybe also `INTEL_MEI_VSC_HW` and `INTEL_MEI_VSC`, but the dependency for IPU6 does not enforce that.
2. **Issue 2:** CSE authentication only works, if the IPU6 driver, the i915 driver and the options from issue 1 are built as modules. If the drivers are statically linked into the kernel, then CSE authentication still fails.
3. **Isssue 3 (for kernel version 6.12.2):** If the drivers and options above are built as modules, CSE authentication finally succeeds, but the kernel shows a DMA bug and kernel trace after that. 

Links to Pastebins:

 - Kernel 6.12.1
    - Config without `INTEL_MEI_GSC` and friends; CSE authentication fails: [https://pastebin.com/rmKdRR6q](https://pastebin.com/rmKdRR6q)
    - Config with `INTEL_MEI_GSC=m` and friends as modules; CSE authentication succeeds:  [https://pastebin.com/q1Tnzw69](https://pastebin.com/q1Tnzw69)
 - Kernel 6.12.2
    - With `INTEL_MEI_GSC=y` and friends linked in statically; CSE authentication fails.
       - Config: [https://pastebin.com/MrZUU7q1](https://pastebin.com/MrZUU7q1)
       - Dmesg: [https://pastebin.com/3rsZZNX7](https://pastebin.com/3rsZZNX7)
    - With `INTEL_MEI_GSC=y` and friends as modules; CSE authentication succeeds, but followed by DMA error
       - Config: [https://pastebin.com/dMPY1YC9](https://pastebin.com/dMPY1YC9)
       - Dmesg: [https://pastebin.com/k3rTNkTV](https://pastebin.com/k3rTNkTV)

_Note:_ All kernels have been self-compiled on the target machine using the same tool chain. If firmware is embedded into the kernel, this firmware is identical to the firmware which is dynamically loaded at run-time from `/lib/firmware`.

## Details

### _Issue 1:_ Successful CSE authentication for IPU6 requires at least `INTEL_MEI_GSC`, `CONFIG_INTEL_MEI_VSC_HW` and `CONFIG_INTEL_MEI_VSC`

It seems that indeed some kernel options were missing. The relevant options are `CONFIG_INTEL_MEI_GSC`, `CONFIG_INTEL_MEI_VSC_HW` and `CONFIG_INTEL_MEI_VSC`. The help texts for those options are

>
> ```
> CONFIG_INTEL_MEI_GSC:
>
> Intel auxiliary driver for GSC devices embedded in Intel graphics devices.
> An MEI device here called GSC can be embedded in an Intel graphics devices,
> to support a range of chassis tasks such as graphics card firmware update
> and security tasks.
> 
> 
> CONFIG_INTEL_MEI_VSC_HW:
>
> Intel SPI transport driver between host and Intel visual sensing controller
> (IVSC) device. This driver can also be built as a module. If so, the module
> will be called mei-vsc-hw.
> 
> 
> CONFIG_INTEL_MEI_VSC:
>
> Intel MEI over SPI driver for Intel visual sensing controller (IVSC) device
> embedded in IA platform. It supports camera sharing between IVSC for
> context sensing and IPU for typical media usage. Select this config should
> enable transport layer for IVSC device. This driver can also be built as a
> module. If so, the module will be called mei-vsc.
> ```
>

I believe the crucial points here are "firmware update", "security task" and "IPU". My [original trimmed-down configuration for 6.12.1](https://pastebin.com/rmKdRR6q) disabled those option completely. The [distribution-provided default configuration for 6.12.1](https://pastebin.com/q1Tnzw69) enables those options as modules.

It seems we have the following dependency chain here:

 - FW authentication for the IPU6 depends on `INTEL_MEI_GSC` (GSC stands for graphics security controller,  believe), and
 - `INTEL_MEI_GSC` depends on the `DRM_I915` in turn.

However, the first dependency is not reflected by the kernel options.

Maybe one solution could be to update Kconfig such that `VIDEO_INTEL_IPU6` depends on `INTEL_MEI_GSC`? I don't know how the other two options (`CONFIG_INTEL_MEI_VSC_HW` and `CONFIG_INTEL_MEI_VSC`) come into play.

### _Issue 2:_ Successful CSE authentication only works with dynamic module loading not with statically linked-in drivers
[... 63 lines cut ...]
 bus_add_driver+0xe3/0x1c0
 driver_register+0x6d/0xc0
 __auxiliary_driver_register+0x69/0xd0
 ? __pfx_isys_driver_init+0x10/0x10 [intel_ipu6_isys]
 do_one_initcall+0x46/0x1c0
 ? __kmalloc_cache_noprof+0x164/0x1f0
 do_init_module+0x5b/0x1f0
 init_module_from_file+0x81/0xc0
 idempotent_init_module+0x103/0x300
 __x64_sys_finit_module+0x57/0x90
 do_syscall_64+0x4b/0x110
 entry_SYSCALL_64_after_hwframe+0x71/0x79
RIP: 0033:0x7f5aa732b89d
Code: ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 5b a5 0c 00 f7 d8 64 89 01 48
RSP: 002b:00007ffda0d3d4a8 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
RAX: ffffffffffffffda RBX: 00005575b6dda570 RCX: 00007f5aa732b89d
RDX: 0000000000000000 RSI: 00007f5aa7163379 RDI: 000000000000003e
RBP: 0000000000000000 R08: 0000000000000000 R09: 00005575b6ce8a40
R10: 00007f5aa73f6ac0 R11: 0000000000000246 R12: 00007f5aa7163379
R13: 0000000000020000 R14: 00005575b6dc6bc0 R15: 00005575b6dbd6e0
 </TASK>
---[ end trace 0000000000000000 ]---
```

Given the observation that this seems somehow be related to DMA, this also might be a result of the [commits 6ac269ab](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=linux-6.12.y&id=6ac269abab9ca5ae910deb2d3ca54351c3467e99) and [11b0543e](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=linux-6.12.y&id=11b0543efe54357b94dccb88dac2099c59358cd4) which have been recently back-ported into 6.12.2 to fix a DMA issue. Hence, this issue might be unrelated to the other two. I haven't had the time to test it with 6.12.1.

bingbucao, 2025-04-07:
Do you have these 2 patches?
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v6.12.21&id=d38610a978d68cbc381bbbafb7e6945de24fb188

https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v6.12.21&id=1496ec94bd38bdb25ca13b1dd4f8e7a6176ea89d