intel/ipu6-drivers #309: HP Spectre360x IPU6 Camera Linux Mint 22: Invalid media configuration

avidoni, 2024-12-15:
as a new linux user and first time mint installer (cinnamon 22), i ran into some problems trying to get the cursed webcamera of my HP Spectre 360x to work. While trying to install the intel-ipu6 drivers to get the HP Truevision Camera (Sensor HI556) to work, I get an error which I could not find any information about anywhere.

I installed the [complete IPU6 driver setup from github](https://github.com/intel/ipu6-drivers), including the camera bins, camera hal, icamerasrc.

First of all: for 'sudo dmesg | grep -i ipu6', i get:

> [    2.649513] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x2)
> [    2.649537] intel-ipu6 0000:00:05.0: physical base address 0x601e000000
> [    2.649539] intel-ipu6 0000:00:05.0: mapped as: 0x000000004483b1ee
> [    2.649658] intel-ipu6 0000:00:05.0: IPU in secure mode
> [    2.649660] intel-ipu6 0000:00:05.0: IPU secure touch = 0x80000000
> [    2.649662] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
> [    2.660888] intel-ipu6 0000:00:05.0: IPC reset done
> [    2.662501] intel-ipu6 0000:00:05.0: FW version: 20230925
> [    2.664496] intel-ipu6 0000:00:05.0: Found supported sensor INT3537:00
> [    2.664680] intel-ipu6 0000:00:05.0: Connected 1 cameras
> [    2.667190] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
> [    2.676127] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
> [    2.745475] intel-ipu6 0000:00:05.0: CSE authenticate_run done
> [    2.745498] intel-ipu6 0000:00:05.0: IPU6-v0 driver version 1.0
> [    2.995548] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
> [    2.995654] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
> [    3.024045] intel-ipu6-isys intel-ipu6-isys0: bind hi556 4-0028 nlanes is 2 port is 2
> [    3.024274] intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.
> [30953.497371] intel-ipu6 0000:00:05.0: IPU in secure mode
> [61513.919732] intel-ipu6 0000:00:05.0: IPU in secure mode

The sensor seems to be found by the intel driver. I have already implemented the config file tweak proposed in [this commit which seemed to have helped a lot of people](https://github.com/intel/ipu6-drivers/commit/3f813580116cc66099fb26e9a7ccf7e8f72424ec#commitcomment-131060757). So far, it does not seem to make a difference due to the following issues:

When i try to start icamerasrc by writing:
`sudo -E gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! ximagesink`
as proposed for the hi556 sensor in the [icamerasrc repo](https://github.com/intel/icamerasrc/tree/icamerasrc_slim_api), I only get to:

> [12-07 22:33:56.890] CamHAL[ERR] Invalid media configuration id 0 for ipu6
> [12-07 22:33:56.890] CamHAL[ERR] Unable to enumerate entities for device /dev/media0
> [12-07 22:33:56.890] CamHAL[ERR] Enum Info failed.

Additionally, upon restart, i also get the following error when starting icamerasrc:
Code: [Select all](https://forums.linuxmint.com/viewtopic.php?t=435614#)

`** (gst-plugin-scanner:3903): CRITICAL **: 22:36:58.046: _dma_fmt_to_dma_drm_fmts: assertion 'fmt != GST_VIDEO_FORMAT_UNKNOWN' failed`
As i cant find these errors anywhere on the internet and also dont find any documentation on this, I would appreciate some help. Thanks alot!

sneumann, 2024-12-20:
Hi, I am on a Dell Latitude 7440, and had a working camera on Ubuntu 24.04 with `6.8.0-10xx-oem` until early December,
and things stopped recently, and current setup with `6.8.0-1019-oem` is not working, but unsure where the regression is. 
Also getting `sh[1538]: [12-20 10:26:00.572] CamHAL[ERR] Invalid media configuration id 0 for ipu6` in logs. 
I'll try to add some more info later. Or should I open a separate issue ? Yours, Steffen

```
dmesg | grep -i ipu6
[    4.326332] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    4.326488] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    4.326523] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    4.326525] intel-ipu6 0000:00:05.0: mapped as: 0x00000000da5f8962
[    4.326592] intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.326595] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.326596] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    4.344262] intel-ipu6 0000:00:05.0: IPC reset done
[    4.349719] intel-ipu6 0000:00:05.0: FW version: 20230925
[    4.354978] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    4.354994] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    4.354996] intel-ipu6 0000:00:05.0: mapped as: 0x00000000da5f8962
[    4.355046] intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.355049] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.355050] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    4.357720] intel-ipu6 0000:00:05.0: IPC reset done
[    4.360917] intel-ipu6 0000:00:05.0: FW version: 20230925
[    4.366442] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    4.366468] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    4.366470] intel-ipu6 0000:00:05.0: mapped as: 0x0000000056bb26fa
[    4.366656] intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.366659] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.366660] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    4.369352] intel-ipu6 0000:00:05.0: IPC reset done
[    4.372505] intel-ipu6 0000:00:05.0: FW version: 20230925
[    4.379113] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    4.379120] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    4.379121] intel-ipu6 0000:00:05.0: mapped as: 0x000000005c67c1f3
[    4.379163] intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.379166] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.379168] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    4.384582] intel-ipu6 0000:00:05.0: IPC reset done
[    4.387046] intel-ipu6 0000:00:05.0: FW version: 20230925
[    4.390058] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    4.390083] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    4.390085] intel-ipu6 0000:00:05.0: mapped as: 0x00000000f4ff8d40
[    4.390222] intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.390225] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.390226] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    4.392095] intel-ipu6 0000:00:05.0: IPC reset done
[    4.395632] intel-ipu6 0000:00:05.0: FW version: 20230925
[    4.401309] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    4.401323] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    4.401324] intel-ipu6 0000:00:05.0: mapped as: 0x0000000056bb26fa
[    4.401382] intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.401384] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.401385] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    4.411557] intel-ipu6 0000:00:05.0: IPC reset done
[    4.414697] intel-ipu6 0000:00:05.0: FW version: 20230925
[    4.431809] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    4.431834] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    4.431836] intel-ipu6 0000:00:05.0: mapped as: 0x0000000056bb26fa
[    4.431898] intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.431901] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.431903] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    4.433730] intel-ipu6 0000:00:05.0: IPC reset done
[    4.437267] intel-ipu6 0000:00:05.0: FW version: 20230925
[    4.485736] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    4.485744] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    4.485745] intel-ipu6 0000:00:05.0: mapped as: 0x000000005c67c1f3
[    4.485920] intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.485937] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.485938] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[... 140 lines cut ...]
[   16.046862] intel-ipu6 0000:00:05.0: mapped as: 0x0000000059934dc9
[   16.046924] intel-ipu6 0000:00:05.0: IPU in secure mode
[   16.046925] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[   16.046926] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[   16.049222] intel-ipu6 0000:00:05.0: IPC reset done
[   16.050499] intel-ipu6 0000:00:05.0: FW version: 20230925
[   17.888715] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[   17.888753] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[   17.888754] intel-ipu6 0000:00:05.0: mapped as: 0x000000001bef8d30
[   17.888890] intel-ipu6 0000:00:05.0: IPU in secure mode
[   17.888892] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[   17.888893] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[   17.907655] intel-ipu6 0000:00:05.0: IPC reset done
[   17.908859] intel-ipu6 0000:00:05.0: FW version: 20230925
[   17.909778] intel-ipu6 0000:00:05.0: Found supported sensor OVTI02C1:00
[... cut ...]
[   17.928199] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[   17.998427] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[   17.998459] intel-ipu6 0000:00:05.0: IPU6-v3 driver version 1.0
[   18.061680] intel-ipu6-isys intel-ipu6-isys0: bind Intel IVSC CSI nlanes is 2 port is 1
[   18.062402] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[   18.062519] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
[   18.114347] intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.
```

jersteth, 2024-12-22:
I can confirm the same. Lenovo Carbon X1 Gen12, ov08x40 sensor on Ubuntu 24.04 was working with 6.8.0-1017-oem on november 25 till early december. Now I have the exact same error message, also with that kernel version.
The ipu6 drivers were updated, as well as intel firmware updates (should have stopped that maybe).

`
sudo dmesg|grep ipu6
[    2.989298] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    2.989573] intel-ipu6 0000:00:05.0: Device 0x7d19 (rev: 0x4)
[    2.989588] intel-ipu6 0000:00:05.0: physical base address 0x4059000000
[    2.989590] intel-ipu6 0000:00:05.0: mapped as: 0x00000000c36770a2
[    2.989660] intel-ipu6 0000:00:05.0: IPU in secure mode
[    2.989662] intel-ipu6 0000:00:05.0: IPU secure touch = 0x80000000
[    2.989663] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    3.002658] intel-ipu6 0000:00:05.0: IPC reset done
[    3.005636] intel-ipu6 0000:00:05.0: FW version: 20230925
[    3.008354] intel-ipu6 0000:00:05.0: Found supported sensor OVTI08F4:00
[    3.008397] intel-ipu6 0000:00:05.0: Connected 1 cameras
[    3.015433] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[    3.055091] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[    3.103071] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[    3.103095] intel-ipu6 0000:00:05.0: IPU6-v4 driver version 1.0
[    3.490765] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[    3.490870] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
[    3.516531] intel-ipu6-isys intel-ipu6-isys0: bind ov08x40 4-0010 nlanes is 4 port is 0
[    3.516870] intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.
`
and
`
gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! ximagesink
[12-22 18:07:46.560] CamHAL[ERR] Invalid media configuration id 0 for ipu6
[12-22 18:07:46.560] CamHAL[ERR] Unable to enumerate entities for device /dev/media0
[12-22 18:07:46.560] CamHAL[ERR] Enum Info failed.
WARNING: erroneous pipeline: could not link camerasrc0 to videoconvert0, camerasrc0 can't handle caps video/x-raw, format=(string)NV12, width=(int)1280, height=(int)720
`

What was upgrade is probably libcamhal0 and libcamhal-ipu6epmtl  and libcamhal-ipu6epmtl-common

dh-harald, 2025-01-16:
I had the same issue with ubuntu 22.04 / 6.8.0-51-generic. I was using https://github.com/jwrdegoede/ipu6-drivers for kernel module. After the update (kernel/camerahal) my camera stops working. I switched back to https://github.com/intel/ipu6-drivers solved my issue