intel/ipu6-drivers #311: Lenovo Thinkbook X AI 2024 with IPU6 cannot wakeup from suspend

peigongdsd, 2024-12-17:
uname -a output:
```
Linux ThinkBookX 6.11.3 #1-NixOS SMP PREEMPT_DYNAMIC Thu Oct 10 10:04:18 UTC 2024 x86_64 GNU/Linux
```
CPU: Core Ultra 9 185H
OS: Nixos 24.11(unstable) 

System cannot wakeup from suspend. Call stack looks like
```
<4>[  833.875513] Call Trace:
<4>[  833.875516]  <TASK>
<4>[  833.875518]  ? __die+0x23/0x80
<4>[  833.875524]  ? page_fault_oops+0x173/0x5b0
<4>[  833.875529]  ? exc_page_fault+0x71/0x160
<4>[  833.875536]  ? asm_exc_page_fault+0x26/0x30
<4>[  833.875544]  ? ipu6_fw_com_open+0xf/0xa0 [intel_ipu6]
<4>[  833.875553]  ipu_fw_psys_open+0x2a/0x100 [intel_ipu6_psys]
<4>[  833.875563]  psys_runtime_pm_resume+0xe8/0x170 [intel_ipu6_psys]
<4>[  833.875569]  ? __pfx_bus_pm_runtime_resume+0x10/0x10 [intel_ipu6]
<4>[  833.875573]  bus_pm_runtime_resume+0x36/0x60 [intel_ipu6]
<4>[  833.875578]  __rpm_callback+0x41/0x170
<4>[  833.875586]  ? __pfx_bus_pm_runtime_resume+0x10/0x10 [intel_ipu6]
<4>[  833.875590]  rpm_callback+0x59/0x70
<4>[  833.875593]  ? __pfx_bus_pm_runtime_resume+0x10/0x10 [intel_ipu6]
<4>[  833.875598]  rpm_resume+0x4d1/0x6d0
<4>[  833.875601]  ? usleep_range_state+0x63/0xa0
<4>[  833.875606]  __pm_runtime_resume+0x4b/0x80
<4>[  833.875609]  ipu6_resume+0x89/0x130 [intel_ipu6]
Oops#1 Part3
<4>[  833.875613]  ? __pfx_pci_pm_resume+0x10/0x10
<4>[  833.875619]  dpm_run_callback+0x8b/0x1d0
<4>[  833.875623]  device_resume+0xb0/0x2b0
<4>[  833.875625]  async_resume+0x1d/0x30
<4>[  833.875627]  async_run_entry_fn+0x31/0x130
<4>[  833.875631]  process_one_work+0x18f/0x3b0
<4>[  833.875636]  worker_thread+0x230/0x340
<4>[  833.875639]  ? __pfx_worker_thread+0x10/0x10
<4>[  833.875643]  kthread+0xcd/0x100
<4>[  833.875646]  ? __pfx_kthread+0x10/0x10
<4>[  833.875647]  ret_from_fork+0x31/0x50
<4>[  833.875650]  ? __pfx_kthread+0x10/0x10
<4>[  833.875652]  ret_from_fork_asm+0x1a/0x30
<4>[  833.875657]  </TASK>

```

Here are more crashdumps from pstore
[dmesg.txt](https://github.com/user-attachments/files/18163455/dmesg.txt)

peigongdsd, 2024-12-17:
Blacklisting intel-ipu6 and intel-ipu6-psys solves the issue.

matejc, 2024-12-17:
Related? https://github.com/NixOS/nixpkgs/issues/364221

peigongdsd, 2024-12-17:
> Related? [NixOS/nixpkgs#364221](https://github.com/NixOS/nixpkgs/issues/364221)

Not quite the same symptom. I think his issue is about i915 instead of ipu6 according to the journals attached.

peigongdsd, 2025-01-31:
I think this bug is not happening anymore now, closing the issue

closed 2025-01-31 by peigongdsd