intel/ipu6-drivers #326: Webcam is not detected on Dell Precision 5490 running AlmaLinux

sm62hi, 2025-02-03:
Hello, my webcam is not detected on my Dell Precision 5490 laptop running Redhat 9 (AlmaLinux). I tried to compile the ipu6 driver without success.
Have you ever tried to make this camera work under Redhat ? how to make this camera work ?
Thanks in advance,
best regards.

bnicolae, 2025-02-03:
It is a mess. It used to work with earlier kernel versions. Then they changed something and as of kernel 6.13 it's not working anymore. Try running "sudo dmesg | grep -i ipu6" and "sudo dmesg | grep -i vsc". Then look for errors. For example:
```
[    6.161921] vsc-tp spi-INTC10D0:00: wait rom failed ret: -110
[    6.162208] intel_vsc intel_vsc: CMD_DUMP_MEM error -110 token 0
[    6.162511] intel_vsc intel_vsc: hw_reset failed ret = -110
[    6.908407] vsc-tp spi-INTC10D0:00: wait rom failed ret: -110
[    6.909089] intel_vsc intel_vsc: CMD_DUMP_MEM error -110 token 0
[    6.909675] intel_vsc intel_vsc: hw_reset failed ret = -110
[    7.663427] vsc-tp spi-INTC10D0:00: wait rom failed ret: -110
[    7.663917] intel_vsc intel_vsc: CMD_DUMP_MEM error -110 token 0
[    7.664423] intel_vsc intel_vsc: hw_reset failed ret = -110
[    7.664925] intel_vsc intel_vsc: reset: reached maximal consecutive resets: disabling the device
[    7.665429] intel_vsc intel_vsc: reset failed ret = -19
[    7.665919] intel_vsc intel_vsc: link layer initialization failed.
[    7.666313] intel_vsc intel_vsc: error -ENODEV: init hw failed
```

sm62hi, 2025-02-04:
Hello, thank you for your reply. I have reinstalled Ubuntu and will be using a podman container for AlmaLinux. I'm sorry but I can't run the greps commands anymore. Thanks a lot. Best regards

gravit8n, 2025-02-22:
Did you see this [thread](https://github.com/intel/ipu6-drivers/issues/228#issuecomment-2675918636) ?