intel/ipu6-drivers #329: Not working in Debian Trixie in a Samsung Galaxy Book4 Pro 360

P3lUZa, 2025-02-17:
My system:
Debian 13 Trixie (new install)
Kernel 6.12.12-amd64
Samsung Galaxy Book4 Pro 360
```
$ sudo dkms status
ipu6-drivers/0.0.0, 6.12.12-amd64, x86_64: installed
usbio-drivers/0.0.1, 6.12.12-amd64, x86_64: installed
```

```
$ sudo dmesg | grep -iE 'ipu6|02c1|int34'
[   21.722323] int3472-discrete INT3472:01: reset \_SB.GPI0 pin number mismatch _DSM 12 resource 108
[   21.722372] int3472-discrete INT3472:01: power-enable \_SB.GPI0 pin number mismatch _DSM 41 resource 201
[   21.722533] int3472-discrete INT3472:01: privacy-led \_SB.GPI0 pin number mismatch _DSM 33 resource 129
[   21.722660] int3472-discrete INT3472:01: clk-enable \_SB.GPI0 pin number mismatch _DSM 36 resource 388
[   21.807000] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[   21.807147] intel-ipu6 0000:00:05.0: IPU6 in secure mode touch 0x80000000 mask 0x0
[   21.813224] intel-ipu6 0000:00:05.0: FW version: 20230925
[   21.814121] intel-ipu6 0000:00:05.0: Found supported sensor OVTI02C1:00
[   21.814201] intel-ipu6 0000:00:05.0: Connected 1 cameras
[   21.815257] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[   21.876423] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[   21.953870] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[   21.953876] intel-ipu6 0000:00:05.0: IPU6-v4[7d19] hardware version 6
```
```
$ sudo -E LANG=C gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! ximagesink
[02-17 21:09:01.845] CamHAL[ERR] readlink sysName /sys/dev/char/0:0 failed ret -1.
[02-17 21:09:01.845] CamHAL[ERR] readlink sysName /sys/dev/char/0:0 failed ret -1.
[02-17 21:09:01.845] CamHAL[ERR] readlink sysName /sys/dev/char/0:0 failed ret -1.
[02-17 21:09:01.845] CamHAL[ERR] readlink sysName /sys/dev/char/0:0 failed ret -1.
[02-17 21:09:01.845] CamHAL[ERR] readlink sysName /sys/dev/char/0:0 failed ret -1.
[02-17 21:09:01.845] CamHAL[ERR] readlink sysName /sys/dev/char/0:0 failed ret -1.
[02-17 21:09:01.882] CamHAL[ERR] readlink sysName /sys/dev/char/0:0 failed ret -1.
[02-17 21:09:01.882] CamHAL[ERR] readlink sysName /sys/dev/char/0:0 failed ret -1.
[02-17 21:09:01.882] CamHAL[ERR] readlink sysName /sys/dev/char/0:0 failed ret -1.
[02-17 21:09:01.882] CamHAL[ERR] readlink sysName /sys/dev/char/0:0 failed ret -1.
[02-17 21:09:01.882] CamHAL[ERR] readlink sysName /sys/dev/char/0:0 failed ret -1.
[02-17 21:09:01.882] CamHAL[ERR] readlink sysName /sys/dev/char/0:0 failed ret -1.
WARNING: erroneous pipeline: could not link camerasrc0 to videoconvert0, camerasrc0 can't handle caps video/x-raw, format=(string)NV12, width=(int)1280, height=(int)720
```
```
$ v4l2-ctl --list-devices
ipu6 ():
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/video6
	/dev/video7
	/dev/video8
	/dev/video9
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video13
	/dev/video14
	/dev/video15
	/dev/video16
	/dev/video17
	/dev/video18
	/dev/video19
	/dev/video20
	/dev/video21
	/dev/video22
	/dev/video23
	/dev/video24
	/dev/video25
	/dev/video26
	/dev/video27
	/dev/video28
	/dev/video29
	/dev/video30
	/dev/video31
	/dev/video32
	/dev/video33
	/dev/video34
	/dev/video35
	/dev/video36
	/dev/video37
	/dev/video38
	/dev/video39
	/dev/video40
	/dev/video41
	/dev/video42
	/dev/video43
	/dev/video44
	/dev/video45
	/dev/video46
	/dev/video47

ipu6 (PCI:0000:00:05.0):
	/dev/media0
```
Also not working with Cheese or any other webcam app.

P3lUZa, 2025-02-25:
Now its working, the steps I did were:
1. Update the firmware (I had to install Windows 11 and run **both** 'Samsung Update' and 'Windows Update') There is no workaround for this, I tried very hard. Before finishing, make sure camera is working in Windows and then you can wipe out the operating system.

2.
`echo 'export GST_PLUGIN_PATH=/usr/lib/gstreamer-1.0' >> ~/.bashrc
source ~/.bashrc`

3. Restart the computer

`sudo -E LANG=C gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! ximagesink`

P3lUZa, 2025-02-25:
Following [](https://github.com/intel/ipu6-drivers/issues/177#issuecomment-1702176999 ) I was able to get something.

P3lUZa, 2025-02-26:
Finally its working:

1. You have to compile from source v4l2-relayd. Do it from this page: https://launchpad.net/ubuntu/+source/v4l2-relayd/0.1.2-0ubuntu3 following this instructions:

I. Manually download the files in folder v4ld_tmp
 v4l2-relayd_0.1.2.orig.tar.xz	
v4l2-relayd_0.1.2-0ubuntu3.debian.tar.xz
v4l2-relayd_0.1.2-0ubuntu3.dsc

II. Once you are in that folder and have downloaded those files run: 
`dget --extract --allow-unauthenticated https://launchpadlibrarian.net/722477133/v4l2-relayd_0.1.2-0ubuntu3_source.changes`

`cd v4l2-relayd-0.1.2`

III. Now you need to modify the file in `src/v4l2-relayd.c`, do it according to: https://gitlab.com/vicamo/v4l2-relayd/-/commit/35a06d89747ff31fbaabf744b7dae585bfa86723 

Another option is that you can directly download that file in gitlab and replace it for the one you have (this step is very important, so do it).

IV.

`sudo mk-build-deps --install --remove`
`dpkg-buildpackage --build=binary --no-sign`
`sudo d`pkg -i ../v4l2-relayd_0.1.2-0ubuntu3_amd64.deb`

Congratulations, you built from source a ppa package in Debian...

V. Some tests:

run and don't close it

```
# sudo gst-launch-1.0 -v icamerasrc ! video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! videoconvert ! video/x-raw,format=YUY2 ! v4l2sink device=/dev/video0 
```

Now open VLC: Media > Open Capture Device > video device name use /dev/video0 
and you should see the camera.

Now make it permanent:

VI.

```
# modprobe -r v4l2loopback   && modprobe  v4l2loopback  video_nr=0 card_label=MYCamera exclusive_caps=1
# cat /sys/devices/virtual/video4linux/video0/name
MYCamera
```

create file /etc/modprobe.d/v4l2loopback.conf 

and write
`options v4l2loopback card_label=MYCamera exclusive_caps=1 video_nr=0`

create the file: 
/etc/systemd/system/v4l2-relayd.service

and write:
```
[Unit]
Description=v4l2-relay daemon service
After=modprobe@v4l2loopback.service systemd-logind.service

[Service]
Type=simple
Environment="GST_PLUGIN_PATH=/usr/lib/gstreamer-1.0"
EnvironmentFile=/etc/default/v4l2-relayd
EnvironmentFile=-/etc/v4l2-relayd
ExecCondition=/usr/bin/test -n "${VIDEOSRC}"
ExecCondition=/usr/bin/test -n "$FORMAT"
ExecCondition=/usr/bin/test -n "$WIDTH"
ExecCondition=/usr/bin/test -n "$HEIGHT"
ExecCondition=/usr/bin/test -n "$FRAMERATE"
ExecCondition=/usr/bin/test -n "${CARD_LABEL}"
ExecStart=/bin/sh -c 'DEVICE=$(grep -l -m1 -E "^${CARD_LABEL}$" /sys/devices/virtual/video4linux/*/name | cut -d/ -f6); exec /usr/bin/v4l2-relayd -i "${VIDEOSRC}" $${SPLASHSRC:+-s "${SPLASHSRC}"} -o "appsrc name=appsrc caps=video/x-raw,format=${FORMAT},width=${WIDTH},height=${HEIGHT},framerate=${FRAMERATE} ! videoconvert ! v4l2sink name=v4l2sink device=/dev/$${DEVICE}"'
Restart=always

[Install]
WantedBy=multi-user.target
```

Now create file /etc/default/v4l2-relayd

and write:

```
# GStreamer source element name:
#VIDEOSRC="videotestsrc"
VIDEOSRC="icamerasrc"
#SPLASHSRC="filesrc location=/.../splash.png ! pngdec ! imagefreeze num-buffers=4 ! videoscale ! videoconvert"

# Output format, width, height, and frame rate:
#FORMAT=YUY2
FORMAT=NV12
WIDTH=1280
HEIGHT=720
FRAMERATE=30/1

# Virtual video device name:
#CARD_LABEL="Virtual Camera"
CARD_LABEL="MYCamera"

# Extra options to pass to v4l2-relayd:
#EXTRA_OPTS=-d
```
Finally:

`sudo systemctl enable v4l2-relayd.service`
`sudo systemctl daemon-reload`
`sudo systemctl start v4l2-relayd.service`

FInally reboot the computer and next time you login it shoud work: tested with Gnome Camera (snapshot), cheese, chrome, firefox, zoom.

Andycodeman, 2026-02-13:
I've put together an install script that automates the IPU6/OV02C10 webcam setup for Samsung Galaxy Book4 laptops:

https://github.com/Andycodeman/samsung-galaxy-book4-linux-fixes/tree/main/webcam-fix

It handles IVSC module loading, intel-ipu6-camera-hal installation, and PipeWire device classification. Tested on Galaxy Book4 Ultra, Ubuntu 24.04, kernel 6.17.