intel/ipu6-drivers #331: Intel MIPI Camera on Ubuntu 24.04 with kernel 6.11.0-17-generic does not work

VLK14, 2025-02-27:
I have a Lenovo ThinkPad X1 Carbon Gen 12 running Ubuntu 24.04 with kernel version 6.11.0-17-generic.

When I first got the laptop and installed the drivers following the [official documentation](https://wiki.ubuntu.com/Lenovo), everything worked correctly.

After a few days of use, the Intel MIPI Camera stopped working. Now when I use [webcamtests](https://webcamtests.com/) to check the camera, it gets stuck at starting the camera, and when I use other solutions (Like gst-launch-1.0 v4l2src ! glimagesink), it returns a black screen.

If anyone could help me fix this issue, I would appreciate it.

rfkill list

1: tpacpi_bluetooth_sw: Bluetooth
	Soft blocked: yes
	Hard blocked: no
2: nfc0: NFC
	Soft blocked: no
	Hard blocked: no
3: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
                                                                                                                                                                                              
dmesg | grep -i ipu6
[   18.054143] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[   18.054428] intel-ipu6 0000:00:05.0: Device 0x7d19 (rev: 0x4)
[   18.054572] intel-ipu6 0000:00:05.0: physical base address 0x4059000000
[   18.054574] intel-ipu6 0000:00:05.0: mapped as: 0x000000000e68b0b1
[   18.054651] intel-ipu6 0000:00:05.0: IPU in secure mode
[   18.054653] intel-ipu6 0000:00:05.0: IPU secure touch = 0x80000000
[   18.054654] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[   18.067435] intel-ipu6 0000:00:05.0: IPC reset done
[   18.069089] intel-ipu6 0000:00:05.0: FW version: 20230925
[   18.075120] intel-ipu6 0000:00:05.0: Found supported sensor OVTI08F4:00
[   18.075173] intel-ipu6 0000:00:05.0: Connected 1 cameras
[   18.077110] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[   18.124533] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[   18.173290] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[   18.173340] intel-ipu6 0000:00:05.0: IPU6-v4 driver version 1.0
[   18.475574] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[   18.475727] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0

xrn, 2025-02-28:
Hey everyone - I can confirm, found the same issue in fedora rawhide

dirkbonenkamp, 2025-04-15:
Dell Latitude 7450 with Ubuntu 24.04 and running 6.11.0-21-generic, same problem.

javi-llorente, 2025-04-16:
Dell XPS 14 9440 with Ubuntu 24.10, 6.11.0-21-generic. Same thing here.

Beor18, 2025-04-18:
same problem with black screen

Beor18, 2025-04-18:
@kity-linuxero tengo el mismo problema en la samsung galaxy book 3 pro con ubuntu 22.04 lo más gracioso que el año pasado lo habia hecho funcionar y de repente (seguramente con alguna actualización) dejo de funcionar hace poco, osea parece que arranca la camera pero se queda la imagen en negro.

picousse, 2025-05-07:
same thing here

gravit8n, 2025-05-09:
can you test this ? https://github.com/intel/ipu6-drivers/issues/228#issuecomment-2850599946

ArgentumCation, 2025-10-28:
Reproduced on Lenovo Thinkpax X1 Carbon 10th Gen 24.04 6.14.0-33-generic