intel/ipu6-drivers #333: Webcam not working in samsung book5 pro 360

ycervandez, 2025-03-06:
I am unable to find a solution to this, tried multiple things and nothing worked. Any help is appreciated

dmesg | grep -i ipu6
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.0-55-generic root=UUID=39cc68f9-f965-4dd0-98d5-fb515146c328 ro quiet splash intel_ipu6=1
[    0.043087] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.0-55-generic root=UUID=39cc68f9-f965-4dd0-98d5-fb515146c328 ro quiet splash intel_ipu6=1
[    0.043194] Unknown kernel command line parameters "splash BOOT_IMAGE=/boot/vmlinuz-6.8.0-55-generic intel_ipu6=1", will be passed to user space.
[    1.305382]     intel_ipu6=1
[    2.790811] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    2.791110] intel-ipu6 0000:00:05.0: Device 0x7d19 (rev: 0x4)
[    2.791137] intel-ipu6 0000:00:05.0: physical base address 0x5020000000
[    2.791139] intel-ipu6 0000:00:05.0: mapped as: 0x000000008ad2253d
[    2.791214] intel-ipu6 0000:00:05.0: IPU in secure mode
[    2.791219] intel-ipu6 0000:00:05.0: IPU secure touch = 0x80000000
[    2.791220] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    2.803773] intel-ipu6 0000:00:05.0: IPC reset done
[    2.804187] intel-ipu6 0000:00:05.0: loading /lib/firmware/intel/ipu6epmtl_fw.bin failed with error -22
[    2.804193] intel-ipu6 0000:00:05.0: Direct firmware load for intel/ipu6epmtl_fw.bin failed with error -22
[    2.804197] intel-ipu6 0000:00:05.0: Requesting signed firmware failed
[    2.804476] intel-ipu6: probe of 0000:00:05.0 failed with error -22

bingbucao, 2025-03-07:
Please set:
`CONFIG_VIDEO_INTEL_IPU6=m`
 in your kernel config instead of 
`CONFIG_VIDEO_INTEL_IPU6=y`

ycervandez, 2025-03-07:
Hi, 
thank you for the response, after I tried that I still cannot load the driver: 

 lsmod | grep -E 'intel_ipu6|ov02c10'
intel_ipu6_isys       159744  0
videobuf2_dma_contig    24576  1 intel_ipu6_isys
videobuf2_v4l2         36864  1 intel_ipu6_isys
videobuf2_common       81920  4 videobuf2_dma_contig,videobuf2_v4l2,intel_ipu6_isys,videobuf2_memops
intel_ipu6_psys       114688  0
intel_ipu6            118784  2 intel_ipu6_isys,intel_ipu6_psys
ipu_bridge             16384  1 intel_ipu6
ov02c10                28672  0
v4l2_fwnode            36864  2 ov02c10,intel_ipu6_isys
v4l2_async             28672  3 v4l2_fwnode,ov02c10,intel_ipu6_isys
videodev              352256  5 v4l2_async,v4l2_fwnode,videobuf2_v4l2,ov02c10,intel_ipu6_isys
mc                     81920  6 v4l2_async,videodev,videobuf2_v4l2,ov02c10,intel_ipu6_isys,videobuf2_common

any idea of where can i find the right driver? I trully appreciate your time

bingbucao, 2025-03-10:
> Hi, thank you for the response, after I tried that I still cannot load the driver:
>
> lsmod | grep -E 'intel_ipu6|ov02c10' intel_ipu6_isys 159744 0 videobuf2_dma_contig 24576 1 intel_ipu6_isys [...]

Could you share the `dmesg` log during boot?