intel/ipu6-drivers #344: Not working on Ubuntu 25.05 with kernel 6.14

bryan977, 2025-04-18:
I am creating this new issue for Ubuntu 25.05; however, it is a continuation of the [issue for Ubuntu 24.04](https://github.com/intel/ipu6-drivers/issues/228).

My System: Dell Inc. XPS 13 9340
My Built-in Webcam: Intel MIPI
Ubuntu Release: 25.04
Kernel: 6.14.0-15-generic]
Drivers:
```
$ ubuntu-drivers list
oem-somerville-treecko-meta
libcamhal0
intel-ipu6-dkms
```

I was able to get the webcam working under Ubuntu 24.04 using the process in the issue above. I tried to use the same process for Ubuntu 25.05; however, have been unsuccessful at this time. https://webcamtests.com/ DOES properly identify my camera, then I click the test button and the gear spins for awhile. Eventually, I receive a response back "The video track is paused." I never see any video.

The9Cat, 2025-04-18:
@bryan977 I suggest posting your boot output.  E.g. stuff related to the ipu6 drivers.

bryan977, 2025-04-19:
> [@bryan977](https://github.com/bryan977) I suggest posting your boot output. E.g. stuff related to the ipu6 drivers.

Updated above. Thank you!

jokla, 2025-04-21:
Hi, the camera was functioning on Ubuntu 24.10, but it is not working on Ubuntu 25.10. Dell Inc. XPS 16 9640. Please let me know if I can provide any additional information.

My System: Dell Inc. XPS 16 9640
My Built-in Webcam: Intel MIPI
Ubuntu Release: `25.04`
Kernel: `6.14.0-15-generic`

$ ubuntu-drivers list
```
nvidia-driver-570-*
intel-ipu6-dkms
```

$ sudo apt list --installed '*ipu6*'
```
intel-ipu6-dkms/plucky,now 0~git202411190607.0ad49882-0ubuntu3 amd64 [installed]
```

$ v4l2-ctl --list-devices
```
ipu6 ():
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/video6
	/dev/video7
	/dev/video8
	/dev/video9
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video13
	/dev/video14
	/dev/video15
	/dev/video16
	/dev/video17
	/dev/video18
	/dev/video19
	/dev/video20
	/dev/video21
	/dev/video22
	/dev/video23
	/dev/video24
	/dev/video25
	/dev/video26
	/dev/video27
	/dev/video28
	/dev/video29
	/dev/video30
	/dev/video31
	/dev/video32
	/dev/video33
	/dev/video34
	/dev/video35
	/dev/video36
	/dev/video37
	/dev/video38
	/dev/video39
	/dev/video40
	/dev/video41
	/dev/video42
	/dev/video43
	/dev/video44
	/dev/video45
	/dev/video46
	/dev/video47

ipu6 (PCI:0000:00:05.0):
	/dev/media0
```

$ sudo dmesg | grep ipu
```
[    2.151005] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[   14.267998] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[   14.303934] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[   14.349992] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[   14.350003] intel-ipu6 0000:00:05.0: IPU6-v4[7d19] hardware version 6
[   24.772367] auxiliary intel_ipu6.psys.40: deferred probe pending: (reason unknown)
```

$ lspci
```  
00:00.0 Host bridge: Intel Corporation Device 7d01 (rev 04)
00:01.0 PCI bridge: Intel Corporation Device 7ecc (rev 10)
00:02.0 VGA compatible controller: Intel Corporation Meteor Lake-P [Intel Arc Graphics] (rev 08)
00:04.0 Signal processing controller: Intel Corporation Meteor Lake-P Dynamic Tuning Technology (rev 04)
00:05.0 Multimedia controller: Intel Corporation Meteor Lake IPU (rev 04)
00:06.0 PCI bridge: Intel Corporation Device 7e4d (rev 20)
00:06.2 PCI bridge: Intel Corporation Device 7ecb (rev 10)
00:07.0 PCI bridge: Intel Corporation Meteor Lake-P Thunderbolt 4 PCI Express Root Port #0 (rev 10)
00:07.1 PCI bridge: Intel Corporation Meteor Lake-P Thunderbolt 4 PCI Express Root Port #1 (rev 10)
00:08.0 System peripheral: Intel Corporation Meteor Lake-P Gaussian & Neural-Network Accelerator (rev 20)
00:0a.0 Signal processing controller: Intel Corporation Meteor Lake-P Platform Monitoring Technology (rev 01)
00:0b.0 Processing accelerators: Intel Corporation Meteor Lake NPU (rev 04)
00:0d.0 USB controller: Intel Corporation Meteor Lake-P Thunderbolt 4 USB Controller (rev 10)
00:0d.2 USB controller: Intel Corporation Meteor Lake-P Thunderbolt 4 NHI #0 (rev 10)
00:12.0 Serial controller: Intel Corporation Meteor Lake-P Integrated Sensor Hub (rev 20)
00:14.0 USB controller: Intel Corporation Meteor Lake-P USB 3.2 Gen 2x1 xHCI Host Controller (rev 20)
00:14.2 RAM memory: Intel Corporation Device 7e7f (rev 20)
00:15.0 Serial bus controller: Intel Corporation Meteor Lake-P Serial IO I2C Controller #0 (rev 20)
00:15.2 Serial bus controller: Intel Corporation Meteor Lake-P Serial IO I2C Controller #2 (rev 20)
00:16.0 Communication controller: Intel Corporation Meteor Lake-P CSME HECI #1 (rev 20)
00:19.0 Serial bus controller: Intel Corporation Meteor Lake-P Serial IO I2C Controller #4 (rev 20)
00:19.1 Serial bus controller: Intel Corporation Meteor Lake-P Serial IO I2C Controller #5 (rev 20)
00:1c.0 PCI bridge: Intel Corporation Device 7e38 (rev 20)
00:1f.0 ISA bridge: Intel Corporation Device 7e02 (rev 20)
00:1f.3 Audio device: Intel Corporation Meteor Lake-P HD Audio Controller (rev 20)
00:1f.4 SMBus: Intel Corporation Meteor Lake-P SMBus Controller (rev 20)
00:1f.5 Serial bus controller: Intel Corporation Meteor Lake-P SPI Controller (rev 20)
01:00.0 VGA compatible controller: NVIDIA Corporation AD106M [GeForce RTX 4070 Max-Q / Mobile] (rev a1)
01:00.1 Audio device: NVIDIA Corporation AD106M High Definition Audio Controller (rev a1)
02:00.0 Non-Volatile memory controller: MAXIO Technology (Hangzhou) Ltd. NVMe SSD Controller MAP1602 (DRAM-less) (rev 01)
03:00.0 Network controller: Intel Corporation Wi-Fi 7(802.11be) AX1775*/AX1790*/BE20*/BE401/BE1750* 2x2 (rev 1a)
74:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5260 PCI Express Card Reader (rev 01)

```

$ lspci -nnk -d ::0480
```
00:05.0 Multimedia controller [0480]: Intel Corporation Meteor Lake IPU [8086:7d19] (rev 04)
	Subsystem: Dell Device [1028:0c62]
	Kernel driver in use: intel-ipu6
	Kernel modules: intel_ipu6
```

comcamot, 2025-04-23:
Same problems here (xps 13 9340 with ubuntu 25.,04). Just comment that **qcam works** with the fresh 25.04 install (with  intel-ipu6-dkms as the only driver). This is a good improvement over previoius versions. Cheese, guvcview and webcamoid, however, do not work. Kamoso can take pictures but does not work well. So I guess they are close.

sn99, 2025-04-25:
Hi, thanks to everyone for their work. I can get the camera working on Ubuntu 24 but not on Ubuntu 25. I did a fresh install of Ubuntu 25 to test.

The camera just shows rainbows(dummy output) on any app unless started using `gst-launch-1.0 libcamerasrc ! video/x-raw ! autovideosink` which works.

```
uname -r

6.14.0-15-generic
```

```
v4l2-ctl --list-devices   

ipu6 ():
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/video6
	/dev/video7
	/dev/video8
	/dev/video9
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video13
	/dev/video14
	/dev/video15
	/dev/video16
	/dev/video17
	/dev/video18
	/dev/video19
	/dev/video20
	/dev/video21
	/dev/video22
	/dev/video23
	/dev/video24
	/dev/video25
	/dev/video26
	/dev/video27
	/dev/video28
	/dev/video29
	/dev/video30
	/dev/video31
	/dev/video32
	/dev/video33
	/dev/video34
	/dev/video35
	/dev/video36
	/dev/video37
	/dev/video38
	/dev/video39
	/dev/video40
	/dev/video41
	/dev/video42
	/dev/video43
	/dev/video44
	/dev/video45
	/dev/video46
	/dev/video47
	/dev/video48

ipu6 (PCI:0000:00:05.0):
	/dev/media0

Virtual Camera (platform:v4l2loopback-000):
	/dev/video0
```

```
ubuntu-drivers list

intel-ipu6-dkms
```

```
lsmod | grep ipu6

intel_ipu6_psys       106496  0
intel_ipu6_isys       126976  0
videobuf2_dma_sg       20480  1 intel_ipu6_isys
videobuf2_v4l2         36864  1 intel_ipu6_isys
videobuf2_common       90112  4 videobuf2_v4l2,intel_ipu6_isys,videobuf2_dma_sg,videobuf2_memops
intel_ipu6             73728  2 intel_ipu6_isys,intel_ipu6_psys
ipu_bridge             20480  3 intel_ipu6,ivsc_csi,intel_ipu6_isys
v4l2_fwnode            40960  4 ivsc_csi,intel_ipu6_isys,ov02e10
v4l2_async             28672  4 v4l2_fwnode,ivsc_csi,intel_ipu6_isys,ov02e10
videodev              364544  8 v4l2_async,v4l2_fwnode,videobuf2_v4l2,ivsc_csi,v4l2loopback,intel_ipu6_isys,ov02e10
mc                     86016  7 v4l2_async,videodev,videobuf2_v4l2,ivsc_csi,intel_ipu6_isys,videobuf2_common,ov02e10
```

```
sudo grep ipu /var/log/dmesg

[    2.588524] kernel: intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[   14.749386] kernel: pci 0000:00:05.0: deferred probe pending: intel-ipu6: IPU6 bridge init failed
[   15.095025] kernel: intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[   15.179071] kernel: intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[   15.183475] kernel: intel-ipu6 0000:00:05.0: CSE authenticate_run done
[   15.183599] kernel: intel-ipu6 0000:00:05.0: IPU6-v4[7d19] hardware version 6
```

```
● v4l2-relayd.service - v4l2-relay daemon service
     Loaded: loaded (/usr/lib/systemd/system/v4l2-relayd.service; enabled; preset: enabled)
     Active: active (running) since Fri 2025-04-25 23:55:44 IST; 25min ago
 Invocation: 31890de5d6a14837895d5508a4884403
   Main PID: 2225 (v4l2-relayd)
      Tasks: 5 (limit: 74700)
     Memory: 477.7M (peak: 841.3M)
        CPU: 4.839s
     CGroup: /system.slice/v4l2-relayd.service
             └─2225 /usr/bin/v4l2-relayd -i videotestsrc -o "appsrc name=appsrc caps=video/x-raw,format=YUY2,width=1280,height=720,framerate=30/1 ! videoconvert ! v>

Apr 25 23:55:43 alfred-work systemd[1]: Starting v4l2-relayd.service - v4l2-relay daemon service...
Apr 25 23:55:44 alfred-work systemd[1]: Started v4l2-relayd.service - v4l2-relay daemon service.
Apr 25 23:55:44 alfred-work (sh)[2225]: v4l2-relayd.service: Referenced but unset environment variable evaluates to an empty string: SPLASHSRC
```

```
lsmod | grep -i v4l2loopback
v4l2loopback           61440  1
videodev              364544  8 v4l2_async,v4l2_fwnode,videobuf2_v4l2,ivsc_csi,v4l2loopback,intel_ipu6_isys,ov02e10
```

[image]

Although, this works without issue (the camera works) :

```
gst-launch-1.0 libcamerasrc ! video/x-raw ! autovideosink
```

Some relevant links that might help:

- https://discourse.ubuntu.com/t/intel-mipi-camera-not-working-out-of-the-box-on-25-04-where-it-did-on-24-10/59072/7
- https://wiki.ubuntu.com/IntelMIPICamera
- https://bugs.launchpad.net/oem-priority/+bug/2043174
- https://askubuntu.com/questions/1508977/how-to-get-ipu6-webcam-xps-14-9440-to-work-on-meteor-lake-platform
- https://askubuntu.com/a/1537621

One thing I have noticed every site mentions to run `sudo gst-launch-1.0 icamerasrc ! autovideosink` instead of replacing `icamerasrc->libcamerasrc`.

Another thing, the camera work ootb for Nobara 41 (unofficial spin of fedora 41), it also has the latest stable kernel but I did have to install `gstreamer1-plugins-icamerasrc` which further installed:

```
gstreamer1-plugins-icamerasrc
    akmod-intel-ipu6
    ipu6-camera-bins
    ipu6-camera-hal
    v4l2-relayd
```

sn99, 2025-04-27:
Which driver am I supposed to use? 

[image]

I installed `libcamhal0` via:

```
$ sudo apt install ubuntu-oem-keyring
$ sudo add-apt-repository "deb http://dell.archive.canonical.com/ noble somerville"
$ ubuntu-drivers list
...
libcamhal0
...
$ sudo apt install libcamhal0
```

If I try:

[image]

- On https://mozilla.github.io/webrtc-landing/gum_test.html I get a black screen.
- `sudo gst-launch-1.0 libcamerasrc ! autovideosink` still works.

```
sudo gst-launch-1.0 icamerasrc ! autovideosink

Setting pipeline to PAUSED ...
[04-27 14:34:29.403] CamHAL[INF] aiqb file name OV08X40_CJFME55_MTL.aiqb
[04-27 14:34:29.404] CamHAL[INF] aiqb file name OV08X40_CJFME55_MTL.aiqb
[04-27 14:34:29.405] CamHAL[ERR] invalid media format, default value used.
[04-27 14:34:29.405] CamHAL[INF] aiqb file name OV13B10_09B13_MTL_PDAF_DPC_T3.aiqb
[04-27 14:34:29.405] CamHAL[INF] aiqb file name OV13B10_09B13_MTL_PDAF_DPC_T3.aiqb
[04-27 14:34:29.405] CamHAL[ERR] invalid media format, default value used.
[04-27 14:34:29.405] CamHAL[INF] aiqb file name OV13B10_09B13_MTL_PDAF_DPC_T3.aiqb
[04-27 14:34:29.405] CamHAL[INF] aiqb file name OV13B10_09B13_MTL_PDAF_DPC_T3.aiqb
[04-27 14:34:29.406] CamHAL[ERR] invalid media format, default value used.
[04-27 14:34:29.406] CamHAL[INF] aiqb file name ov5675.aiqb
[04-27 14:34:29.406] CamHAL[INF] aiqb file name ov5675.aiqb
[04-27 14:34:29.406] CamHAL[ERR] invalid media format, default value used.
[04-27 14:34:29.406] CamHAL[INF] aiqb file name ov01a1s.aiqb
[04-27 14:34:29.406] CamHAL[INF] aiqb file name ov01a1s.aiqb
[04-27 14:34:29.407] CamHAL[ERR] invalid media format, default value used.
[04-27 14:34:29.407] CamHAL[INF] aiqb file name ov01a10.aiqb
[04-27 14:34:29.407] CamHAL[INF] aiqb file name ov01a10.aiqb
[04-27 14:34:29.407] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[04-27 14:34:29.407] CamHAL[INF] aiqb file name ov01a10.aiqb
[04-27 14:34:29.407] CamHAL[INF] aiqb file name ov01a10.aiqb
[04-27 14:34:29.408] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[04-27 14:34:29.408] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[04-27 14:34:29.408] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[04-27 14:34:29.408] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[04-27 14:34:29.408] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[04-27 14:34:29.408] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[04-27 14:34:29.408] CamHAL[ERR] invalid media format, default value used.
[04-27 14:34:29.408] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[04-27 14:34:29.408] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[04-27 14:34:29.409] CamHAL[ERR] invalid media format, default value used.
[04-27 14:34:29.409] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[04-27 14:34:29.409] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[04-27 14:34:29.409] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[04-27 14:34:29.409] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[04-27 14:34:29.409] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[04-27 14:34:29.409] CamHAL[ERR] invalid media format, default value used.
[04-27 14:34:29.409] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[04-27 14:34:29.409] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[04-27 14:34:29.410] CamHAL[ERR] invalid media format, default value used.
[04-27 14:34:29.410] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[04-27 14:34:29.410] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[04-27 14:34:29.410] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[04-27 14:34:29.410] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[... 137 lines cut ...]
```
v4l2-ctl --list-devices   

ipu6 (PCI:0000:00:05.0):
	/dev/media0

Intel MIPI Camera (platform:v4l2loopback-000):
	/dev/video0
```

```
sudo systemctl status v4l2-relayd.service

[sudo] password for sn99: 
● v4l2-relayd.service - v4l2-relay daemon service
     Loaded: loaded (/usr/lib/systemd/system/v4l2-relayd.service; enabled; preset: enabled)
     Active: active (running) since Sun 2025-04-27 13:58:34 IST; 58min ago
 Invocation: 057e088166914609b119c16546cdac29
   Main PID: 2035 (v4l2-relayd)
      Tasks: 5 (limit: 74700)
     Memory: 382.3M (peak: 394M)
        CPU: 9.370s
     CGroup: /system.slice/v4l2-relayd.service
             └─2035 /usr/bin/v4l2-relayd -i "icamerasrc buffer-count=7" -o "appsrc name=appsrc caps=video/x-raw,f>
```

luigimassa, 2025-05-07:
> Which driver am I supposed to use?

Equal

dlsoltero, 2025-05-08:
Hello,
I don't know if it helps at all but I have the same issues,  I'm using  a Dell Inc. Precision 5470 with  ubuntu 25.04. I selected "using the IPU6 driver from intel-ipu6-dkms" in the Software & Updates app or whatever it is called.  I installed a few packages for v4l2 and gstream  and although `cheese` fails and the `snapshot` or any other app that uses `/dev/video0` return a video test image. But the camera work with both `gst-launch-1.0 libcamerasrc ! video/x-raw,width=1024,height=768 ! autovideosink` or `qcam`.

`v4l2-ctl --list-devices`  list my camera as a Virtual Camera (platform:v4l2loopback-000) and not MIPI as should be.

Logicwax, 2025-05-22:
nope, not working unfortunately.

The9Cat, 2025-05-22:
I'm still using 24.04 with the patched 6.11.  But just thought I'd ask: has anyone here tried **not** using the `intel-ipu6-dkms` and just relying on the in-tree drivers which have been patched for many sensors?  That's my 24.04 patched 6.11 solution and it works fine for me.

bryan977, 2025-05-23:
> I'm still using 24.04 with the patched 6.11. But just thought I'd ask: has anyone here tried **not** using the `intel-ipu6-dkms` and just relying on the in-tree drivers which have been patched for many sensors? That's my 24.04 patched 6.11 solution and it works fine for me.

I do see it installed. I was using "Using HAL library for MIPI camera through Intel IPU6 from libcamhal0 (open source). I switched to "Using Intel Integrated Image Processing Unit 6 (IPU6) driver from intel-ipu6-dkms (open source) and have the same non-working result. Is there a process I need to perform in addition to selecting the driver?

```
$ ubuntu-drivers list
oem-somerville-treecko-meta
libcamhal0
intel-ipu6-dkms
```

kieranatkins, 2025-06-03:
Hi all,

I'm having the same issue on my Dell Latitude 7450. I installed the 'oem-somerville-oddish-meta' package on a fresh 25.04 install after adding the somerville noble repository, and selected the libcamhal0 driver. However on the Mozilla gUM test page the webcam still appears as black. None of the other driver options seem to work either.

Note, `sudo dmesg | grep -i ipu6` does output something I don't see on my working 24.04 LTS partition: 
```
[    2.125139] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    2.138094] intel-ipu6 0000:00:05.0: Found supported sensor INT3537:00
[    2.138212] intel-ipu6 0000:00:05.0: Connected 1 cameras
[    2.139574] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[    2.185721] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[    2.234500] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[    2.234507] intel-ipu6 0000:00:05.0: IPU6-v4[7d19] hardware version 6
[   14.958875] auxiliary intel_ipu6.psys.40: deferred probe pending: (reason unknown)
```
The final line I don't see on the working partition, and it says the same for others in this thread?

dario-colombo, 2025-06-05:
Why are we doomed to struggle with this driver, like forever.. /rant

ph818, 2025-06-18:
Hey, is there a general flow to troubleshooting this webcam? I have a Lenovo Thinkpad X12 Detachable Gen 1. The Webcam was working in 24.04, but is no longer working in 25.04. I could downgrade to 24.04, but then the hardware volume keys don't work.

adrianofoschi, 2025-06-19:
The same behavior on my Dell XPS 13, black screen output.
It worked with ubuntu 24.10.

Jojoooo1, 2025-06-23:
Same here using XPS Dell Inc. XPS 16 9640 with ubuntu 25.04.

enbzz, 2025-06-25:
Same here. Lenovo X1 carbon Gen 12. was working on ubuntu 24.10

jaimebamboo, 2025-06-27:
has anybody tried with the patch for kernel 6.16 released today?

lainsw, 2025-06-27:
> How can we test it?
> 
> Il Ven 27 Giu 2025, 09:35 Jaime ***@***.***> ha scritto:
> […](#)

> has anybody tried with the patch for kernel 6.16 released today?

Based on https://cdn.kernel.org/ , the kernel version released today (27-Jun-2025)  is 6.15.2. 

6.16 is still on release candidate 2 (22-Jun-2025), but will be a mainline release.

The [release notes for 6.15.2](https://cdn.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.15.4#:~:text=commit%20bd96f6681185391447b401612b89ae8cd421daf3%0AAuthor,gregkh%40linuxfoundation.org%3E) seem to indicate that there is a faulty error code that has been fixed when trying to gain access to cameras: 

`uvc_gpio_parse() can return -EPROBE_DEFER when the GPIOs it depends on
have not yet been probed. This return code should be propagated to the caller of uvc_probe() to ensure that probing is retried when the required GPIOs become available.`
  `Currently, this error code is incorrectly converted to -ENODEV, causing some internal cameras to be ignored.`

I am on 24.04.2, which has the same problem (running Kernel 6.8.0). The DEFER/retry kind of makes sense, because I got my external Logitech camera to work exactly once - must have gotten lucky - but haven't had it work since.

24.10 runs 6.11, but is about to go end-of-life.

And as of now, 25.10 is planning kernel 6.17, with an expected release of October 2025.

So, I might limp along until then, or wait for the mainline 6.16 kernel to be officially released. 

But 6.16 is probably worth a shot for 25.04, I would think. I don't know if LTS status for 24.04 still applies if you swap out the kernel for 6.16 (or 6.17).

berglh, 2025-07-07:
> has anybody tried with the patch for kernel 6.16 released today?

- Dell Precision 5490
- 00:05.0 Multimedia controller [0480]: Intel Corporation Meteor Lake IPU [8086:7d19] (rev 04)
	Subsystem: Dell Device [1028:0cc7]
	Kernel driver in use: intel-ipu6
	Kernel modules: intel_ipu6

I installed `6.16.0-061600rc5-generic` today and basically have the same issue. I get the 48 /dev/video devices and the one /dev/media device.

`gst-launch-1.0 libcamerasrc ! video/x-raw ! autovideosink` is showing video just fine, but cheese and everything else fails to detect it as a camera.

I've only tried using the Ubuntu Drivers installation version of the driver, with and without I get pretty much the same thing. Not sure if there is anything else anyone would like me to try, happy to give stuff a whirl.

yoelii, 2025-07-25:
Hi, 

I have the same issue with Dell Precision 5690, ubuntu 24.04.2 LTS, Bios 1.13.1 kernel 6.14.0-24-generic

jnzhnd, 2025-07-30:
Same issue on a Dell Precision 5680, Ubuntu 24.04.2 LTS, Kernel: Linux 6.14.0-27-generic, Firmware Version: 1.13.0

ksket, 2025-08-01:
> * Dell Precision 5490
> * 00:05.0 Multimedia controller [0480]: Intel Corporation Meteor Lake IPU [8086:7d19] (rev 04)
>   Subsystem: Dell Device [1028:0cc7]
>   Kernel driver in use: intel-ipu6
>   Kernel [...]

I was experiencing the same results as you, works fine with `gst-launch1.0` but not in browser. I checked the _v4l2-relayd_ service status (enabled and running). 

So i tried to run `sudo modprobe v4l2loopback` and got:

> ➜ ~ sudo modprobe v4l2loopback
> modprobe: ERROR: could not insert 'v4l2loopback': Key was rejected by service

I followed instructions on how to import a key into EFI in this [post](https://vcheng.org/2023/01/22/ubuntu-linux-secure-boot-dkms/) (critical part here!)

Reboot

Restart _v4l2-relayd_ service with `sudo systemctl restart v4l2-relayd.service`

Test in the [web browser](https://webcamtests.com/)

It works _**for me**_ (until next kernel upgrade ?)

luigimassa, 2025-08-04:
Hi,
You can leave the original file as is and just edit it
```
ExecStart=/usr/bin/gst-launch-1.0 \
  libcamerasrc ! video/x-raw,format=RGBA,width=1272,height=800,framerate=30/1 \
  ! videoconvert ! video/x-raw,format=YUY2 ! v4l2sink device=/dev/video48
Restart=always
```
Assign the correct device number in `/dev/video48`

berglh, 2025-08-04:
> Hi, You can leave the original file as is and just edit it
>
> ```
> ExecStart=/usr/bin/gst-launch-1.0 \
>   libcamerasrc ! video/x-raw,format=RGBA,width=1272,height=800,framerate=30/1 \
>   ! videoconvert ! [...]

Yes, indeed you can. Just be warned the package may update this file and you'll need to make the changes again in the future. Putting in `/etc/systemd/system` acts as override to the pacakge managed file.

luigimassa, 2025-08-04:
the problem is the light of the webcam, now
The light remains always on

luigimassa, 2025-08-04:
yes... a manual temporary solution
sob

kieranatkins, 2025-08-13:
Hi all,

So, for my Latitude 7450 I managed to get the webcam to work on a fresh Ubuntu 25.04 install. This required using the Ubuntu LTS dell repositories.

First, I add the dell somerville repositories for my specific laptop (oddish)

`deb http://dell.archive.canonical.com/ noble somerville-oddish`
`deb http://dell.archive.canonical.com/ noble somerville`

These repositories require signing, but I'm lazy so I just used the `apt update` flag that allows insecure repositories.

`sudo apt update --allow-insecure-repositories`.

Install the required packages

`sudo apt install somerville-oddish-meta libcamhal0 libcamhal-ipu6epmtl intel-ipu6-dkms`

Restart. (I did need to restart a few times and make sure the `libcamhal-ipu6epmtl` driver was selected in the ubuntu-drivers GUI) 

For reference my kernel is 6.14.0-27-generic, and did a fresh install of 25.04 and did all updates as of yesterday.

RodolpheGohard, 2026-09-18:
Fixed with help from an agent. The pb on my side was a missing package intel-ipu6-psys

`sudo apt install linux-modules-ipu6-generic-hwe-24.04 && sudo modprobe intel-ipu6-psys`

kskalski, 2026-09-19:
Finally after >1 years of trying I got the camera with proper image quality working

### Working on ThinkPad X1 Carbon Gen 12 (ov08x40), Ubuntu 26.04, kernel 7.0.0-31

**The OEM PPA now publishes a `resolute` (26.04) series**, so nothing has to be pinned to `noble` any more. The builds are current: `gstreamer1.0-icamera` from Sept 2026, `libcamhal` from Jan 2026. Check before you start:

```bash
sudo add-apt-repository ppa:oem-solutions-group/intel-ipu6
sudo apt update
apt-cache policy libcamhal0   # candidate must say resolute, not noble
```

Then install the HAL, the GStreamer source, and the relay:

```bash
sudo apt install libcamhal-ipu6epmtl gstreamer1.0-icamera v4l2-relayd
sudo reboot
```

Note `libcamhal-ipu6epmtl`, not just `libcamhal-ipu6epmtl-common`. The `-common` package holds only the `.aiqb` tuning data, so installing it alone gives you a detected camera and a black screen.

I did **not** need `linux-modules-ipu6-generic-hwe-*` — `intel_ipu6_psys` was already loaded on the stock 26.04 kernel. Check `lsmod | grep psys` before chasing that.

No need for "somerville" / "dacia" packages.

bossie, 2026-09-28:
@kskalski that fixed it for me (Ubuntu 26.04 on a Dell Inc. Precision 5690), thanks! 👍