intel/ipu6-drivers #356: IPU6 on Linux: Frustrating Experience, Lack of Usable Drivers

gutopardini, 2025-05-27:
Hi team,

I'm writing this issue out of deep frustration.

I've spent countless hours trying to get my IPU6-based camera to work properly on Linux. I’ve followed every guide, tried multiple distros (Fedora, Ubuntu, etc.), installed the IPU6 drivers from this repository and others, and even compiled kernels manually — all in vain.

The camera is either not detected at all or fails to initialize.

There's barely any meaningful feedback in the logs.

Many people across forums and GitHub issues seem to face the exact same situation.

Documentation is outdated or incomplete.

The few success cases rely on very specific kernel and distro combinations, which are unrealistic for regular users.

This has been a maddening experience — so maddening, in fact, that I'm literally switching laptops just to get away from this nightmare. That’s how desperate it’s gotten.

Why is Intel still shipping hardware that isn't properly supported in upstream Linux?

I truly want to help improve this situation, but we need:

Clear documentation on supported distros and kernel versions.

A realistic timeline for mainline support.

At the very least, an installer script or DKMS module that makes this process manageable.

Please consider making this driver usable for real-world Linux users.

Thanks for reading. I hope this can move the needle forward — even a little.

bnicolae, 2025-05-29:
Intel is in financial trouble and in survival mode, Linux support is probably not their top priority right now. Shame, because it will lose them customers in the long run.

harrykipper, 2025-07-25:
> It's been two months since I changed my laptop and I've never had a problem again

did you get a non-Intel laptop?

saintcodec, 2026-08-25:
Just wanted to drop a positive update here. On a Dell Latitude 9430, installed Fedora Workstation 44, and the camera just works out of the box, no extra packages, no DKMS, nothing. (Probably thanks to Hans de Goede)
Colors are a bit blue-ish but not a big deal for me.
Only issue is Chrome won't show the permission prompt at all (some xdg-desktop-portal-gnome bug maybe), but Firefox works fine so I'm just using that for now.

saintcodec, 2026-08-28:
@cachapa 
I'm on libcamera-0.7.1-1.fc44.x86_64
On Snapshot I get  ~9% CPU for the app and ~10% for wireplumber.
On a Google Meet call in Firefox, the "Isolated Web Co" process is around 73% CPU and wireplumber around 8%.
But tbh, i don't feel any performance issues.

Johndeep, 2026-09-04:
With Debian 13 stable, I tried to selective upgrade to libcamera 7.2-1 from testing. The kernel is stable at version 6.12.107-deb13. I additionally installed the current ipu6-drivers from this repo since it is not yet included in the kernel 6.x.

Cold start gnome-snapshot the first time, I get a frozen image with low brightness. Starting the second time, I get video signal, but due to it adjusting the brightness, the footage is stuttering. After reaching correct brightness, the footage is finally stable. This smooth state is remembered until system restart.

In Firefox, I get additional ~5% CPU load if webcam is running.

The reason for libcamera 7.2+ is because of this blog https://jetm.github.io/blog/posts/ipu6-webcam-libcamera-on-linux/, where the AGC flicker problem bug is supposedly fixed from then on. That is what lead to the annoying brightness adjustments, which often spirals out of control, prior in libcamera version 7.1 and less I think.