intel/ipu6-drivers #369: Cannot build intel-ipu6-dkms for kernel 6.14.0-24-generic

azurtoise5, 2025-07-18:
Dear all,

I updated to 6.14.0-24-generic and lost functionality of my web camera. Could you update the headers "#include <asm/unaligned.h" and a few other compilation errors, see log below?

**System:**
Hardware: Dell Inc. XPS 16 9640
OS: Ubuntu 24.04.2 LTS
Kernel: 6.14.0-24-generic

**Steps to reproduce:**

```
sudo apt install intel-ipu6-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  ...
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  intel-ipu6-dkms
0 upgraded, 1 newly installed, 0 to remove and 13 not upgraded.
Need to get 0 B/199 kB of archives.
After this operation, 1 443 kB of additional disk space will be used.
Selecting previously unselected package intel-ipu6-dkms.
(Reading database ... 387245 files and directories currently installed.)
Preparing to unpack .../intel-ipu6-dkms_0~git202406240945.aecec2aa-0ubuntu2~24.04.3_amd64.deb ...
Unpacking intel-ipu6-dkms (0~git202406240945.aecec2aa-0ubuntu2~24.04.3) ...
Setting up intel-ipu6-dkms (0~git202406240945.aecec2aa-0ubuntu2~24.04.3) ...
Loading new ipu6-drivers-0~git202406240945.aecec2aa-0ubuntu2~24.04.3 DKMS files...
Building for 6.14.0-24-generic
Building initial module for 6.14.0-24-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/intel-ipu6-dkms.0.crash'
Error! Bad return status for module build on kernel: 6.14.0-24-generic (x86_64)
Consult /var/lib/dkms/ipu6-drivers/0~git202406240945.aecec2aa-0ubuntu2~24.04.3/build/make.log for more information.
dpkg: error processing package intel-ipu6-dkms (--configure):
 installed intel-ipu6-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 intel-ipu6-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

The log, /var/lib/dkms/ipu6-drivers/0~git202406240945.aecec2aa-0ubuntu2~24.04.3/build/make.log, states

```
DKMS make.log for ipu6-drivers-0~git202406240945.aecec2aa-0ubuntu2~24.04.3 for kernel 6.14.0-24-generic (x86_64)
fre 18 jul 2025 10:05:46 CEST
make -C /lib/modules/6.14.0-24-generic/build M=/var/lib/dkms/ipu6-drivers/0~git202406240945.aecec2aa-0ubuntu2~24.04.3/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.14.0-24-generic'
make[2]: Entering directory '/var/lib/dkms/ipu6-drivers/0~git202406240945.aecec2aa-0ubuntu2~24.04.3/build'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  You are using:           gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  CC [M]  drivers/media/i2c/hm11b1.o
  CC [M]  drivers/media/i2c/ov01a1s.o
  CC [M]  drivers/media/i2c/ov01a10.o
  CC [M]  drivers/media/i2c/ov02c10.o
  CC [M]  drivers/media/i2c/ov02e10.o
  CC [M]  drivers/media/i2c/ov08a10.o
  CC [M]  drivers/media/i2c/ov08x40.o
  CC [M]  drivers/media/i2c/ov13858_intel.o
  CC [M]  drivers/media/i2c/ov2740.o
  CC [M]  drivers/media/i2c/ov8856.o
  CC [M]  drivers/media/i2c/hm2170.o
  CC [M]  drivers/media/i2c/hm2172.o
  CC [M]  drivers/media/i2c/hi556.o
drivers/media/i2c/ov02c10.c:4:10: fatal error: asm/unaligned.h: No such file or directory
    4 | #include <asm/unaligned.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
drivers/media/i2c/hm11b1.c:4:10: fatal error: asm/unaligned.h: No such file or directory
    4 | #include <asm/unaligned.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
drivers/media/i2c/ov01a10.c:4:10: fatal error: asm/unaligned.h: No such file or directory
    4 | #include <asm/unaligned.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
....
make[5]: *** [/usr/src/linux-headers-6.14.0-24-generic/scripts/Makefile.build:207: drivers/media/i2c/hm2170.o] Error 1
  CC [M]  drivers/media/pci/intel/ipu6/../ipu.o
  CC [M]  drivers/media/pci/intel/ipu6/../ipu-bus.o
  CC [M]  drivers/media/pci/intel/ipu6/../ipu-dma.o
  CC [M]  drivers/media/pci/intel/ipu6/../ipu-mmu.o
  CC [M]  drivers/media/pci/intel/ipu6/../ipu-buttress.o
  CC [M]  drivers/media/pci/intel/ipu6/../ipu-trace.o
  CC [M]  drivers/media/pci/intel/ipu6/../ipu-cpd.o
  CC [M]  drivers/media/pci/intel/ipu6/ipu6.o
  CC [M]  drivers/media/pci/intel/ipu6/../ipu-fw-com.o
  CC [M]  drivers/media/pci/intel/ipu6/../ipu-isys.o
  CC [M]  drivers/media/pci/intel/ipu6/../ipu-isys-csi2.o
  CC [M]  drivers/media/pci/intel/ipu6/ipu6-isys.o
  CC [M]  drivers/media/pci/intel/ipu6/ipu6-isys-phy.o
  CC [M]  drivers/media/pci/intel/ipu6/ipu6-isys-dwc-phy.o
  CC [M]  drivers/media/pci/intel/ipu6/ipu6-isys-csi2.o
  CC [M]  drivers/media/pci/intel/ipu6/ipu6-isys-gpc.o
  CC [M]  drivers/media/pci/intel/ipu6/../ipu-isys-csi2-be-soc.o
  CC [M]  drivers/media/pci/intel/ipu6/../ipu-fw-isys.o
  CC [M]  drivers/media/pci/intel/ipu6/../ipu-isys-video.o
  CC [M]  drivers/media/pci/intel/ipu6/../ipu-isys-queue.o
  CC [M]  drivers/media/pci/intel/ipu6/../ipu-isys-subdev.o
drivers/media/pci/intel/ipu6/../ipu-trace.c:520:19: error: ‘no_llseek’ undeclared here (not in a function); did you mean ‘noop_llseek’?
  520 |         .llseek = no_llseek,
      |                   ^~~~~~~~~
      |                   noop_llseek
make[6]: *** [/usr/src/linux-headers-6.14.0-24-generic/scripts/Makefile.build:207: drivers/media/pci/intel/ipu6/../ipu-trace.o] Error 1
make[6]: *** Waiting for unfinished jobs....
In file included from /usr/src/linux-headers-6.14.0-24-generic/include/linux/module.h:22,
                 from /usr/src/linux-headers-6.14.0-24-generic/include/linux/device/driver.h:21,
                 from /usr/src/linux-headers-6.14.0-24-generic/include/linux/device.h:32,
                 from /usr/src/linux-headers-6.14.0-24-generic/include/linux/acpi.h:14,
                 from drivers/media/pci/intel/ipu6/../ipu.c:4:
drivers/media/pci/intel/ipu6/../ipu.c:964:18: error: expected ‘,’ or ‘;’ before ‘INTEL_IPU_BRIDGE’
  964 | MODULE_IMPORT_NS(INTEL_IPU_BRIDGE);
      |                  ^~~~~~~~~~~~~~~~
/usr/src/linux-headers-6.14.0-24-generic/include/linux/moduleparam.h:26:61: note: in definition of macro ‘__MODULE_INFO’
   26 |                 = __MODULE_INFO_PREFIX __stringify(tag) "=" info
      |                                                             ^~~~
/usr/src/linux-headers-6.14.0-24-generic/include/linux/module.h:301:33: note: in expansion of macro ‘MODULE_INFO’
  301 | #define MODULE_IMPORT_NS(ns)    MODULE_INFO(import_ns, ns)
      |                                 ^~~~~~~~~~~
make[6]: *** [/usr/src/linux-headers-6.14.0-24-generic/scripts/Makefile.build:207: drivers/media/pci/intel/ipu6/../ipu.o] Error 1
drivers/media/i2c/ov13858_intel.c: In function ‘ov13858_do_get_pad_format’:
drivers/media/i2c/ov13858_intel.c:1585:29: warning: unused variable ‘sd’ [-Wunused-variable]
 1585 |         struct v4l2_subdev *sd = &ov13858->sd;
      |                             ^~
```

a5ehren, 2025-07-18:
I'm like 99% sure this is a Ubuntu problem where they didn't take their patches (that are already in this repo) for the 24.04 version of intel-ipu6-dkms.

eric-jahn, 2025-07-20:
fwiw, I got out of the problem by doing 
`sudo dpkg --purge intel-ipu6-dkms `
and the updating again.

NVolcz, 2025-07-22:
I had the same issue and the purging the package made the kernel panic that I had go away but the webcam doesn't work. My guess would be that I upgraded from a kernel <6.12 to 6.14 which has renamed the asm/unaligned.h. Here is a [similar issue](https://github.com/DIGImend/digimend-kernel-drivers/issues/706)

yoelii, 2025-07-24:
I have the same Issue : no camera.
Dell Precision 5690
Ubuntu 24.04.2 LTS, Kernel 6.14.0-24-generic.

I try to purge intel-ipu6-dkms and to re-install libcamhal0 who works previously but nothing work now.

bentsi, 2025-07-25:
same issue 

Ubuntu 24.04
6.11.0-29-generic #29~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Jun 26 14:16:59 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

```
DKMS make.log for ipu6-drivers-0~git202406240945.aecec2aa-0ubuntu2~24.04.3 for kernel 6.14.0-24-generic (x86_64)
Fri Jul 25 11:12:15 AM IDT 2025
make -C /lib/modules/6.14.0-24-generic/build M=/var/lib/dkms/ipu6-drivers/0~git202406240945.aecec2aa-0ubuntu2~24.04.3/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.14.0-24-generic'
make[2]: Entering directory '/var/lib/dkms/ipu6-drivers/0~git202406240945.aecec2aa-0ubuntu2~24.04.3/build'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  You are using:           gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  CC [M]  drivers/media/i2c/hm11b1.o
  CC [M]  drivers/media/i2c/ov01a1s.o
  CC [M]  drivers/media/i2c/ov01a10.o
  CC [M]  drivers/media/i2c/ov02c10.o
  CC [M]  drivers/media/i2c/ov02e10.o
  CC [M]  drivers/media/i2c/ov08a10.o
  CC [M]  drivers/media/i2c/ov08x40.o
  CC [M]  drivers/media/i2c/ov13858_intel.o
  CC [M]  drivers/media/i2c/ov2740.o
  CC [M]  drivers/media/i2c/ov8856.o
  CC [M]  drivers/media/i2c/hm2170.o
  CC [M]  drivers/media/i2c/hm2172.o
drivers/media/i2c/ov02c10.c:4:10: fatal error: asm/unaligned.h: No such file or directory
    4 | #include <asm/unaligned.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
  CC [M]  drivers/media/i2c/hi556.o
drivers/media/i2c/hm11b1.c:4:10: fatal error: asm/unaligned.h: No such file or directory
    4 | #include <asm/unaligned.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
drivers/media/i2c/ov01a1s.c:4:10: fatal error: asm/unaligned.h: No such file or directory
    4 | #include <asm/unaligned.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [/usr/src/linux-headers-6.14.0-24-generic/scripts/Makefile.build:207: drivers/media/i2c/hm11b1.o] Error 1
make[5]: *** Waiting for unfinished jobs....
drivers/media/i2c/ov01a10.c:4:10: fatal error: asm/unaligned.h: No such file or directory
    4 | #include <asm/unaligned.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [/usr/src/linux-headers-6.14.0-24-generic/scripts/Makefile.build:207: drivers/media/i2c/ov02c10.o] Error 1
make[5]: *** [/usr/src/linux-headers-6.14.0-24-generic/scripts/Makefile.build:207: drivers/media/i2c/ov01a1s.o] Error 1
drivers/media/i2c/ov02e10.c:4:10: fatal error: asm/unaligned.h: No such file or directory
    4 | #include <asm/unaligned.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [/usr/src/linux-headers-6.14.0-24-generic/scripts/Makefile.build:207: drivers/media/i2c/ov01a10.o] Error 1
drivers/media/i2c/ov08a10.c:4:10: fatal error: asm/unaligned.h: No such file or directory
    4 | #include <asm/unaligned.h>
      |          ^~~~~~~~~~~~~~~~~
make[5]: *** [/usr/src/linux-headers-6.14.0-24-generic/scripts/Makefile.build:207: drivers/media/i2c/ov02e10.o] Error 1
compilation terminated.
make[5]: *** [/usr/src/linux-headers-6.14.0-24-generic/scripts/Makefile.build:207: drivers/media/i2c/ov08a10.o] Error 1
drivers/media/i2c/ov08x40.c:4:10: fatal error: asm-generic/unaligned.h: No such file or directory
    4 | #include <asm-generic/unaligned.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [/usr/src/linux-headers-6.14.0-24-generic/scripts/Makefile.build:207: drivers/media/i2c/ov08x40.o] Error 1
drivers/media/i2c/hm2172.c:4:10: fatal error: asm/unaligned.h: No such file or directory
    4 | #include <asm/unaligned.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
drivers/media/i2c/ov8856.c:4:10: fatal error: asm/unaligned.h: No such file or directory
    4 | #include <asm/unaligned.h>
      |          ^~~~~~~~~~~~~~~~~
[... 138 lines cut ...]
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/intel-ipu6-dkm
s.0.crash'
Error! Bad return status for module build on kernel: 6.14.0-24-generic (x86_64)
Consult /var/lib/dkms/ipu6-drivers/0~git202406240945.aecec2aa-0ubuntu2~24.04.3/
build/make.log for more information.
dkms autoinstall on 6.14.0-24-generic/x86_64 succeeded for usbio-drivers
dkms autoinstall on 6.14.0-24-generic/x86_64 failed for ipu6-drivers(10)
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
 * dkms: autoinstall for kernel 6.14.0-24-generic
   ...fail!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
dpkg: error processing package linux-image-6.14.0-24-generic (--configure):
 installed linux-image-6.14.0-24-generic package post-installation script subpr
ocess returned error exit status 11
No apport report written because MaxReports is reached already
                                                              Errors were encou
ntered while processing:
 linux-headers-6.14.0-24-generic
 linux-headers-generic-hwe-24.04
 linux-generic-hwe-24.04
 linux-image-6.14.0-24-generic
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

lucasresck, 2025-07-26:
- Hi, everyone.
- I also had my webcam stop working recently.
- It is a Dell Latitude 7450 running Ubuntu 24.04.2 LTS and kernel Linux 6.14.0-24-generic.
- Trying to fix this, I was following this discussion: https://discourse.ubuntu.com/t/dell-7450-camera-not-found/55121
- Then, I came to a similar error to this issue when I run try to upgrade/install `intel-ipu6-dkms`.
- I'm afraid of downgrading the kernel as this is my work laptop so I may wait for another solution soon.
- People at this thread is also suferring from this issue: https://bugs.launchpad.net/ubuntu/+source/dkms/+bug/2117143
- Here we go again!

- I was able to fix the issue by downgrading the kernel and installing `intel-ipu6-dkms` again.
- To downfrade the kernel, go to the GRUB while the machine is booting and select "Advanced Ubuntu options"
- Select an older kernel
- Boot the system
- Uninstall the current kernel and install the package again:
```

bentsi, 2025-07-28:
Confirming, works when downgrading to `6.11.0-29-generic`

Thanks @lucasresck! 
In my case apt removing does't help to remove from grub menu and from the packages
when I run `sudo apt-get remove linux-image-unsigned-6.14.0-24-generic`
I get in the `dpkg --list`
```
ii  linux-image-unsigned-6.14.0-24-generic           6.14.0-24.24~24.04.3                        amd64        Linux kernel image for version 6.14.0
```
what helped me 
- `sudo apt remove linux-modules-ipu6-6.14.0-24-generic`
- `sudo apt-get remove linux-modules-usbio-6.14.0-24-generic`
- `sudo apt-get purge linux-image-unsigned-6.14.0-24-generic`
- `sudo dpkg --purge linux-image-6.14.0-24-generic`

Federica-Rike, 2025-08-01:
I just added this to the main thread where I suggested the solution to the MIPI camera issues, so more people can get help downgrading. Thanks for opening this, and for all the contributions!

ksket, 2025-08-01:
Hi,

After many hours (days?) spent on this, I finally managed to make my _Intel MIPI Camera_ works on my _Dell Precision 5690_ with **Ubuntu 24.04** with kernel **6.14.0-27-generic**.

It is not an _out of the box_ solution, i make it work with the _**v4l2loopback + v4l2-relayd**_ combo.

### System specs<hr>

- Machine: Dell Precision 5690
- OS: Ubuntu 24.04.2 LTS (KDE Plasma)
- Kernel: Linux 6.14.0-27-generic

### What has worked for me<hr>

So i got this new laptop at work, installed Ubuntu but webcam was not working at all, unusable in Firefox, nor Chrome.
I tried many things  before i make it work (for how long though ?).

- Install ipu6-drivers from the PPA as described in the [Intel MIPI Camera wiki](https://wiki.ubuntu.com/IntelMIPICamera), mine is named **libcamhal-ipu6epmtl**
I have followed the instructions at _**Install the userspace stack manually from the development PPA**_ section. (do not forget to _remove all existing IPU6/IPU7 packages first_ , this step seems essential)

- Make sure your driver is used, in GUI:

[image]

- reboot 

- Test that the camera is properly working by typing in terminal:
`sudo gst-launch-1.0 icamerasrc ! autovideosink`

- Remove the usage of ppa:oem-solutions-group/intel-ipu6 (step described in the wiki)

- Now, that the webcam is functional, we need to make it work in applications and web browsers especially, To do so, I followed this nice [article](https://dshedd.com/2023/07/18/fixing-lenovos-mipi-camera-problems-on-ubuntu-22-04/). Go straight to the steps where you need to install _v4l2loopback-dkms_ and _v4l2-relayd_ (drivers are already installed on previous steps described here)

- when i ran `v4l2-ctl --list-devices`, i had a permission issue, so i had to put myself in the `video` group (in terminal: `sudo usermod -a -G video YOUR_USERNAME`), then log out and log in, verify that you are part of the group by running `groups` command.

- i checked the _v4l2-relayd_ service status, it was enabled and running, **but** still no camera detected in web browser !

- something is off, so, i ran `sudo modprobe v4l2loopback` and got:

> ➜  ~ sudo modprobe v4l2loopback
> modprobe: ERROR: could not insert 'v4l2loopback': Key was rejected by service

- Follow instructions on how to import a key into EFI in this [post](https://vcheng.org/2023/01/22/ubuntu-linux-secure-boot-dkms/)  (**critical part here!**)

- After rebooting and making sure your key is enrolled, check the _v4l2-relayd_ service status again (you might need to restart)

- Test your webcam in the [web browser](https://webcamtests.com/)

Good luck !

### Related links<hr>

- Ubuntu Intel MIPI webcam wiki: https://wiki.ubuntu.com/IntelMIPICamera
- v4l2loopback + v4l2-relayd workaround: https://dshedd.com/2023/07/18/fixing-lenovos-mipi-camera-problems-on-ubuntu-22-04/
- Enable MOK on Ubuntu: https://vcheng.org/2023/01/22/ubuntu-linux-secure-boot-dkms/

LuisVasquezBSC, 2025-08-01:
> Hi,
> 
> After many hours (days?) spent on this, I finally managed to make my _Intel MIPI Camera_ works on my _Dell Precision 5690_ with **Ubuntu 24.04** with kernel **6.14.0-27-generic**.
> 
> It is not an _out of the box_ solution, i make it work with the _**v4l2loopback + v4l2-relayd**_ combo.
> ### System specs
> 
>     * Machine: Dell Precision 5690
> 
>     * OS: Ubuntu 24.04.2 LTS (KDE Plasma)
> 
>     * Kernel: Linux 6.14.0-27-generic
> 
> 
> ### What has worked for me
> 
> So i got this new laptop at work, installed Ubuntu but webcam was not working at all, unusable in Firefox, nor Chrome. I tried many things before i make it work (for how long though ?).
> 
>     * Install ipu6-drivers from the PPA as described in the [Intel MIPI Camera wiki](https://wiki.ubuntu.com/IntelMIPICamera), mine is named **libcamhal-ipu6epmtl**
>       I have followed the instructions at _**Install the userspace stack manually from the development PPA**_ section. (do not forget to _remove all existing IPU6/IPU7 packages first_ , this step seems essential)
> 
>     * Make sure your driver is used, in GUI:
> 
> 
> <img alt="Image" width="1019" height="729" src="https://private-user-images.githubusercontent.com/10031068/473470361-949ad3bc-5a2f-4c5e-acde-b98e4d9220e3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQwNjQ2ODUsIm5iZiI6MTc1NDA2NDM4NSwicGF0aCI6Ii8xMDAzMTA2OC80NzM0NzAzNjEtOTQ5YWQzYmMtNWEyZi00YzVlLWFjZGUtYjk4ZTRkOTIyMGUzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODAxVDE2MDYyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMxMWRhNTg0ZTQ4N2U1YTc4NDNkYjQ0MmE5YjYzZTZjYTY5NzY5ZDcyNTYwMTczMGE1NGY2MWFjNTYwYmY2M2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.3Qfwcl4rY7JvAcBlG8gyAhNoQK4auS6naSmQZfwtHjM">
> 
>     * reboot
> 
>     * Test that the camera is properly working by typing in terminal:
>       `sudo gst-launch-1.0 icamerasrc ! autovideosink` (you might need to `sudo apt-get install gstreamer1.0-tools`)
> 
>     * Remove the usage of ppa:oem-solutions-group/intel-ipu6 (step described in the wiki)
> 
>     * Now, that the webcam is functional, we need to make it work in applications and web browsers especially, To do so, I followed this nice [article](https://dshedd.com/2023/07/18/fixing-lenovos-mipi-camera-problems-on-ubuntu-22-04/). Go straight to the steps where you need to install _v4l2loopback-dkms_ and _v4l2-relayd_ (drivers are already installed on previous steps described above)
> 
>     * when i ran `v4l2-ctl --list-devices`, i had a permission issue, so i had to put myself in the `video` group (in terminal: `sudo usermod -a -G video YOUR_USERNAME`), then log out and log in, verify that you are part of the group by running `groups` command.
> 
>     * i checked the _v4l2-relayd_ service status, it was enabled and running, **but** still no camera detected in web browser !
> 
>     * something is off, so, i ran `sudo modprobe v4l2loopback` and got:
> 
> 
> > ➜  ~ sudo modprobe v4l2loopback
> > modprobe: ERROR: could not insert 'v4l2loopback': Key was rejected by service
> 
>     * Follow instructions on how to import a key into EFI in this [post](https://vcheng.org/2023/01/22/ubuntu-linux-secure-boot-dkms/)  (**critical part here!**)
> 
>     * After rebooting and making sure your key is enrolled, check the _v4l2-relayd_ service status again (you might need to restart the service)
> 
>     * Test your webcam in the [web browser](https://webcamtests.com/)
> 
> 
> Good luck !
> ### Related links
> 
>     * Ubuntu Intel MIPI webcam wiki: https://wiki.ubuntu.com/IntelMIPICamera
> 
>     * v4l2loopback + v4l2-relayd workaround: https://dshedd.com/2023/07/18/fixing-lenovos-mipi-camera-problems-on-ubuntu-22-04/
> 
>     * Enable MOK on Ubuntu: https://vcheng.org/2023/01/22/ubuntu-linux-secure-boot-dkms/

I followed this until the first `reboot` step, and my camera is working on Chromium and other apps (not Firefox, though)

It is very important to remove all existing IPU6/IPU7 packages first, and this is in the **Troubleshooting** section of the [Intel MIPI Camera wiki](https://wiki.ubuntu.com/IntelMIPICamera),

picousse, 2025-08-01:
> Hi,
>
> After many hours (days?) spent on this, I finally managed to make my _Intel MIPI Camera_ works on my _Dell Precision 5690_ with **Ubuntu 24.04** with kernel **6.14.0-27-generic**.
>
> It is not an [...]

works on xps 14 9440. 
HERO!

Federica-Rike, 2025-08-01:
I've just solved by removing libcamhal0 and installing the new driver that appeared in my list.

It works without switching to the development ppa, and without installing the ipu6-dkms.

Solution is posted in the main thread on discourse as well.

newcolour, 2025-08-02:
@eric-jahn would you care to say what you did in more detail? Did you purge the repo and then update? Because nothing is showing up for me yet. Thanks!

yoelii, 2025-08-04:
Thanks @ksket  and @LuisVasquezBSC

I just followed the first steps until the firts reboot.

```
sudo apt-get remove intel-ipu6-dkms
sudo apt-get autoremove
sudo apt-get update
sudo apt-get upgrade
```

```
ubuntu-drivers list
...
libcamhal-ipu6epmtl
...
```

```
sudo apt-get install libcamhal-ipu6epmtl
```

Check in the "Software and Updates" panel that the additional driver "libcamera-ipu6" is selected and that the camera is working correctly.

Thanks a lot.

Have a nice day

Federica-Rike, 2025-08-07:
Glad to hear it is working!

Federica-Rike, 2025-08-21:
Hi @matthiasheimerl ,

Have you installed also the dkms one? Please remove it.

A number of people (including me) solved this way.

matthiasheimerl, 2025-08-21:
@Federica-Rike 
I removed everything described here https://wiki.ubuntu.com/IntelMIPICamera#Troubleshooting and also executed the steps I marked in my first post. 
I just checked to be sure. intel-ipu6-dkms is not installed

Federica-Rike, 2025-08-21:
@matthiasheimerl That's interesting! Thanks for sharing.

I assume also libcamhal0 is not installed, right?

matthiasheimerl, 2025-08-21:
No this is installed @Federica-Rike 

The execution of `apt-cache depends libcamhal-ipu6epmtl`  shows that  `libcamhal0` is required

Federica-Rike, 2025-08-21:
@matthiasheimerl I solved my issues by removing both libcamhal0 and the dkms, then installing the libcamhal-ipu6epmtl.

Federica-Rike, 2025-08-21:
Let me see if I find in my history how I did it. I remember that was not working and I removed the libcamhal0.

matthiasheimerl, 2025-08-21:
@Federica-Rike  I did it in the same way.
And the package `libcamhal-ipu6epmtl`  installs again `libcamhal0` . At least for me. 

For `libcamhal-ipu6epmtl`  I have the version `0~git202501070628.8346d93~ubuntu24.04.2`

Federica-Rike, 2025-08-21:
@matthiasheimerl, I have just noticed you have a different package!

Mine is libcamhal-ipu6ep.

Senectus, 2025-08-21:
> @Federica-Rike <https://github.com/Federica-Rike> I'm using a Dell
> Precision 5690
> I guess some here have the same notebook running.
> Maybe they can verify it, if it's working on their side.
>
> Hopefully they are reading this

Model: Precision 5690
Kernel: 6.14.0-27-generic
ubuntu-drivers list: libcamhal-ipu6epmtl and intel-ipu6-dkms

This is working for me (after it broke in  6.14.0-24). I can try 6.14.0-27
a bit later.

Senectus, 2025-08-22:
@matthiasheimerl I just updated to 6.14.0-28-generic, still have libcamhal-ipu6epmtl and intel-ipu6-dkms and the cam is still working for me. (on the same Precision 5690) let me know if there is anything else I need to do to find out why we have a different experience.

matthiasheimerl, 2025-08-22:
Hi @Senectus  Interesting that it's working on your side.  I have only installed  `libcamhal-ipu6epmtl`, because the installation of ` intel-ipu6-dkm` didn't worked. 
I only installed the updates. 
I have version `0~git202501070628.8346d93~ubuntu24.04.2` for `libcamhal-ipu6epmtl` and installed it from `http://dell.archive.canonical.com noble/somerville`

Senectus, 2025-08-22:
same here:

apt-cache policy libcamhal-ipu6epmtl
libcamhal-ipu6epmtl:
  Installed: 0~git202501070628.8346d93~ubuntu24.04.2
  Candidate: 0~git202501070628.8346d93~ubuntu24.04.2
  Version table:
 *** 0~git202501070628.8346d93~ubuntu24.04.2 500
        500 http://dell.archive.canonical.com noble/somerville amd64
Packages
        100 /var/lib/dpkg/status

Senectus, 2025-08-25:
> You have really installed `intel-ipu6-dkms`? I don't have it, because this was no longer necessary as, when I understood the previous comments correct to fix the camera
> 
> > same here:
> > apt-cache policy libcamhal-ipu6epmtl
> > libcamhal-ipu6epmtl:
> > Installed: 0~git202501070628.8346d93~ubuntu24.04.2
> > Candidate: 0~git202501070628.8346d93~ubuntu24.04.2
> > Version table:
> > *** 0~git202501070628.8346d93~ubuntu24.04.2 500
> > 500 http://dell.archive.canonical.com noble/somerville amd64
> > Packages
> > 100 /var/lib/dpkg/status
> > […](#)
it appears it got installed 11 days ago. 
I dont explicitly remember installing it, but its is still working.

matthiasheimerl, 2025-08-25:
> it appears it got installed 11 days ago. I dont explicitly remember installing it (and cant see any evidence that _i_ selectively installed it in the bash logs), but its is still working.

I tried to install the intel-ipu6-dkms again, but it failed as described in that topic.  I removed it again completely

kivra-joscan, 2025-08-29:
Still getting build error on latest kernel version of 24.04.
Hardware: ThinkPad X1 Carbon Gen 11
Kernel version: 6.14.0-29-generic (64-bit)

> Building for 6.14.0-29-generic
Building initial module for 6.14.0-29-generic
Error! Bad return status for module build on kernel: 6.14.0-29-generic (x86_64)
Consult /var/lib/dkms/ipu6-drivers/0~git202406240945.aecec2aa-0ubuntu2~24.04.3/build/make.log for more information.
dpkg: error processing package intel-ipu6-dkms (--configure):
 installed intel-ipu6-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 intel-ipu6-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

Any ideas?

matthiasheimerl, 2025-08-29:
I'm wondering that you have installed `intel-ipu6-dkms` and kernel updates are working for you. I had to remove it because I saw the same error as @kivra-joscan 

Today I updated to kernel `6.14.0-29` but the webcam is still not working. I'm only able to run the webcam with version `6.14.0-27` 
I'm not sure if some logs are necessary to see why it's not working.

> > You have really installed `intel-ipu6-dkms`? I don't have it, because this was no longer necessary as, when I understood the previous comments correct to fix the camera
> > > same here:
> > > apt-cache policy libcamhal-ipu6epmtl
> > > libcamhal-ipu6epmtl:
> > > Installed: 0~git202501070628.8346d93~ubuntu24.04.2
> > > Candidate: 0~git202501070628.8346d93~ubuntu24.04.2
> > > Version table:
> > > *** 0~git202501070628.8346d93~ubuntu24.04.2 500
> > > 500 http://dell.archive.canonical.com noble/somerville amd64
> > > Packages
> > > 100 /var/lib/dpkg/status
> > > […](#)
> 
> it appears it got installed 11 days ago. I dont explicitly remember installing it (and cant see any evidence that _i_ selectively installed it in the bash logs), but its is still working.

matthiasverstraete, 2025-09-22:
I haven't been able to get my webcam working in kernel 6.14.0-29-generic (it used to work in 6.11). I get the same compile error as everyone else.
I tried manually installing by following the instructions in the [README](https://github.com/intel/ipu6-drivers?tab=readme-ov-file#3-build-with-dkms). The module installation didn't report any errors and it's listed in `ubuntu-drivers list` and `dkms status`, however, the webcam is still not working.
Any suggestions on how I can investigate this further?

yoelii, 2026-02-13:
Yes, I confirmed. same issue @nathou .

```
Paramétrage de intel-ipu6-dkms (0~git202406240945.aecec2aa-0ubuntu2~24.04.3) ...
Removing old ipu6-drivers-0~git202406240945.aecec2aa-0ubuntu2~24.04.3 DKMS files...
Deleting module ipu6-drivers-0~git202406240945.aecec2aa-0ubuntu2~24.04.3 completely from the DKMS tree.
Loading new ipu6-drivers-0~git202406240945.aecec2aa-0ubuntu2~24.04.3 DKMS files...
Building for 6.17.0-14-generic
Building initial module for 6.17.0-14-generic
Error! Bad return status for module build on kernel: 6.17.0-14-generic (x86_64)
Consult /var/lib/dkms/ipu6-drivers/0~git202406240945.aecec2aa-0ubuntu2~24.04.3/build/make.log for more information.
dpkg: erreur de traitement du paquet intel-ipu6-dkms (--configure) :
 le sous-processus paquet intel-ipu6-dkms script post-installation installé a renvoyé un état de sortie d'erreur 10
Des erreurs ont été rencontrées pendant l'exécution :
 intel-ipu6-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

realill, 2026-02-21:
I went into a rabbit whole fixing this on my Dell laptop using Cursor AI CLI tool with Opus 4.6 model. Was successful to make it running. Took me 2 hours. I asked my agent to summarize issues and solutions. Hope it is useful for you guys. Take it with the grain of salt that some of it may have been hallucinated problems.

---

# Intel MIPI Camera Fix — Ubuntu 24.04 HWE Kernel 6.17

## Problem

Intel MIPI camera (IPU6 + ov01a10 sensor) not working on Ubuntu 24.04 LTS with the HWE kernel (6.17.0-14-generic). The `v4l2-relayd@default.service` crashes repeatedly with GStreamer errors:

```
gst_element_set_state: assertion 'GST_IS_ELEMENT (element)' failed
v4l2allocator: buffer 0 was not queued, this indicate a driver bug
```

## Root Causes

Three separate issues needed to be fixed:

### 1. Empty v4l2loopback modprobe config

`/etc/modprobe.d/v4l2-relayd.conf` was empty. The `v4l2-relayd` service expects the v4l2loopback device to be labeled `"Intel MIPI Camera"` (configured in `/etc/v4l2-relayd.d/default.conf`), but without the modprobe option, v4l2loopback loads with the default name `"Dummy video device (0x0000)"`. The relay can't find the output device and the pipeline fails.

**Fix:**

```bash
echo 'options v4l2loopback card_label="Intel MIPI Camera" exclusive_caps=1' | sudo tee /etc/modprobe.d/v4l2-relayd.conf
```

### 2. v4l2loopback 0.15.0 kernel bug (LP#2136769)

The HWE kernel 6.17 ships v4l2loopback version 0.15.0, which contains upstream commit `5092ea9` that breaks GStreamer's `v4l2sink` buffer queue management on Ubuntu 24.04 (Noble). This is a known bug: [Launchpad #2136769](https://bugs.launchpad.net/ubuntu/+source/v4l2loopback/+bug/2136769).

The v4l2loopback `prepare_buffer_queue()` function pre-populates buffers and returns them via `VIDIOC_DQBUF` without them ever being queued via `VIDIOC_QBUF`, violating the V4L2 spec. See [GitHub #656](https://github.com/v4l2loopback/v4l2loopback/issues/656).

The fix exists in the `linux-oem-6.17` kernel but has not yet been backported to the generic HWE kernel.

**Fix:** Install the patched `v4l2loopback-dkms` (0.12.7) from noble-updates, which overrides the broken in-kernel module:

```bash
sudo apt install v4l2loopback-dkms
```

### 3. Secure Boot MOK key not enrolled

With Secure Boot enabled, the DKMS-built v4l2loopback module is signed with a local MOK (Machine Owner Key), but that key was not enrolled in the firmware. The kernel refuses to load unsigned/untrusted modules.

**Fix:**

```bash
sudo mokutil --import /var/lib/shim-signed/mok/MOK.der
# Set a temporary password when prompted
sudo reboot
```

On reboot, the blue MOK Manager screen appears:

1. Select **"Enroll MOK"**
2. Select **"Continue"**
3. Enter the temporary password
4. Select **"Reboot"**

## Verification

After all fixes and a reboot:

```bash
# Confirm DKMS module is loaded (should show updates/dkms path, not kernel/)
modinfo v4l2loopback | head -3

# Confirm relay service is running
systemctl status v4l2-relayd@default.service

# Confirm camera device exists
cat /sys/devices/virtual/video4linux/*/name | grep "Intel MIPI Camera"
```

## System Info

- **OS:** Ubuntu 24.04.4 LTS (Noble)
- **Kernel:** 6.17.0-14-generic (HWE)
- **CPU:** 13th Gen Intel i7-1360P (Raptor Lake)
- **Camera sensor:** ov01a10
- **IPU:** Intel IPU6
```

Senectus, 2026-02-22:
> > > Thanks [@ksket](https://github.com/ksket) and [@LuisVasquezBSC](https://github.com/LuisVasquezBSC)
> > > I just followed the first steps until the firts reboot.
> > > ```
> > > sudo apt-get remove intel-ipu6-dkms
> > > sudo apt-get autoremove
> > > sudo apt-get update
> > > sudo apt-get upgrade
> > > ```
> > >   
> > > ```
> > > ubuntu-drivers list
> > > ...
> > > libcamhal-ipu6epmtl
> > > ...
> > > ```
> 
> > > ```
> > > sudo apt-get install libcamhal-ipu6epmtl
> > > ```
> 
> > > Check in the "Software and Updates" panel that the additional driver "libcamera-ipu6" is selected and that the camera is working correctly.
> > > Thanks a lot.
> > > Have a nice day
> > 
> > 
> > Confirmed this worked for me also (once i reverted back to 6.14 in the grub config.)
> 
> _sigh_ I'm back again, this time this hasn't worked. I cant get the webcam to work on either kernel 6.14 or 6.17

Ok weirdly, after trying all this again, and giving up I decided if I'm not going to have webcam I might as well use the latest kernel and switched back to 6.17

Now its working again.
unfortunately towards the end i was getting impatient and didn't follow good troubleshooting practice so I'm not sure what change I made that suddenly allowed the webcam to work on 6.17.

fernandob1974, 2026-08-30:
## Reproduction on Samsung Galaxy Book NP960QFG (Raptor Lake + OV02C1)

Adding another data point to this issue. The same broken behavior occurs on
**kernel 7.0.0-30-generic** (Ubuntu 24.04 HWE / Zorin OS 18.1) with both the
in-kernel driver and the upstream `intel-ipu6-drivers` main.

### Hardware
- **Laptop**: Samsung Galaxy Book NP960QFG-KA1AR
- **CPU**: Intel Raptor Lake-P (12th/13th gen, host bridge `0xa707`)
- **IPU6**: PCI `0000:00:05.0` `[8086:a75d]` (IPU6-v3, hardware version 5)
- **Sensor**: OVTI02C1:00 (Omnivision OV02C1) on i2c-0 address 0x36
- **Subsystem**: Samsung `[144d:c1ca]`
- **BIOS**: P07ALN.260.240415.SH
- **Secure Boot**: disabled

### Driver tested
- `intel-ipu6-dkms 0~git202603270946.51fe7248` (from Zorin PPA)
- `linux-modules-ipu6-generic-7.0` (Ubuntu in-kernel)
- `intel/ipu6-drivers` main with manual DWC PHY HSFREQRANGE patch (commit `1f025c6`)

All three exhibit the same failure.

### dmesg (intel-ipu* lines, only)
```
[    2.633510] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    2.642939] intel-ipu6 0000:00:05.0: Found supported sensor OVTI02C1:00
[    2.643149] intel-ipu6 0000:00:05.0: Connected 1 cameras
[    2.644075] intel-ipu6 0000:00:05.0: IPU6-v3[a75d] hardware version 5
[    2.672743] intel_ipu6_isys: loading out-of-tree module taints kernel.
[    2.711217] intel_ipu6_psys.psys intel_ipu6.psys.40: pkg_dir entry count:8
[    2.713832] intel_ipu6_psys.psys intel_ipu6.psys.40: psys probe minor: 0
```

### Key observation
- The PSYS auxiliary device probes and `/dev/ipu-psys0` is created.
- **The ISYS auxiliary device (`intel_ipu6.isys.40`) is created and bound to
  `intel_ipu6_isys.isys` driver, but the probe never logs completion.**
- There is **no `isys probe minor: 0` line** in dmesg.
- There is **no `Loading IPU6 ISYS firmware: ipu6ep_fw.bin`** request.
- There is **no `Loading IPU6 ISP firmware: shisp_2401a0_v21.bin`** request.
- Both firmware files are present in `/lib/firmware/intel/ipu/` (decompressed
  to `.bin` as well as the original `.zst`).

### Reproduction
```bash
$ sudo modprobe intel-ipu6 intel-ipu6-isys intel-ipu6-psys ov02c10
$ ls /dev/video* | wc -l
33
$ v4l2-ctl -d /dev/video10 --stream-mmap=4 --stream-count=5 --stream-to=/tmp/test.raw
        VIDIOC_STREAMON returned -1 (Link has been severed)
$ ls -la /tmp/test.raw
-rw-r--r-- 1 ... 0 ... /tmp/test.raw   # empty
```

### Media topology
The sensor subdev exposes `SGRBG10_1X10/1928x1092` with `link_freq=400 MHz,
pixel_rate=80 MHz` and the link to CSI-2 0 is `[ENABLED, IMMUTABLE]`. But the
CSI-2 receiver reports `fmt: SGRBG10_1X10/4096x3072` (default, unconfigured),
and `media-ctl -V` cannot force it (subdev rejects `VIDIOC_SUBDEV_S_FMT` with
`ENOTTY` / "Inappropriate ioctl").

### HAL `libcamhal-ipu6ep` + `gst-launch-1.0 icamerasrc`
```
CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
ERROR: src pad: Internal data flow error.
CamHAL[ERR] Stream config is not supported. format:V4L2_PIX_FMT_NV12 (1920x1080)
CamHAL[ERR] @configure, analyzeStream failed
CamHAL[ERR] failed to config streams.
```

The HAL loads `AR0234_TGL_10bits.aiqb` which is a Tiger Lake AI profile
(AR0234 sensor), not an OV02C1 one. There is no `OV02C1_*.aiqb` in
`/usr/share/libcamhal/` or the upstream repo, so the HAL is essentially
unusable on this hardware today.

### Suspected root cause
The ISYS probe appears to abort before reaching the `request_firmware()` calls
for `ipu6ep_fw.bin` and `shisp_2401a0_v21.bin`. Compare with a healthy boot
where dmesg shows:
```
intel_ipu6_isys.isys intel_ipu6.isys.40: isys probe minor: 0
intel-ipu6 0000:00:05.0: Loading IPU6 ISYS firmware: ipu6ep_fw.bin
intel-ipu6 0000:00:05.0: Loading IPU6 ISP firmware: shisp_2401a0_v21.bin
intel-ipu6 0000:00:05.0: Loaded firmware: ...
```
None of those lines appear on this RPL a75d system. The most likely places
where the probe aborts are in `intel_ipu6_isys_probe()` ->
`ipu6_isys_init()` -> the `IPU6_ISYS_PCI_CTL` handshake with the ISYS BAR,
or in the CSE communication path that boots the ISYS firmware.

The DWC PHY HSFREQRANGE band selection patch (upstream commit `1f025c6`)
targets Arrow Lake / OV08X40 with `osc_freq_target=208` issues. On RPL
OV02C1 the symptom is identical ("Link has been severed" on STREAMON) but
the link frequency is 400 MHz and the relevant `osc_freq_target` is 335,
so the patch does not apply here.

### Cross-references
- #329 (Galaxy Book4 Pro 360, similar IPU6 setup)
- #369 (this issue, originally)
- #446, #453 (kernel 7.0 backports that I tested)

Happy to provide more diagnostics, run `dynamic_debug`/`ftrace`, or test
patches if you can point me at what to instrument.