intel/ipu6-drivers #372: Linux fedora 6.16.0-0.rc7.60.fc43 Regresion

MRabenda, 2025-07-28:
Hey,

I have Lenovo X1 Carbon Gen 12 with Intel IPU6 - I was running successfully Fedora 42 with working camera. I did an upgrade to rawhide and cammera stopped working - it seems that drivier is crashing - I am using akmod-intel-ipu6-0.0-20.20250115git13c466e.fc43.x86_64

sudo dmesg | grep -i "ipu6\|ivsc"
[   15.374277] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[   15.407083] intel-ipu6 0000:00:05.0: Found supported sensor OVTI08F4:00
[   15.407146] intel-ipu6 0000:00:05.0: Connected 1 cameras
[   15.411304] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[   15.446739] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[   15.493276] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[   15.493288] intel-ipu6 0000:00:05.0: IPU6-v4[7d19] hardware version 6
[   15.748226] RIP: 0010:ipu6_psys_probe+0x384/0x3b0 [intel_ipu6_psys]
[   15.748260]  ? __pfx_ipu6_psys_probe+0x10/0x10 [intel_ipu6_psys]
[   15.748296]  ? __pfx_ipu_psys_init+0x10/0x10 [intel_ipu6_psys]
[   15.748305]  ipu_psys_init+0x86/0xff0 [intel_ipu6_psys]
[   15.748416] Modules linked in: snd_soc_acpi x86_pkg_temp_thermal crc8 intel_powerclamp soundwire_bus coretemp iwlmvm(+) intel_ipu6_psys(OE+) snd_soc_sdca kvm_intel snd_soc_core intel_ipu6_isys mhi_wwan_ctrl mhi_wwan_mbim(+) snd_compress wwan mac80211 ac97_bus videobuf2_dma_sg spi_nor kvm snd_pcm_dmaengine videobuf2_memops videobuf2_v4l2 iTCO_wdt mei_gsc_proxy mtd snd_hda_intel intel_pmc_bxt iTCO_vendor_support videobuf2_common snd_intel_dspcfg irqbypass libarc4 rapl snd_intel_sdw_acpi intel_rapl_msr intel_cstate btusb snd_hda_codec btrtl snd_hda_core btintel btbcm snd_hwdep btmtk snd_seq intel_uncore iwlwifi think_lmi(+) pcspkr hid_sensor_accel_3d snd_seq_device bluetooth mhi_pci_generic hid_sensor_trigger snd_pcm firmware_attributes_class hid_sensor_iio_common processor_thermal_device_pci i2c_i801 industrialio_triggered_buffer snd_ctl_led spi_intel_pci wmi_bmof kfifo_buf mhi usbio(OE) snd_timer spi_intel i2c_smbus processor_thermal_device mei_me cfg80211 industrialio processor_thermal_wt_hint
[   15.748454]  platform_temperature_control mei processor_thermal_rfim processor_thermal_rapl idma64 intel_rapl_common thinkpad_acpi processor_thermal_wt_req intel_ipu6 processor_thermal_power_floor nxp_nci_i2c ipu_bridge platform_profile processor_thermal_mbox igen6_edac nxp_nci snd nci soundcore intel_pmc_core soc_button_array nfc intel_skl_int3472_tps68470 int3403_thermal int340x_thermal_zone tps68470_regulator ov08x40(OE) rfkill clk_tps68470 v4l2_fwnode v4l2_async pmt_telemetry pmt_class intel_skl_int3472_discrete int3400_thermal intel_hid acpi_thermal_rel acpi_pad intel_skl_int3472_common sparse_keymap intel_pmc_ssram_telemetry acpi_tad joydev loop zram lz4hc_compress lz4_compress dm_crypt xe drm_ttm_helper drm_suballoc_helper gpu_sched drm_gpuvm drm_exec drm_gpusvm hid_sensor_hub intel_ishtp_hid i915 ucsi_acpi typec_ucsi typec i2c_algo_bit drm_buddy ttm nvme nvme_core drm_display_helper intel_ish_ipc polyval_clmulni video ghash_clmulni_intel hid_multitouch sha512_ssse3 intel_vpu nvme_keyring sha1_ssse3 intel_ishtp
**[   15.748508] RIP: 0010:ipu6_psys_probe+0x384/0x3b0 [intel_ipu6_psys]**

peigongdsd, 2025-08-22:
Seems like not only did intel_ipu6_psys dead, but libcamera also suffered from API changes of ov02c10 kernel module.
```
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.638285885] [4539]  INFO IPAManager ipa_manager.cpp:137 libcamera is not installed. Adding '/nix/store/src/ipa' to the IPA search path
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.638390870] [4539]  INFO Camera camera_manager.cpp:326 libcamera v0.5.1
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.645820436] [4914]  WARN CameraSensor camera_sensor_legacy.cpp:354 'ov02c10 15-0036': Recommended V4L2 control 0x009a0922 not supported
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.645830982] [4914] ERROR V4L2 v4l2_subdevice.cpp:1196 'ov02c10 15-0036': Unable to get rectangle 2 on pad 0/0: Inappropriate ioctl for device
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.645838132] [4914]  WARN CameraSensor camera_sensor_legacy.cpp:401 'ov02c10 15-0036': The PixelArraySize property has been defaulted to 1928x1092
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.645841107] [4914] ERROR V4L2 v4l2_subdevice.cpp:1196 'ov02c10 15-0036': Unable to get rectangle 1 on pad 0/0: Inappropriate ioctl for device
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.645844676] [4914]  WARN CameraSensor camera_sensor_legacy.cpp:412 'ov02c10 15-0036': The PixelArrayActiveAreas property has been defaulted to (0, 0)/1928x1092
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.645849933] [4914] ERROR V4L2 v4l2_subdevice.cpp:1196 'ov02c10 15-0036': Unable to get rectangle 0 on pad 0/0: Inappropriate ioctl for device
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.645853354] [4914]  WARN CameraSensor camera_sensor_legacy.cpp:420 'ov02c10 15-0036': Failed to retrieve the sensor crop rectangle
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.645855752] [4914]  WARN CameraSensor camera_sensor_legacy.cpp:426 'ov02c10 15-0036': The sensor kernel driver needs to be fixed
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.645857611] [4914]  WARN CameraSensor camera_sensor_legacy.cpp:428 'ov02c10 15-0036': See Documentation/sensor_driver_requirements.rst in the libcamera sources for more information
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.645992755] [4914]  WARN CameraSensorProperties camera_sensor_properties.cpp:463 No static properties available for 'ov02c10'
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.645996881] [4914]  WARN CameraSensorProperties camera_sensor_properties.cpp:465 Please consider updating the camera sensor properties database
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.645998565] [4914]  WARN CameraSensor camera_sensor_legacy.cpp:594 'ov02c10 15-0036': Failed to retrieve the camera location
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.645999952] [4914]  WARN CameraSensor camera_sensor_legacy.cpp:616 'ov02c10 15-0036': Rotation control not available, default to 0 degrees
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.646009160] [4914]  WARN CameraSensor camera_sensor_legacy.cpp:501 'ov02c10 15-0036': No sensor delays found in static properties. Assuming unverified defaults.
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.646486308] [4914]  INFO IPAProxy ipa_proxy.cpp:226 libcamera is not installed. Loading proxy workers from '/nix/store/src/libcamera/proxy/worker'
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.646492593] [4914] ERROR IPAProxy soft_ipa_proxy.cpp:52 Failed to get proxy worker path
Aug 21 03:02:20 ThinkBookX wireplumber[4539]: [0:01:36.646494611] [4914] ERROR IPAManager ipa_manager.h:47 Failed to load proxy
Aug 21 03:02:20 ThinkBookX systemd-coredump[4919]: Process 4539 (wireplumber) of user 1000 dumped core.

```

mmack, 2025-09-04:
Same problem here.

smallorange, 2025-09-10:
Same here
```
[    6.697129] BUG: unable to handle page fault for address: 00000000ff78d008
[    6.697134] #PF: supervisor read access in kernel mode
[    6.697135] #PF: error_code(0x0000) - not-present page
[    6.697137] PGD 0 P4D 0 
[    6.697141] Oops: Oops: 0000 [#1] SMP NOPTI
[    6.697145] CPU: 12 UID: 0 PID: 1149 Comm: (udev-worker) Tainted: G           OE       6.16.5-200.fc42.x86_64 #1 PREEMPT(lazy) 
[    6.697149] Tainted: [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
[    6.697150] Hardware name: LENOVO 21HQSIT024/21HQSIT024, BIOS N3XET40W (1.15 ) 05/22/2023
[    6.697152] RIP: 0010:ipu6_psys_probe+0x384/0x3b0 [intel_ipu6_psys]
[    6.697164] Code: ff c7 44 24 10 fb fd ff ff e9 f8 fc ff ff c7 44 24 10 f4 ff ff ff eb cb 49 8b 84 24 98 03 00 00 48 c7 c6 cf 51 ad c1 4c 89 e7 <48> 8b 40 08 89 43 18 89 c2 e8 8e a5 e3 e4 80 3d 33 6b f3 ff 00 0f
[    6.697166] RSP: 0018:ffffcee981177860 EFLAGS: 00010246
[    6.697169] RAX: 00000000ff78d000 RBX: ffff8bc153c8e028 RCX: 0000000000002000
[    6.697171] RDX: ffff8bc146d58020 RSI: ffffffffc1ad51cf RDI: ffff8bc14e7d1000
[    6.697172] RBP: ffff8bc146d58020 R08: 0000000000000002 R09: ffffcee980ba7000
[    6.697174] R10: ffffcee980ba7000 R11: ffff8bc141b00480 R12: ffff8bc14e7d1000
[    6.697175] R13: ffff8bc153c8e440 R14: ffff8bc146d5a020 R15: ffff8bc146d5a000
[    6.697176] FS:  00007feae3a8d3c0(0000) GS:ffff8bc4e565f000(0000) knlGS:0000000000000000
[    6.697179] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    6.697180] CR2: 00000000ff78d008 CR3: 0000000104809001 CR4: 0000000000f70ef0
[    6.697182] PKRU: 55555554
[    6.697183] Call Trace:
[    6.697186]  <TASK>
[    6.697189]  ? __pfx_ipu6_psys_probe+0x10/0x10 [intel_ipu6_psys]
[    6.697196]  auxiliary_bus_probe+0x46/0x80
[    6.697202]  really_probe+0xdb/0x340
[    6.697206]  ? pm_runtime_barrier+0x55/0x90

```

smallorange, 2025-09-15:
Thank you, Hans  
I pushed a PR to fix the issue.

closed 2025-09-16 by hao-yao