intel/ipu6-drivers #375: Fedora rpmfusion IPU6 packages are out of date

croci, 2025-08-06:
Not sure where to put this so creating an issue here.

All IPU6 packages on Fedora (akmod-intel-ipu6, ipu6-camera-bins, gstreamer1-plugins-icamerasrc, ipu6-camera-hal) have stopped being updated (last time updated around December 2024/January 2025).

xrn, 2025-08-20:
I can confirm - last available version for me is `0.0-21.20250115` for some reason

croci, 2025-09-15:
@smallorange I see you've been working hard on this, thank you very much!

I think a big problem currently is a lack of up-to-date documentation about the current status of IPU6 support. It is very hard to tell which solutions are out of date/out of active development/support, which solutions work, and how to install them/set them up.

For instance, how to make the software ISP solution work? What I can find online are only packages that have not been updated ever since kernel 6.7. Installing rawhide packages on stable Fedora also does not sound like a stable solution. The akmod package is currently the simplest solution from a user viewpoint, but it hasn't been updated in many months...

Anyhow, I saw your pull request and I will be waiting for the akmods to be updated, thank you very much again!

smallorange, 2025-09-16:
Hi,

The upstream solution (upstream isys driver + libcamera software ISP) is available for most of the IPU6-based cameras. By updating to the latest Fedora kernel and installing libcamera, most of the cameras should work (not for all cameras).

If the camera still can't work, the out-of-tree driver can resolve it. The out-of-tree includes an Intel usbio-driver and it drives additional hardware and makes the camera run. All the packages can be found in the RPMfusion repo.

Enable RPMfusion repo:
https://rpmfusion.org/Configuration/

The latest packages can be found in RPMfusion rawhide branch, since #327 hasn't been resolved. But now the solution is there, I'll test it with all the required packages and then push them to the rest of the branch.

smallorange, 2025-09-22:
> So if the package `ipu6-camera-bin` is in the rawhide branch, does this means it will be released in the stable release of fedora also soon? Or in other words, is it safe to install it via rawhide repo when you're on the stable repos?

It stays in RPMfusion. Fedora doesn't ship this solution.
By the way, the upstream IPU6 + libcamera softISP solution can be found in Fedora and you can use it :)

smallorange, 2025-09-22:
> Accually this version of akomd is not able to build with F43, kernel `6.17.0-0.rc5.42.fc43.x86_64`. In the same time we can see multiple new tags this year in this repo. I fully understand that everyone is here to help and appreciate this, but I am confused as a user how should I get working intel-ipu6 with latest patches

The latest build for F43 (rawhide)and F42 can be found in RPMfusion updates-testing repo. 
You can enable it using "Software" app or run
`$sudo dnf config-manager setopt rpmfusion-nonfree-updates-testing.enabled=1`
`$sudo dnf config-manager setopt rpmfusion-free-updates-testing.enabled=1`

After enabling the repo, you can find those updated package.

MRabenda, 2025-09-22:
@smallorange it seems that someone yesterday finally did an update of RPMFusion and we received an upgrade. Te sad part is that someone released everything but without yours fix for kernels +6.16 :( 

```
rpm -qi akmod-intel-ipu6
Name        : akmod-intel-ipu6
Version     : 0.0
Release     : 22.20250909git4bb5b4d.fc43
Architecture: x86_64
Install Date: Sun 21 Sep 2025 09:43:18 PM CEST
```

smallorange, 2025-09-24:
I found a usbio-drivers build issue against F43 kernel. I fixed it and built a new release of the akmod package to include the fix. The package will be available after it gets synced to the updates-testing repo.

MRabenda, 2025-09-24:
Awesome, thank you - curious, should that usbio-drivers bug fix be reflected also in this repo? https://github.com/intel/usbio-drivers/pull/38

smallorange, 2025-09-25:
Included all of them :)