intel/ipu6-drivers #381: Linux kernel 6.16+ breaks suspend to memory

T0mK0, 2025-09-02:
When intel-ipu6-psys module is loaded suspend( s2idle)  does not work.
It just freezes to black screen with no errors on the console and only sysrq or reboot gets out of this state.
Drivers are loaded/compiled with dkms , latest version pulled. 
Removing the modules with:
`dkms  remove ipu6-drivers/0.0.0  --force -k 6.16.X`
makes suspend to work again.

Hardware : 
Operating System: Arch Linux 
Kernel Version: 6.16.4-zen1-1-zen (64-bit)
Processors: 20 × 13th Gen Intel® Core™ i7-13800H
Memory: 32 GiB of RAM (30.9 GiB usable)
Graphics Processor: Intel® Iris® Xe Graphics
Manufacturer: Dell Inc.
Product Name: Precision 5680

Tested on mainline kernels from 6.16rc ( 16.0 16.1. 16.2. 16.3 16.4 ) series to 6.17rc ( rc1 rc2 ) series and all present the same behavior.

**6.15.X and lower work fine.**

peigongdsd, 2025-09-03:
See also https://github.com/intel/ipu6-drivers/issues/372

twouters, 2025-09-03:
I suppose this is a dupe of #372

T0mK0, 2025-09-03:
The camera works ( before the first attempt to suspend ), intel-ipu6-psys is NOT crashing for me.
The only thing that is broken for me is the suspend to ram.
I don't think it has anything to do with #372 , seems to be different problem, IMHO.

twouters, 2025-09-04:
Oh, OK. I noticed a failing suspend and freezes when rebooting before I blacklisted intel-ipu6-psys so I mistook it as the same issue.

twouters, 2025-09-09:
just wondering: does "the camera works" mean that you're able to use it with the gstreamer icamera stuff and binary blobs from https://github.com/intel/ipu6-camera-bins (probably in combination with v4l2-relayd) or do you use the mainline implementation with pipewire in your browser and/or qcam?

The latter currently "works" for me (Dell XPS 9315) but the image quality is inferior to the former, which no longer works since 6.16 because psys module fails to load properly.

paulmenzel, 2026-02-26:
Can you confirm this happens with Linux 6.19 or 7.0-rc1?

Also the article [Intel IPU6 Webcam on Linux: From Proprietary Stack to Mainline](https://jetm.github.io/blog/posts/ipu6-webcam-libcamera-on-linux/) has a good write-up on how to tune some parameters when using the upstream stack.