intel/ipu6-drivers #382: [Bug] Webcam stream fails on Dell Precision 5480 with "Link has been severed" error

milk4candy, 2025-09-03:
Hello,

The webcam on my Dell Precision 5480 is not working on Linux Mint 21.3. After successfully compiling and installing the latest drivers from the main branch, the kernel modules load and the /dev/video* device nodes are created. However, attempting to stream from the device fails.

**Hardware Information:**
- Laptop: Dell Precision 5480 (2023)
- OS: Linux Mint 21.3 Cinnamon
- Kernel: 6.8.0-65-generic
- Camera Sensor (from dmesg): OVTI01A0:00 (OmniVision OV01A10)

**Symptoms:**
- Cheese and Guvcview show a black screen with a "There was an error playing video from the webcam" type of error.
- The user is correctly in the 'video' group.

**Key Diagnostic Output:**
When running `v4l2-ctl --device /dev/video0 --stream-mmap --stream-count=1 --stream-to=test.raw`, I get the following error:
`VIDIOC_STREAMON returned -1 (Link has been severed)`

The output of `v4l2-ctl --list-formats-ext` is:
milk4candy@Precision5480:~$ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture Multiplanar

	[0]: 'Y10 ' (10-bit Greyscale)
	[1]: 'BX24' (32-bit XRGB 8-8-8-8)
	[2]: 'Y12I' (Interleaved 12-bit Greyscale)
	[3]: 'Z16 ' (16-bit Depth)
	[4]: 'YUYV' (YUYV 4:2:2)
	[5]: 'Y8I ' (Interleaved 8-bit Greyscale)
	[6]: 'BG12' (12-bit Bayer BGBG/GRGR)
	[7]: 'GB12' (12-bit Bayer GBGB/RGRG)
	[8]: 'BA12' (12-bit Bayer GRGR/BGBG)
	[9]: 'RG12' (12-bit Bayer RGRG/GBGB)
	[10]: 'BG10' (10-bit Bayer BGBG/GRGR)
	[11]: 'GB10' (10-bit Bayer GBGB/RGRG)
	[12]: 'BA10' (10-bit Bayer GRGR/BGBG)
	[13]: 'RG10' (10-bit Bayer RGRG/GBGB)
	[14]: 'BA81' (8-bit Bayer BGBG/GRGR)
	[15]: 'GBRG' (8-bit Bayer GBGB/RGRG)
	[16]: 'GRBG' (8-bit Bayer GRGR/BGBG)
	[17]: 'RGGB' (8-bit Bayer RGRG/GBGB)
	[18]: 'GREY' (8-bit Greyscale)

milk4candy@Precision5480:~$ v4l2-ctl --device /dev/video0 --stream-mmap --stream-count=1 --stream-to=test.raw
		VIDIOC_STREAMON returned -1 (Link has been severed)

milk4candy@Precision5480:~$ ls -l test.raw
-rw-rw-r-- 1 milk4candy milk4candy 0  9月  3 10:30 test.raw

The output of dmesg after loading the modules is:
milk4candy@Precision5480:~$ dmesg | grep -E "ipu6|ivsc|camera|i2c_hid_acpi"
[    8.501145] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    8.501249] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[    8.501262] intel-ipu6 0000:00:05.0: physical base address 0x628c000000
[    8.501263] intel-ipu6 0000:00:05.0: mapped as: 0x00000000860c2f84
[    8.501308] intel-ipu6 0000:00:05.0: IPU in secure mode
[    8.501309] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    8.501310] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    8.517113] intel-ipu6 0000:00:05.0: IPC reset done
[    8.518988] intel-ipu6 0000:00:05.0: FW version: 20230102
[    8.527356] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[    8.527381] intel-ipu6 0000:00:05.0: physical base address 0x628c000000
[    8.527383] intel-ipu6 0000:00:05.0: mapped as: 0x00000000860c2f84
[    8.527434] intel-ipu6 0000:00:05.0: IPU in secure mode
[    8.527435] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    8.527436] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    8.534819] intel-ipu6 0000:00:05.0: IPC reset done
[    8.535266] intel-ipu6 0000:00:05.0: FW version: 20230102
[    8.536133] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[    8.536153] intel-ipu6 0000:00:05.0: physical base address 0x628c000000
[... 182 lines cut ...]
[   21.323623] intel-ipu6 0000:00:05.0: FW version: 20230102
[   21.394129] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
[   21.394179] intel-ipu6 0000:00:05.0: physical base address 0x628c000000
[   21.394184] intel-ipu6 0000:00:05.0: mapped as: 0x00000000860c2f84
[   21.394372] intel-ipu6 0000:00:05.0: IPU in secure mode
[   21.394376] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[   21.394380] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[   21.412920] intel-ipu6 0000:00:05.0: IPC reset done
[   21.413913] intel-ipu6 0000:00:05.0: FW version: 20230102
[   21.416139] intel-ipu6 0000:00:05.0: Found supported sensor OVTI01A0:00
[   21.416361] intel-ipu6 0000:00:05.0: Connected 1 cameras
[   21.418361] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[   21.436152] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[   21.518510] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[   21.518600] intel-ipu6 0000:00:05.0: IPU6-v3 driver version 1.0
[   21.684008] intel-ipu6-isys intel-ipu6-isys0: bind Intel IVSC CSI nlanes is 1 port is 2
[   21.686358] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[   21.686426] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
[   21.704102] intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.
[  924.696548] intel-ipu6-isys intel-ipu6-isys0: removed
[  924.697210] intel-ipu6-isys intel-ipu6-isys0: unbind Intel IVSC CSI
[  963.615405] intel-ipu6-isys intel-ipu6-isys0: bind Intel IVSC CSI nlanes is 1 port is 2
[  963.616532] intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.

It seems the driver initializes correctly but fails when the stream is started. Please let me know if any other logs are needed. Thank you.

sthibaul, 2025-11-10:
To be noted, though: qcam does get it to work:

```
€ qcam
[1:51:11.744670870] [97716]  INFO Camera camera_manager.cpp:330 libcamera v0.5.2
[1:51:11.755499276] [97720]  WARN CameraSensorProperties camera_sensor_properties.cpp:484 No static properties available for 'ov01a10'
[1:51:11.755506575] [97720]  WARN CameraSensorProperties camera_sensor_properties.cpp:486 Please consider updating the camera sensor properties database
[1:51:11.755510516] [97720]  WARN CameraSensor camera_sensor_legacy.cpp:501 'ov01a10 19-0036': No sensor delays found in static properties. Assuming unverified defaults.
[1:51:11.756808943] [97720]  WARN IPAProxy ipa_proxy.cpp:177 Configuration file 'ov01a10.yaml' not found for IPA module 'simple', falling back to 'uncalibrated.yaml'
[1:51:11.756853781] [97720]  WARN IPASoft soft_simple.cpp:103 IPASoft: Failed to create camera sensor helper for ov01a10
[1:51:11.757022265] [97720]  INFO Camera camera_manager.cpp:220 Adding camera '\_SB_.PC00.LNK1' for pipeline handler simple
[1:51:11.845522853] [97716]  INFO Camera camera.cpp:1215 configuring streams: (0) 1272x800-ABGR8888/Unset
[1:51:11.845731841] [97720]  INFO IPASoft soft_simple.cpp:264 IPASoft: Exposure 4-888, gain 0-65535 (1)
Zero-copy enabled
[1:51:13.834702418] [97723]  INFO Debayer debayer_cpu.cpp:828 Processed 30 frames in 72015us, 2400 us/frame

```

(and the image shows up)

sthibaul, 2025-11-10:
I made wireplumber switch to it with 

```
wireplumber.profiles = {
  main = {
    monitor.v4l2 = disabled
    monitor.libcamera = optional
  }
}
```

It shows up in helvum:

```
Video
 ├─ Devices:
 │      70. ov01a10                             [libcamera]
 │  
 ├─ Sinks:
 │  
 ├─ Sources:
 │  *   86. Built-in Front Camera              
 │  
 ├─ Filters:
 │  
 └─ Streams:
```

but gnome snapshot, or firefox configured with pipewire support, say there is no camera.