intel/ipu6-drivers #386: ov08x40: no camera content available (Dell Pro 13 Premium)

mathstuf, 2025-09-14:
I'm running Fedora 42 and have built the following based on the rpmfusion.org packages and updating snapshots:

- v4l2-relayd 0.1.5
- intel-ipu6-kmod commit intel/ipu6-drivers@4bb5b4d8128fbf7f4730cd364a8f7fc13a0ef65b
  - intel/usbio-drivers@4fb690c6d15a81c492954636c2db396cb700a119 is bundled in this package
- intel/ipu6-camera-hal@c933525a6efe8229a7129b7b0b66798f19d2bef7
- intel/ipu6-camera-bins@30e87664829782811a765b0ca9eea3a878a7ff29

After configuring `v4l2-relayd` to actually start doing something useful, all Cheese and Camera see are all-black. `mpv` kind of hangs and doesn't even open a window. `qcam` sees nothing. `firefox` (via Google Meet) sees the camera, but ends up acting as if nothing is there.

elisei, 2025-10-03:
same thing here...  xps 16 9640

azizkhon96, 2025-12-04:
Why intel doesnt support this sensor, i have hp spectre x360 2024, two years almost , but still no supported driver for this sensor